indilib/indi-3rdparty #269: Streaming with QHY268C not possible

Juergen889, 2020-12-23:
**Describe the bug**
I'm using the actual astroberry server incl. all updates (kstars 3.5.0, libindi 1.8.7.) It is not possible to stream with the camera. Taking pictures workes very well, but no streaming. I just get a "black" picture in the "live video" window (see picture QHY268C_streaming_1.jpg). [image]

After trying to stream with the camera and close the live video window I get odd pictures when taking photos (see picture QHY268C_streaming_2.jpg) To get good pictures again I have to restart kstars!

[image]

In the pictures you see the current settings for the camera streaming and the version of SDK (20.11.26) and indi_qhy_ccd (2.7)

[image]
[image]

Regarding Encoder and Recorder I tryed raw/mjpeg and ser/ogv in all combinations, everytime same result: no streaming. I also set the USB buffer to 4096, without positive result.

I had the same problems with kstars 3.4.3, SDK 20.8.26 and libindi 1.8.6.

**To Reproduce**
1. start streaming
2. black in live video window
3. stop streaming
4. get "black and white picture when capturing
5. restart kstars
6. capturing "good" pictures is ok again, but no streaming

**Expected behavior**
start streaming --> get live video

**Screenshots**
see text

**Desktop (please complete the following information):**
 - OS: astroberry 2.0.3 with actual updates
 - Raspi 4, 4GB memory

**Log Files**
[log_12-38-37.txt](https://github.com/indilib/indi-3rdparty/files/5734944/log_12-38-37.txt)

knro, 2020-12-23:
Can you please try the latest nightly to see if it resolves this issue?

Juergen889, 2020-12-23:
Hi Jasem, 
quick reply :-) 
do you mean the indi-qhy or kstars? Problem is, that I dont have experience in compiling. Or are there already compiled files available?

Regards

Jürgen

Juergen889, 2020-12-23:
just saw, that there is guide how to install nightly in Ubuntu... But it dont work with Raspberry... Sorry for beeing noobie :-)

[image]

Juergen889, 2020-12-23:
short update: I installed Ubuntu in vmware. To check if everything goes well, I first installed kstars 3.5.0.

The camera was not detected by kstars ("please power on...")! So I checked lsusb, it says, that the camera ID is 1618:c268 "Cypress Westbridge".

I rechecked with my Raspi Astroberry, camera is recognized by kstars. BUT: lsusb shows 1618:c269 (not c268)!

I tryed it again with Ubuntu, on all USB ports it was 1618:c268 and kstars dont find the camera at all (pop up with "power on camera" comes up). Same with the nightly build.

On astroberry everything works as usual (with kstars 3.5.0, capturing is ok, but no streaming)

knro, 2020-12-23:
You cannot add Ubuntu PPA to Raspbian. Either your OS updates to latest or compile. There is beta channel support in StellarMate, not sure about other distros.

Juergen889, 2020-12-23:
Ok. So I think I wait for update, at the moment I can live with that, since astroberry has no nightly/beta channel. I will give you update if I have news on it.

Do you have an idea regarding the lsusb issue (see my last comment) I have with the ubuntu (ID1618:c268) and astroberry (ID1618:c269)? Shouldnt it be the same? May this affect the problem with streaming?

Juergen889, 2021-01-28:
Update: I tried it with stellarmate (all updates until today, kstars 3.5.1, QHY SDK 20.12.23): no change, still the same problem!

knro, 2021-01-28:
Looks like you might need to wait until I receive it here and then check it out. I'm going to get the QHY268M, but it should be in principle the same.

Juergen889, 2021-02-03:
Ok, thank you. I just saw, that there is a new QHY SDK. The changelog states the following: 

Fixed: cannot get frame continuously under Linux

Perhaps ist works with the new SDK... Will this be implemented in Stellarmate?

knro, 2021-02-03:
Yes we will update soon. However, I think another issue is that streaming at full resolution is not a good idea. So downsampling is the only way to get better frames IMO.

Juergen889, 2021-03-04:
I updated to stellarmate 1.5.7 (kstars 3.5.2, indi 1.8.9, all other updates till today). Still no streaming possible with QHY268C, still black screen. Good thing after the update: after all the strange behaviour after trying to stream (the black/white pictures) is not happening any more.

Streaming with ASI290mm is possible.

i tried with directly on the Raspi 4B (4GB) and remote over WLAN and LAN, all the same.

I can live with that, would just be good, if it works.

Juergen889, 2021-03-04:
Forgot to mention, that I tried 1x1, 2x2 and 3x3 binning. Same result on all binnings

Juergen889, 2021-03-05:
The new update from indi-qhy from today works! :-)

closed 2021-05-25 by knro