indilib/indi-3rdparty #271: indi_rpicam unable to set shutter speed for long exposures in EKOS

cnieswand, 2020-12-25:
I am using a Raspberry HQ Cam on a PI 4B running Astroberry 2.0.3.
Running KStars/EKOS using the indi_rpicam driver does not allow me to select any reasonable exposure time. Whatever is entered I always get the same image, especially for exposuretimes larger than a second. I can never saturate the sensor by lengthing the exposure time.
I have cloned the indi-3rdparty git repository and can easily modify, compile and debug the indi_rpicam driver.
However whatever I do, there is no way to set the shutter speed of the camera correctly.
Increasing the gain gives saturated output on all color channels.
I also did a rpi-update with no success.

I have noted that the line in mmalcamera.cpp, where the shutterspeed is set, is already marked:

    // FIXME: Seconds does not work completely ok.
    status = mmal_port_parameter_set_uint32(component->control, MMAL_PARAMETER_SHUTTER_SPEED, shutter_speed);
    MMALException::throw_if(status, "Failed to set shutter speed");

So I guess that this is a known problem but I do not see any issue regarding this problem.
Is there any progress?

knro, 2020-12-26:
@lboclboc Can you please take a look at this?

lboclboc, 2020-12-26:
I know there was problem with this earlier. Make sure you use the latest rpicam version. Im not sure what version astroberry is using.

cnieswand, 2020-12-26:
Meanwhile I modified the code, set the camera to burst mode, set the CAPTURE_STATS_PASS just before setting the ananlog gain.

Result is stunning: Trying to create a 9 second exposure gives me on the first capture a 1 second exposure, all subsequent exposures are correct. I checked this by taking a picture of an analog clock. So it has to do something with the preview feature of this camera.

lboclboc, 2021-01-08:
Hi, I have uploaded some fixes now, mostly made by https://github.com/QRainman,  to my master branch https://github.com/lboclboc/indi-3rdparty  that hopefully solves the long exposure problems. You are welcome to test this out.

cnieswand, 2021-01-17:
Unfortunately this does not help.
Assume I define in EKOS a sequence having 9 secs exposure time per frame. Whenever I start indiserver and "connect" indi_rpicam (compiled from your branch, fresh clone from today) the first exposure is 1 second long. All subsequent frames are 9 seconds ... until I disconnect and reconnect. Then it is again 1 second and all subsequent frames again 9 seconds.
I mean as long as I know, I can live with that "feature" ... but it is not nice ... also because the fits info in this first file is wrong, because it is generated from EKOS settings ...

knro, 2021-11-24:
@lboclboc @cnieswand is this still an issue with the latest RPICam driver?

lboclboc, 2021-11-25:
Yes, sadly this is still a problem. I have not been able to find a solution for this yet.

aaronwmorris, 2021-12-19:
I think the problem might be due to the fps settings not allowing the shutterspeed to be set properly.

Here is some debug output from indi

`2021-12-19T21:41:54: Driver indi_rpicam: setExposureParameters(/home/aaron/git/indi-3rdparty/indi-rpicam/mmalcamera.cpp:102): Camera crop set to 0,0,4056,3040
2021-12-19T21:41:54: Driver indi_rpicam:
2021-12-19T21:41:54: Driver indi_rpicam: setExposureParameters(/home/aaron/git/indi-3rdparty/indi-rpicam/mmalcamera.cpp:127): setting fps range 5/1000 -> 166/1000
2021-12-19T21:41:54: Driver indi_rpicam: setExposureParameters(/home/aaron/git/indi-3rdparty/indi-rpicam/mmalcamera.cpp:134): failed to set fps ranges: low range is 327/65536, high range is 10878/65536
2021-12-19T21:41:54: Driver indi_rpicam: setExposureParameters(/home/aaron/git/indi-3rdparty/indi-rpicam/mmalcamera.cpp:142): Failed to set shutter speed, requested 15000000 but actual value is 994639
2021-12-19T21:41:54: Driver indi_rpicam: setExposureParameters(/home/aaron/git/indi-3rdparty/indi-rpicam/mmalcamera.cpp:144): shutter speed set to 994639
2021-12-19T21:41:54: Driver indi_rpicam: setExposureParameters(/home/aaron/git/indi-3rdparty/indi-rpicam/mmalcamera.cpp:151): gain set to 65536/65536
2021-12-19T21:41:54: Driver indi_rpicam: startCapture(/home/aaron/git/indi-3rdparty/indi-rpicam/cameracontrol.cpp:63): shutter speed after enabling camera: 994639`

aaronwmorris, 2021-12-23:
I discovered another interesting behavior.  Long exposures do work, but only if the first (throw away) exposure is requested as 2 seconds or longer.  If the first exposure is 1s or shorter, none of the exposures will exceed 1s even if you request longer.

aaronwmorris, 2021-12-28:
Correction, a 2 second throw away exposure allows up to a 6-7 second exposure.  Taking a 7 second throw away exposure seems to allow the maximum exposure times.

sajmons, 2022-05-03:
I can confirm that with INDi 1.9.5 and Kstars 3.5.8, taking first exposure at exactly 7 sec, solves the problem. All exposures that you take after that are corect.

sajmons, 2022-07-18:
According to this, we are getting fix for this problem soon! Yeeea!
https://github.com/indilib/indi-3rdparty/pull/603/commits/028c59a1feb66cbf9411f5242e81d5c28abce929

closed 2022-11-16 by knro