indilib/indi-3rdparty #283: Nexdome "open shutter on dome unpark" control not working as expected

xthestreams, 2021-01-06:
**Describe the bug**
The control panel for the the Nexdome driver offers the ability to set an "open on unpark" flag for the driver. 

This is a great idea, especially as EKOS Scheduler does not provide the ability to do more than Unpark the dome at the moment. 

However, I have not been able to get it to work as expected. The "close on park" feature IS working.
 
**To Reproduce**
Exact steps to reproduce the behavior.
1. set driver to enable Open On Unpark 
2. unpark dome (manually or under the control of Ekos Scheduler)
3. dome shutter does NOT open automatically

**Expected behavior**
Dome should open on "unpark" - however I think the preference is that the dome shutter could be controlled under Ekos scheduler

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS:Stellarmate
 - Version 1.8.8
 - 
**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

xthestreams, 2021-01-07:
Jassem - just so you know the new feature works - GREAT!

BUT, it's introduced a new bug. With the feature enabled, the dome does not slave or respond to manual rotate commands as it believes it is still "parked" - even though its state is showing as UnParked" in the control panel.

Once the "open on unpack" feature is disabled the you can park and unpack as expected and the dome slaves as expected.

knro, 2021-01-07:
Can you please attach dome log that shows this? I believe some state is not getting set correctly, let's find out from log what is it.

knro, 2021-01-13:
Any update on the logs?

xthestreams, 2021-01-13:
sorry I posted the logs in the new bug report - can you see it or should I move it here?

knro, 2021-01-13:
Post here or put link to it.

xthestreams, 2021-01-13:
updated!

knro, 2021-01-13:
the log looks wrong, please delete that and attach a file.

xthestreams, 2021-01-13:
done

knro, 2021-01-13:
Thanks for the log, the problem is here:

```
[2021-01-13T18:46:08.495 AEDT DEBG ][           org.kde.kstars.indi] - NexDome : "[DEBUG] RES <:SES,46000,46000,0,0> "
[2021-01-13T18:46:08.496 AEDT DEBG ][           org.kde.kstars.indi] - NexDome : "[DEBUG] Processing event <:SES,46000,46000,0,0> with value <46000,46000,0,0> "
```

The message has the format: position, maximum position, open_limit_switch, close_limit_switch

When you have the shutter fully opened, open_limit_switch should be 1, but it's zero. Do you know why? Before the code was also checking if we reached maximum position as another indicator, but I see it as commented I think perhaps it caused some issues. At any rate, the limit switch should be actuated when the shutter is fully opened but it's not reported as such.

xthestreams, 2021-01-13:
That makes sense. The sensor is in place last I checked, it certainly automatically stops as it opens.

knro, 2021-01-13:
Check if you can see from some other software the limit switch value.

xthestreams, 2021-01-13:
I'm trying to work out how to do this, I might just see if I can putty into the dome and look at what comes out of the port when I send an open command.
Having said that, why do you think the dome rotation problem ONLY occurs when the "open on unpark" option is elected. The shutter opens and the dome rotates fine otherwise?

xthestreams, 2021-01-13:
Thanks for the hint Jassem - I determined that the @RWS command enables the dome owner the set the maximum travel limits for the dome open and as such the shutter will stop where instructed BEFORE it reaches the limit switch (why you woudl want this I am not sure, but it's an option). Somewhere/somehow my dome had been set to 46000 and the limit switch is not/was not reached until 46097, hence the glad was never triggered. I've changed it now and will do testing, I'll also update the driver page to advise of this unless we think the current behaviour can be/shoudl be made to work?

commit f3eb8efd (2021-01-06): Do not use IUFindOnSwitchIndex with NOFMANY switch type. Should fix #283 shutter open on unpark
closed 2021-01-15 by xthestreams