indilib/indi-3rdparty #292: libatik needs a min glibc version requirement

paolostivanin, 2021-01-13:
It seems that with PR #123 , the min glibc version for `libflycapture.bin` has been increased to 2.27

```
strings libatik/x64/libflycapture.bin | grep GLIBC_2.27
GLIBC_2.27
```

We should update the CMakeLists accordingly. What do you think?
As I don't own an ATIK camera, I can't verify if it works correctly with previous glibc version.

knro, 2021-01-13:
@TallFurryMan Can you please check?

TallFurryMan, 2021-01-13:
Good point, I will check. And probably also update de SDK in the process if there is a new version, it's been several months.

TallFurryMan, 2021-01-21:
The SDK is updated on my github branch, I will run a few tests and push a MR. I am not under the impression that the glibc dependency is mandatory. It seems to be an internal static link.

paolostivanin, 2021-01-21:
In case it might help, testing it requires an old Debian (stretch) or an old Ubuntu (xenial) or openSUSE 15.2 :+1:

TallFurryMan, 2021-01-23:
Tested OK on Ubuntu 18.04 32-bit.

TallFurryMan, 2021-01-23:
I sent a mail to Atik support about the C99 `powf` symbol in `libflybycapture`.

I also mentioned that, in my observatory, when the device, initially plugged in the usb port but not powered, is powered, the first driver connection will only get zero frames. The second connection will resolve the problem.

TallFurryMan, 2021-01-31:
Atik did acknowledge my question, awaiting feedback.

knro, 2021-03-12:
Any update on this issue?

knro, 2021-05-25:
@TallFurryMan Checking if there is any update on this issue from Atik?

grantbrown, 2021-09-24:
I believe [this issue](https://github.com/indilib/indi-3rdparty/issues/382) is likely related. I can use an Atik EFW 2.2 on an Ubuntu 18.04 system, but on 20.04 and newer it fails. Any ideas about a workaround?

paolostivanin, 2021-09-24:
I don't think this issue is related to your bug. Ubuntu 20.04 is using glibc 2.31 (while 18.04 was using 2.27), so this shouldn't be an issue. _Usually_, glibc is backwards compatible.

grantbrown, 2021-09-24:
Gotcha. I tried running 18.04 under docker and that didn't fix the issue, since from what I understand that won't change glibc. Under Gnome Boxes 18.04 works *shrug*. Atik promised a reply back in April but I haven't heard back.

paolostivanin, 2021-09-24:
libflycapture:

```
GLIBC_2.3
GLIBC_2.2.5
GLIBC_2.27
GLIBCXX_3.4.20
GLIBCXX_3.4.11
GLIBCXX_3.4.9
GLIBCXX_3.4.15
GLIBCXX_3.4.21
GLIBCXX_3.4
GLIBC_2.9
GLIBC_2.4
GLIBC_2.7
GLIBC_2.14
GLIBC_2.8
GLIBC_2.11
GLIBC_2.3.2
GLIBC_2.3.4
```

libatikcameras:
```
GLIBC_2.2.5
GLIBC_2.4
GLIBC_2.17
GLIBC_2.14
GLIBC_2.3.4
GLIBCXX_3.4.20
GLIBCXX_3.4.11
GLIBCXX_3.4.9
GLIBCXX_3.4.21
GLIBCXX_3.4
```

That's very weird, it should work fine with Ubuntu >= 18.04 and Fedora 33...

paolostivanin, 2021-09-24:
Do you have the following deps installed?

```
$ readelf -a libflycapture.bin| grep NEEDED
 0x0000000000000001 (NEEDED)             Shared library: [libgomp.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libraw1394.so.11]
 0x0000000000000001 (NEEDED)             Shared library: [librt.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libusb-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [ld-linux-x86-64.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libpthread.so.0]
[24/Sep/2021 22:48:40] pstivanin at suse-ws in ~/Documents/obs/Application:Astrophotography/libindi-3rdparty-libs/indi-3rdparty-1.9.2/libatik/x64
```
```
$ readelf -a libatikcameras.bin| grep NEEDED
 0x0000000000000001 (NEEDED)             Shared library: [libusb-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libudev.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libpthread.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libflycapture.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
```

paolostivanin, 2021-09-24:
Just out of curiosity, could you please try also openSUSE Tumbleweed (https://get.opensuse.org/tumbleweed)? You need to install the OS and then add the Astro repo like this:
```
sudo zypper ar -f -p 90 https://download.opensuse.org/repositories/Application:/Astrophotography/openSUSE_Tumbleweed/ Astrophotography
```
and then execute:
```
sudo zypper ref
sudo zypper in -y indi-atik kstars
```

then you can start Kstars -> Ekos, add your profile, etc...

grantbrown, 2021-09-24:
All good questions - as far as the shared libraries, I believe all those should have been available. I'll run a "clean" test tonight in a new Boxes machine so that I can answer precisely and completely.

grantbrown, 2021-09-25:
Progress!!! Following the shared libraries you identified, it seems like most were available, but I installed libgcc1 and ib64stdc++6 on a new install Ubuntu 20.04 and it ended up working. It seems I was entirely incorrect about this being related to glibc, but I'm still confused about why things worked out on 18.04 just following the usual build instructions and cmake output. 

It seems like this should maybe be moved to a new issue about cmake dependencies in indi-atik or the libatik folder? Thanks a million times for your help - let me know if you still want me to test out openSUSE or how else I can help document the solution.

paolostivanin, 2021-09-25:
Cool, I'm glad to hear that it's working :)
About Ubuntu 18.04 I can think of only two things: either you or a package pulled in those deps or they were installed by default.
Nah, if it works when the needed deps are installed, then there's no need to test openSUSE too. Our build system does reverse deps analysis, so it automatically brings in the needed deps, so if the packager forgets things do not break :)

closed 2021-09-28 by knro