indilib/indi-3rdparty #294: Patch for cameras with little control over exposure parameters

simont77, 2021-01-19:
Hi,
I'm using  a Nikon V1 with indi_gphoto_ccd. The camera has several limitation in parameters setting (e.g. exposure and iso), but I can successfully acquire images once I set the parameters in camera. However, with the present version of indi_gphoto_cdd, since no exposure widget is present, kstars timeouts for exposure longer then few seconds (in fact, only 1s and bulb are currently present for cameras with no exposure widget). Moreover, since no iso widget is present, kstars does not allow me to set file format to native. Of coarse, using bulb in not an option, it does not work.
I made a patch (see link below) to add fake widgets for exposure and iso, which of course I cannot use for actual setting, but allow me to operate correctly the camera after I manually set the parameters. 
If it is of general interest for cameras with similar limitations, I can issue a PR.

https://github.com/simont77/indi-3rdparty/commit/0a2268226a598c685f67d8a4fc36e59d8a3ee633

knro, 2021-01-19:
Thank you Simon. Shouldn't this be fixed at libgphoto2 level?

simont77, 2021-01-19:
I'm not sure. The camera cannot be configured (it's a camera limitation), and I see no way in which libgphoto2 may help. My issues arise:
- for the exposure time, since kstars will try to use bulb for not listed exposure time (but bulb is not available on this camera). I can simply leave exposure in kstars at 1s, but it may became confusing. At this point I would prefer that kstars simple avoids any reference to exposure time and remembers me that is set "In Camera".
- for the image format, since kstars assumes that only cameras with iso are DSLR, and exposes the native format option only in this case.

By the way, I've already filed an issue on libgphoto2 repo, because for this camera the timeout on capture was much lower than for the others.

knro, 2021-05-25:
@simont77 How did it go at the libgphoto2 end?

simont77, 2021-05-25:
The timeout issue is solved, so now you can set the camera in manual mode and 30s exposure and it works. I’m currently using my patched version of indi-gphoto-ccd with no issue. Unfortunately as I said, this camera does not support bulb, neither exposure time nor ISO setting from USB, so I believe nothing more can be done in gphoto. 
I understand that this is essentially an issue due to the severely limited remote control capability of the camera, so I fully understand if you are not willing to add these “fake” widgets to the driver. I don’t know how many user of such limited cameras there are, and I suppose that each camera will be limited in a different way. At the end of the day, I can live with my patched driver until I will move to a more serious camera.

knro, 2021-05-26:
Right, thanks for understanding. The driver needs to be as generic as possible and while we added some cases for some camera models, they're usually on the camera manufacturer level. Alright, will close this issue for now.

closed 2021-05-26 by knro