indilib/indi-3rdparty #298: indi-celestronaux does'nt work on USB PC Port of CGX

tgghtp, 2021-01-20:
Hello,

hoped the driver would work when I attach the RPI to the USB PC Port of my CGX, but it does'nt .

Did I misunderstand the usage?

On the other hand, when i connect this port it to my PC, a serial port appears and I can watch the communication with my  written program, it displays the traffic between HC, GPS and so on when scope powered on.

Using the USB Serial on Handcontroller it works as expected.

knro, 2021-01-20:
Please report this here: https://indilib.org/forum/development/8246-indi-celestron-aux.html

Celestron GPS driver doesn't work for your setup?

jochym, 2021-01-21:
If the PC port presents itself as serial and shows AUX traffic it should work. Please provide some more info about your setup.  What is the output of `lsusb` after connecting the scope and what are the lines in `dmesg` after connection.

tgghtp, 2021-01-21:
reported as knro wrote here: https://indilib.org/forum/development/8246-indi-celestron-aux.html
All details there.

tgghtp, 2021-01-21:
direct link: https://indilib.org/forum/development/8246-indi-celestron-aux.html?start=108#66289

knro, 2021-09-28:
Is this still an issue with latest INDI release? Please confirm.

jochym, 2021-09-28:
It should not be a problem. If it is, it should be visible in other drivers as well - CAUX mostly uses INDIs comm procedures - the only difference is the serial handshake functions, which are slightly irregular in celestron scopes and had to be handled manually in our code. I have no hardware to test so the @tgghtp has to confirm if this is an issue. @fabriziop is your scope equipped with USB PC port?

fabriziop, 2021-09-28:
@jochym no, I have only serial ports. Just to refresh what does the driver at handshake time when the port is of serial type. It first checks if hardware handshake is present. If yes, it means that the connection is to AUX or PC serial connectors with 19200 speed. If no, the connection is supposed to be to HC USB port or to a mount USB port with 9600 speed. Just an hypotesis: CGX USB PC port may have a different speed than 9600.

closed 2025-08-04 by knro