indilib/indi-3rdparty #308: INDI celestron_aux fails to build on Raspberry Pi4

ewolfer67, 2021-01-29:
I ran the following script to set up celestron_aux.

```bash
#!/bin/bash

# Download dependencies
sudo apt install cmake libindi-dev build-essential

# Create projects directory
mkdir -p ~/Projects/

# Download indi 3rd party source 
cd ~/Projects/
git clone https://github.com/indilib/indi-3rdparty

# Setup celestron_aux driver
mkdir -p ~/Projects/build/indi-caux
cd ~/Projects/build/indi-caux
cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-celestronaux
make -j4
sudo cp indi_celestron_aux /usr/bin/indi_celestron_aux
```

During the `make -j4` I encountered the following error:
```make -j4
[ 33%] Building CXX object CMakeFiles/indi_celestron_aux.dir/celestronaux.cpp.o
/home/astroberry/Projects/indi-3rdparty/indi-celestronaux/celestronaux.cpp: In member function ‘virtual bool CelestronAUX::updateProperties()’:
/home/astroberry/Projects/indi-3rdparty/indi-celestronaux/celestronaux.cpp:635:9: error: ‘defineProperty’ was not declared in this scope
         defineProperty(&CordWrapSP);
         ^~~~~~~~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-celestronaux/celestronaux.cpp:635:9: note: suggested alternative: ‘deleteProperty’
         defineProperty(&CordWrapSP);
         ^~~~~~~~~~~~~~
         deleteProperty
/home/astroberry/Projects/indi-3rdparty/indi-celestronaux/celestronaux.cpp: In member function ‘virtual void CelestronAUX::ISGetProperties(const char*)’:
/home/astroberry/Projects/indi-3rdparty/indi-celestronaux/celestronaux.cpp:695:5: error: ‘defineProperty’ was not declared in this scope
     defineProperty(&NetDetectSP);
     ^~~~~~~~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-celestronaux/celestronaux.cpp:695:5: note: suggested alternative: ‘deleteProperty’
     defineProperty(&NetDetectSP);
     ^~~~~~~~~~~~~~
     deleteProperty
make[2]: *** [CMakeFiles/indi_celestron_aux.dir/build.make:76: CMakeFiles/indi_celestron_aux.dir/celestronaux.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/indi_celestron_aux.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
It complains about not finding `defineProperty`, which seems to have been added in #299.

**Desktop (please complete the following information):**
 - OS: Raspbian Buster - Astroberry Server 
 - Linux astroberry 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux

ewolfer67, 2021-01-29:
Can confirm that switching to #297 solved the problem, so it must have been introduced in #299.

knro, 2021-01-29:
Maybe install libindi first? latest libindi from Github must be installed to system then re-compile indi-celestronaux ?

ewolfer67, 2021-01-30:
My sh script installed libindi first.

`sudo apt install cmake libindi-dev build-essential`

knro, 2021-01-30:
You need to compile libindi from Github since the change is recent. Or use INDI nightly PPA.

ewolfer67, 2021-01-30:
I tried PPA but got this error:
Error: could not find a distribution template for Raspbian/buster

ewolfer67, 2021-01-30:
I managed to get celestron aux to build successfully by also building libindi from source.
Question: Is there any easier way to get this setup on a Pi? The nightly repo isn't compatible with buster and the shipped version of libindi-dev (astroberry) is broken with my hardware.

closed 2021-01-30 by ewolfer67