indilib/indi-3rdparty #310: indi_celestron_aux driver can't connect with mount motor controllers error.

ewolfer67, 2021-01-30:
Raspberry Pi (Astroberry) connected to bottom of Nexstar hand controller (non-usb) via an RS232-to-USB cable.
Load celestron_aux driver (0.7) and get the following errors attempting to connect:

2021-01-30T10:28:42: [ERROR] Cannot continue without connection to motor controllers. 
2021-01-30T10:28:42: [ERROR] Got no response from target ALT or AZM. 
2021-01-30T10:28:41: [INFO] Trying to contact telescope motor controllers. 
2021-01-30T10:28:41: [INFO] Setting serial speed to 9600 baud. 
2021-01-30T10:28:41: [INFO] Detected Hand Controller serial connection.

Mount was aligned before attempted connection.
Hand controller

dmsg displays:

[    2.071181] usb 1-1.2: new full-speed USB device number 3 using xhci_hcd
[    2.126471] systemd[1]: Set hostname to <raspberrypi>.
[    2.216708] usb 1-1.2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[    2.216728] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.216746] usb 1-1.2: Product: CP2102 USB to UART Bridge Controller
[    2.216764] usb 1-1.2: Manufacturer: Silicon Labs
[    2.216779] usb 1-1.2: SerialNumber: 0001

jochym, 2021-01-30:
It seems like the connection *was* established after all. But I guess it does not work?
Any further info about the communication we can use to debug this?
Logs will be usefull.

ewolfer67, 2021-01-30:
I'm a noob but here are the logs:

[2021-01-30T19:01:44.395 CET INFO ][                org.kde.kstars] - Welcome to KStars 3.5.0 Stable
[2021-01-30T19:01:44.395 CET INFO ][                org.kde.kstars] - Build: 2020-12-06T18:17:19Z
[2021-01-30T19:01:44.395 CET INFO ][                org.kde.kstars] - OS: "raspbian"
[2021-01-30T19:01:44.395 CET INFO ][                org.kde.kstars] - API: "arm-little_endian-ilp32-eabi-hardfloat"
[2021-01-30T19:01:44.395 CET INFO ][                org.kde.kstars] - Arch: "arm"
[2021-01-30T19:01:44.395 CET INFO ][                org.kde.kstars] - Kernel Type: "linux"
[2021-01-30T19:01:44.396 CET INFO ][                org.kde.kstars] - Kernel Version: "5.4.83-v7l+"
[2021-01-30T19:01:44.396 CET INFO ][                org.kde.kstars] - Qt Version: 5.11.3
[2021-01-30T19:01:45.338 CET INFO ][                org.kde.kstars] - Processing  "unnamedstars.dat" , HTMesh Level 3
[2021-01-30T19:01:45.339 CET INFO ][                org.kde.kstars] -   Sky Mesh Size:  512
[2021-01-30T19:01:45.425 CET INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "unnamedstars.dat"
[2021-01-30T19:01:46.299 CET WARN ][                org.kde.kstars] - "Star HD20794 not found."
[2021-01-30T19:01:46.335 CET WARN ][                org.kde.kstars] - "Star HD98230 not found."
[2021-01-30T19:01:46.376 CET DEBG ][                       default] - File opened:  "/usr/share/kstars/ngcic.dat"
[2021-01-30T19:01:46.376 CET INFO ][                org.kde.kstars] - Loading NGC/IC objects
[2021-01-30T19:01:47.840 CET DEBG ][                       default] - File opened:  "/home/astroberry/.local/share/kstars/comets.dat"
[2021-01-30T19:01:48.227 CET WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-30T19:01:48.236 CET WARN ][                org.kde.kstars] - "Object named NGC 6050A not found"
[2021-01-30T19:01:48.238 CET WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-30T19:01:48.270 CET WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-30T19:01:48.270 CET WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-30T19:01:48.271 CET WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-30T19:01:48.271 CET WARN ][                org.kde.kstars] - "Object named Ceres not found"
[2021-01-30T19:01:48.271 CET WARN ][                org.kde.kstars] - "Object named Pallas not found"
[2021-01-30T19:01:48.272 CET WARN ][                org.kde.kstars] - "Object named Vesta not found"
[2021-01-30T19:01:48.272 CET WARN ][                org.kde.kstars] - "Object named Cybele not found"
[2021-01-30T19:01:48.272 CET WARN ][                org.kde.kstars] - "Object named Psyche not found"
[2021-01-30T19:01:48.272 CET WARN ][                org.kde.kstars] - "Object named Juno not found"
[2021-01-30T19:01:48.272 CET WARN ][                org.kde.kstars] - "Object named Themis not found"
[2021-01-30T19:01:48.272 CET WARN ][                org.kde.kstars] - "Object named Astraea not found"
[2021-01-30T19:01:48.273 CET WARN ][                org.kde.kstars] - "Object named Lumen not found"
[2021-01-30T19:01:48.273 CET WARN ][                org.kde.kstars] - "Object named Ida not found"
[2021-01-30T19:01:48.273 CET WARN ][                org.kde.kstars] - "Object named Mathilde not found"
[2021-01-30T19:01:48.273 CET WARN ][                org.kde.kstars] - "Object named Eros (1898 DQ) not found"
[2021-01-30T19:01:48.273 CET WARN ][                org.kde.kstars] - "Object named Hektor (1907 XM) not found"
[2021-01-30T19:01:48.287 CET WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-30T19:01:48.288 CET WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-30T19:01:48.288 CET WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-30T19:01:48.288 CET WARN ][                org.kde.kstars] - "Object named Ceres not found"
[2021-01-30T19:01:48.288 CET WARN ][                org.kde.kstars] - "Object named Pallas not found"
[2021-01-30T19:01:48.288 CET WARN ][                org.kde.kstars] - "Object named Vesta not found"
[2021-01-30T19:01:48.289 CET WARN ][                org.kde.kstars] - "Object named Cybele not found"
[2021-01-30T19:01:48.289 CET WARN ][                org.kde.kstars] - "Object named Psyche not found"
[2021-01-30T19:01:48.289 CET WARN ][                org.kde.kstars] - "Object named Juno not found"
[2021-01-30T19:01:48.289 CET WARN ][                org.kde.kstars] - "Object named Themis not found"
[2021-01-30T19:01:48.289 CET WARN ][                org.kde.kstars] - "Object named Astraea not found"
[2021-01-30T19:01:48.289 CET WARN ][                org.kde.kstars] - "Object named Lumen not found"
[2021-01-30T19:01:48.289 CET WARN ][                org.kde.kstars] - "Object named Ida not found"
[2021-01-30T19:01:48.290 CET WARN ][                org.kde.kstars] - "Object named Mathilde not found"
[2021-01-30T19:01:48.290 CET WARN ][                org.kde.kstars] - "Object named Eros (1898 DQ) not found"
[2021-01-30T19:01:48.290 CET WARN ][                org.kde.kstars] - "Object named Hektor (1907 XM) not found"
[2021-01-30T19:01:48.505 CET WARN ][                org.kde.kstars] - Current icon theme is "PiXflat"
[... cut ...]
[2021-01-30T19:02:02.327 CET INFO ][           org.kde.kstars.indi] - Celestron AUX :  "[INFO] Trying to contact telescope motor controllers. "
[2021-01-30T19:02:03.374 CET INFO ][           org.kde.kstars.indi] - Celestron AUX :  "[ERROR] Got no response from target ALT or AZM. "
[2021-01-30T19:02:03.377 CET INFO ][           org.kde.kstars.indi] - Celestron AUX :  "[ERROR] Cannot continue without connection to motor controllers. "
[2021-01-30T19:02:03.381 CET INFO ][           org.kde.kstars.ekos] - "Celestron AUX is disconnected."

jochym, 2021-01-30:
Versions of your hardware and software would help in diagnosing the issue.

ewolfer67, 2021-01-31:
Apologies.

OS: Raspbian Buster - Astroberry Server
Linux astroberry 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux

knro, 2021-02-27:
Please test against latest git

jochym, 2021-02-27:
Even better against celestron_aux_work branch

kenkelso, 2021-03-20:
This is the setup I have with no connection problem, hardware maybe?

closed 2021-09-28 by knro