indilib/indi-3rdparty #315: ZWO CCD driver crashes INDI Server

ewolfer67, 2021-02-05:
Crashing INDI Server while running Astroberry with Celestron AUX and ZWO CCD drivers.

[image]

OS: Raspbian Buster - Astroberry Server
Linux astroberry 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux

[2021-02-04T16:38:05.349 PST DEBG ][           org.kde.kstars.indi] - Adding managed driver "ZWO CCD"
[2021-02-04T16:38:05.349 PST DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_asi_ccd"
[2021-02-04T16:38:05.350 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_asi_ccd"
[2021-02-04T16:38:05.351 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_asi_ccd"
[2021-02-04T16:38:05.351 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-02-05T00:38:05: Driver indi_asi_ccd: pid=13870 rfd=10 wfd=14 efd=15"
[2021-02-04T16:38:05.351 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2021-02-04T16:38:05.386 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-02-05T00:38:05: Driver indi_asi_ccd: indi_asi_ccd: symbol lookup error: indi_asi_ccd: undefined symbol: _ZThn4768_N4INDI3CCD13GuideCompleteE12INDI_EQ_AXIS"
[2021-02-04T16:38:05.386 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2021-02-04T16:38:05.387 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  "Child process 13870 died"
[2021-02-04T16:38:05.388 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-02-05T00:38:05: Driver indi_asi_ccd: stderr EOF"
[2021-02-04T16:38:05.388 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-02-05T00:38:05: Driver indi_asi_ccd: Terminated after #0 restarts."
[2021-02-04T16:38:05.388 PST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2021-02-04T16:38:05.388 PST CRIT ][           org.kde.kstars.indi] - INDI driver  "indi_asi_ccd"  crashed!

ewolfer67, 2021-02-05:
Update.

It turns out that ANY CCD driver loaded into the Camera slot crashes INDI Server. Only loading the CCD Simulator doesn't crash it.

I hooked my ZWO and mount up to my Mac and it loaded and ran fine. Must be a RPI/Linux issue?

knro, 2021-02-05:
Looks like some libindi was not installed correctly or the driver wasn't build correctly against it.

ewolfer67, 2021-02-06:
I ran this set of commands to install my necessary drivers and still have the issue.

# Update system
`sudo apt update && sudo apt upgrade`

# Download dependencies
`sudo apt-get install -y libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libtiff-dev libfftw3-dev`

`sudo apt remove libindi-dev`

# Create projects directory
`mkdir -p ~/Projects/`
`cd ~/Projects/`

# Setup indi core library
`git clone https://github.com/indilib/indi.git`
`mkdir -p ~/Projects/build/indi-core`
`cd ~/Projects/build/indi-core`
`cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi`
`make -j4`
`sudo make install`

# Download indi 3rd party source 
`cd ~/Projects/`
`git clone https://github.com/indilib/indi-3rdparty`

# Setup celestron_aux driver
`mkdir -p ~/Projects/build/indi-caux`
`cd ~/Projects/build/indi-caux`
`cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-celestronaux`
`make -j4`
`sudo cp indi_celestron_aux /usr/bin/indi_celestron_aux`

knro, 2021-03-12:
I cannot reproduce the issue still.

pawel-soja, 2021-03-15:
```
$ c++filt "_ZThn4768_N4INDI3CCD13GuideCompleteE12INDI_EQ_AXIS"
non-virtual thunk to INDI::CCD::GuideComplete(INDI_EQ_AXIS)
```
It is possible that some packages were not compiled with each other.

@ewolfer67 From your description it appears that you compile **indi**, and only **celestron_aux**.
The problem occurs with **indi_asi_ccd**, which you do not update.

ewolfer67, 2021-03-15:
Resolved itself somehow.

closed 2021-03-15 by ewolfer67