indilib/indi-3rdparty #317: read mode support in indi_qhy_ccd doesn't put camera into new read mode

nrackwit, 2021-02-06:
For the QHY294M-Pro which supports alternate read modes (11MP, 47MP resolutions) I found that the read mode support in the indi_qhy_ccd driver fails to switch my camera to a different read mode. In either read mode the frames I can obtain will only be from the 11MP (read mode = 0) resolution.

To reproduce, I connect the camera through ECOS/Indi. The driver successfully identifies that the camera has two read modes and provides a switch for changing read modes in the main control tab.

Changing the read modes in the main control tab although has no effect on the image resolution in the image settings and image info tabs nor on the obtained frames. When taking exposures with either read mode, the frame resolution  will always be 4164px x 2795px which is the effective area of the imaging sensor excluding the overscan/calibration areas in the 11M mode. I can however use the image settings tab to lower the resolution as long as I keep the frame size at or below 4164px x 2795px. 

I would expect that by toggling the read mode I am able to obtain frames in the two different read modes with different resolutions at 4212x2850 px for the  11M read mode and 8432x5648 px for the 47M read mode. This is the full sensor resolution supported by the camera including overscan/calibration areas. The respective read modes and sensor properties should be also reflected in the image info and settings tabs.

I am using locally compiled versions from master of Indicore and 3rd-party and the same same behavior exists for OS X 11.1 and on linux/arm32.

Attached are the logs and screen shots from a short session of two frames each taken in a different read mode and the corresponding console output. In the screenshot note that the setting switches to read mode = 1 and the camera replies with the higher read mode resolution of the 47M mode but then the driver defaults back to the 11M read mode.  

I have been opening the code and doing a few tests myself including reaching out for help in the QHY forum and I think that I have a good understanding of the problem. Attached is my attempt to patch the issue which Please see my post on the Indi Developer forum (https://indilib.org/forum/development/8760-should-alternate-read-modes-work-on-qhy294m-pro-with-indi-qhy.html). I am also attaching sources for the driver which, at least for my camera, resolve the problem.

Indi console output

2021-02-06T05:43:16: [WARNING] Telescope aperture is missing. 
2021-02-06T05:43:16: [WARNING] Telescope focal length is missing. 
**2021-02-06T05:43:09: [INFO] GetQHYCCDReadModeResolution in this ReadMode: imageW: 4164 imageH: 2795 
2021-02-06T05:43:09: [INFO] GetQHYCCDReadModeResolution in this ReadMode: [ imageW: 8336 > imageEAW: 4164 ] / [ imageH: 5648 > imageEAH: 2795 ] 
2021-02-06T05:43:09: [INFO] Read mode updated to 1** 
2021-02-06T05:43:09: [INFO] GetQHYCCDEffectiveArea: subX :36 subY: 28 subW: 4164 subH: 2795 
2021-02-06T05:42:54: [WARNING] Telescope aperture is missing. 
2021-02-06T05:42:54: [WARNING] Telescope focal length is missing. 
2021-02-06T05:42:45: [INFO] Debug is enabled. 
2021-02-06T05:42:45: [INFO] World Coordinate System is enabled. 
2021-02-06T05:42:44: [INFO] Device configuration applied. 
2021-02-06T05:42:44: [INFO] USB Buffer updated to 4096 
2021-02-06T05:42:44: [INFO] USB Traffic updated to 255 
2021-02-06T05:42:44: [INFO] GetQHYCCDReadModeResolution in this ReadMode: imageW: 4164 imageH: 2795 
2021-02-06T05:42:44: [INFO] GetQHYCCDReadModeResolution in this ReadMode: [ imageW: 4164 > imageEAW: 4164 ] / [ imageH: 2822 > imageEAH: 2795 ] 
2021-02-06T05:42:44: [INFO] Read mode updated to 0 
2021-02-06T05:42:44: [INFO] GetQHYCCDEffectiveArea: subX :36 subY: 28 subW: 4164 subH: 2795 
2021-02-06T05:42:44: [INFO] Offset updated to 0 
2021-02-06T05:42:44: [INFO] Gain updated to 400 
2021-02-06T05:42:44: [INFO] Upload settings set to client only. 
2021-02-06T05:42:44: [INFO] Debug is disabled. 
2021-02-06T05:42:44: [INFO] Loading device configuration... 
2021-02-06T05:42:44: [INFO] USB Traffic Settings: Value: 0 Min: 0 Max: 255 Step 13 
2021-02-06T05:42:44: [INFO] Offset Settings: Value: 0 Min: 0 Max: 200 Step 1 
2021-02-06T05:42:44: [INFO] Gain Settings: Value: 1 Min: 0 Max: 3624 Step 1 
2021-02-06T05:42:44: [INFO] Current read mode: 0 
2021-02-06T05:42:44: [INFO] Camera exposure limits: Min: 0.000001s Max: 3600s Step 0s 
2021-02-06T05:42:44: [INFO] Number of read modes: 140728898420738 
2021-02-06T05:42:44: [INFO] Using QHY SDK version 20.12.11 
2021-02-06T05:42:41: [INFO] Connected to QHY294PROM-86dda84baef688d7b. 
2021-02-06T05:42:41: [INFO] Debug is enabled. 

Indi log
[log_00-42-21.txt](https://github.com/indilib/indi-3rdparty/files/5936331/log_00-42-21.txt)

[image]

Updated driver code

[qhy_ccd.h.txt](https://github.com/indilib/indi-3rdparty/files/5936346/qhy_ccd.h.txt)
[qhy_ccd.cpp.txt](https://github.com/indilib/indi-3rdparty/files/5936345/qhy_ccd.cpp.txt)

knro, 2021-02-06:
Thank you, can you please submit a PR?

knro, 2021-03-12:
was this issue also fixed in the latest few PRs?

nrackwit, 2021-03-12:
The camera re-Initialization after read mode changes was added in [PR #319](https://github.com/indilib/indi-3rdparty/pull/319) and updated again in [PR #341](https://github.com/indilib/indi-3rdparty/pull/341)

closed 2021-03-12 by knro