indilib/indi-3rdparty #318: New feature request to support enabling/ignoring the overscan/calibration regions of my camera

nrackwit, 2021-02-06:
So this request somehow goes along with my other request that I just posted around read modes  (https://github.com/indilib/indi-3rdparty/issues/317) but is about a new feature that's not yet supported in the qhy driver (not sure about others...) which is to add driver support for the overscan/calibration areas of the imaging sensor.

Enabling the calibration/overscan areas may have some good value for calibrating the images during post processing and there isn't a good reason I can think of why the image capture software shouldn't be able to support handling those areas on the imaging sensor. 

Basically, what other imaging applications can do is to give an option whether to include or to ignore the overscan areas. Since the QHY SDK supports querying those properties I think that it's straight forward to implement it into the indi driver and the code I shared in the linked ticket actually also tries to address this.

The idea is that if the sensor is recognized to have an overscan area then the driver will add a new control to the main control tab for including or ignoring the overscan area. Based on this user setting the driver will then adjust frame offset and width of the frame so that the image info and image settings get modified to the respective frame resolution and the user doesn't need to manually account for setting a proper frame offset for excluding overscan areas.

knro, 2021-03-12:
Hello, was this incorporated in the last few patches?

nrackwit, 2021-03-12:
The Overscan Area support was in [PR #319](https://github.com/indilib/indi-3rdparty/pull/319)

closed 2021-03-12 by knro