indilib/indi-3rdparty #324: indi_gpsd constantly reports 'NO FIX' against gpsd 3.22

wgauvin, 2021-02-10:
**Describe the bug**
A clear and concise description of what the bug is.  Provide a general summary of the issue in the Title above as well as affected program name like `indiserver` or `indi_toupcam_ccd`

On both my server and a Raspberry Pi that are running Arch Linux based distro
(server is purely Arch Linux, RPi4 is Manjaro) `indi_gpsd` keeps report 'NO FIX'. This
happens in Ekos as well as running indiserver from command line. When running `cgps`/`xgps`
they both report getting fixes. This doesn't happen on a Raspbian install. The version
of `gpsd` is 3.22 changed `STATUS_NO_FIX` from being 'no' to 'no, or unknown'.

The change in GPSD was commit 094c0af, https://gitlab.com/gpsd/gpsd/-/commit/094c0af4ea3f8e25a84c4643c370c17a88ef7878

However, `indi_gpsd` is expecting status >= STATUS_FIX. This can be seen
https://github.com/indilib/indi-3rdparty/blob/master/indi-gpsd/gps_driver.cpp#L282-L286

I've tried a local patch indi_gpsd reports having a fix.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Use a Linux distro that you can install `gpsd` (I use Arch Linux / Manjaro)
2. Install `gpsd` (v3.22), `libindi` (1.8.8), `libindi-gpsd` (1.8.8)
3. Insert USB GPS dongle, make sure udev has picked up the dongle
4. Run `gpsd` with:
```
/usr/bin/gpsd -b -n -D 2 -N -G /dev/gps0
```
5. Use cgps/xgps to view when the GPS has a fix
6. Start `indiserver` and `indi_gpsd` from command line
```
indiserver -vvv indi_gpsd
```
7. Use `indi_setprop` to have indi_gpsd to listen
```
indi_setprop "GPSD.CONNECTION.CONNECT=On"
```
8. Indi server keeps reporting 'NO FIX'

**Expected behavior**
A clear and concise description of what you expected to happen.

If the mode value is set from MODE_2D or greater than indi_gpsd should
report a fix. This works on versions of gpsd  < 3.22.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
Desktop
 - OS: ArchLinux (5.10.14-arch1-1)
 - Version:  5.10.14-arch1-1
 - Libraries:
gpsd 3.22-1
libindi 1.8.8-1
libindi-gpsd 1.8.8-1

RPi
- OS: Manjaro
- Version: 5.10.14-1-MANJARO-ARM
- Libraries:
gpsd 3.22-1
libindi 1.8.8-1
libindi-gpsd 1.8.8-1

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2021-02-10:
Thank you. Can you submit a PR that handles both versions for gpsd?

wgauvin, 2021-02-10:
@knro - definitely

commit 106aecf4 (2021-02-13): Fix indi gps not getting fix. fixes #324 (#325)
closed 2021-02-13 by knro