indilib/indi-3rdparty #334: indi_eqmod_telescope:  Mount time reverts on each new connection.

setnes, 2021-02-26:
New connections to INDI cause the mount's time to be synced using the previous EQMod Mount.TIME_UTC values.  The mount time is always set to the static values (previously used) on new INDI connections.

**Steps to reproduce**

- Start the EQMod Mount driver and set the time (either manually, or via a client).  I'm using CCDCiel.
- Navigate to the mount's Site Management tab on the INDI control panel so you can see the driver's Julian Date.
- Wait about a minute so the time jump will be obvious.
- Run indi_getprop from a shell window.  There's no need to query anything specific.
- Note that the Julian Date on the Site Management tab jumped backwards.  It was reset to the previously synced time.
- This can be tested with other clients too.  Open PHD2 and connect to the mount.  The same thing will happen to the mount's date.

**Expected behavior.**
Unless the TIME_UTC values are explicitly being updated, the mount's time should not be affected by new INDI connections, queries, or otherwise.

OS:  Raspbian (Astroberry Server)
EQMod Mount Driver Version:  1.0

This bug is hidden by clients or drivers that regularly update the mount's time.  I am aware of the GPSD feature to sync the time every second.  It is not desirable though as that causes location jitter and the mount driver config file is re-written on every location change.  So every second I end up with a new /home/astroberry/.indi/EQMod Mount_config.xml written to my SD card.  🤦

knro, 2021-02-27:
I just tested this and cannot reproduce this behavior with simulators.

setnes, 2021-02-27:
The Julian date never stops incrementing, but it jumps back to the previously set time.  This happens for every single indi_getprop command.  It happens when PHD2 connects.  It happens when I open the INDI control panel.  Every time a new connection to INDI occurs, the time gets reset.

knro, 2021-02-27:
Yes, I don't see it jumping back as well. I've used many indi_getprop and they don't affect it at all. Something else must be happening at your end.

setnes, 2021-02-27:
Hmm... I have never seen it NOT do this since I started using INDI.  I have always had to use the GPSD workaround... and it's not ideal.  I am not using Ekos/Kstars.  Perhaps that code is doing something to prevent this behavior?

knro, 2021-02-27:
Try to isolate this. Run EQMod driver alone and check.

setnes, 2021-02-27:
That test was worthwhile.  It does NOT occur when only the EQMod Mount driver is in use.  It DOES occur when the GPSD driver is in use.  So if the GPSD driver is loaded and has synced the time, then you should be able to reproduce this.  The GPSD and EQMount driver simulators are showing this.

So... steps to be clear.
1) Load EQMod Mount and GPSD drivers.  Simulate and start both.
2) Leave GPSD refresh period at 0, but hit the GPS refresh button once.
3) Navigate to the mount's Site Management tab.
4) Query something with indi_getprop (or connect with something like PHD2 that isn't configured to update the time)
5) You should see the Julian Date jump backwards.

setnes, 2021-02-27:
I made a very short video that shows this jump.

https://user-images.githubusercontent.com/12475102/109395009-b4c10e80-78ef-11eb-9008-c8bd1ded71f5.mp4

closed 2021-02-27 by knro

setnes, 2021-02-27:
Thanks!  I'm excited for this change.  This behavior has confused me for a long time.

I have tried building this though, and it doesn't look like it worked.  On Astroberry I needed to git and build indi core and 3rd party drivers.  I can see your code change in the source that I built, but the time is still jumping backwards when I test it.

knro, 2021-02-27:
Are you sure the updated INDI GPSD driver is installed to the system?

setnes, 2021-02-27:
Yes.  I just tweaked the version number in CMakeLists.txt to version 0.6 and rebuilt the code and installed it.  I can see 0.6 in the INDI control panel.  And here's a screenshot that shows your comment in the code. 

[image]

setnes, 2021-02-27:
I can try rebuilding it all with deep git clones instead of shallow clones (based on the build instructions).  Maybe there's something in that that will make a difference.  <shrug>

setnes, 2021-02-27:
I rebuilt everything from deep git clones.  It's still jumping backwards in time.

knro, 2021-02-27:
I can't replicate that now, just tested again. It stays on time. Check if the Location and Time properties in GPSD become IDLE once indi_getprop is called. THis is the change that happened in this fix. Btw, INDI GPSD driver is the one that got updated, not INDI EQMOd.

setnes, 2021-02-27:
Ah... They switch to idle if you issue an indi_getprop command without options.  Then subsequent calls do not make the time jump. 

However, try testing by querying a specific item after refreshing GPSD.

`indi_getprop "EQMod Mount.CONNECTION.CONNECT"`

instead of just

`indi_getprop`

So it IS better in some situations, but not all.  We were testing with slightly different indi_getprop commands.

commit 67ad76ee (2021-02-27): Fix an issue in GPSD driver where it can send out of date Location and Time properties when a new client tries to connect to it AFTER it already obtained the fix before.

Fixes #334