indilib/indi-3rdparty #339: SONY A7S imageQuality error

peppuccio, 2021-03-01:
**Describe the bug**
When using last gphoto driver, setting Imagequallity to RAW will download first a JPG then the RAW version in the next photo. 

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Take a photo assigning RAW (imagequality set value = 3)
3. Dowload a JPG 
4. Took a second photo (change point of view) then you will see the first one (driver download raw version of previous photo)

**Expected behavior**
Everything seems to happen as Raw+JPG was selected but download take the first photo and second one is downloaded with the next request.
 
**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Rapsberian buster
 - last version

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[log_21-51-23.txt](https://github.com/indilib/indi-3rdparty/files/6064656/log_21-51-23.txt)
See capture line 3780

knro, 2021-03-02:
Can you also report this at libgphoto2 issue tracker as well?

jurgenfd, 2021-03-02:
FYI That would be at https://github.com/gphoto/libgphoto2/issues

peppuccio, 2021-03-02:
> Can you also report this at libgphoto2 issue tracker as well?

Ok.

peppuccio, 2021-03-04:
I think the issue is in indi driver. if I execute "$ gphoto2 --set-config imagequality=3" before entering Kstar/ekos, then only captxxxx.arw are downloaded. I seems that the DEBUG information "setting radio/menu widget imagequality: 3  (RAW)" does not correspond to anything.

knro, 2021-03-04:
The settings returns OK as you can see in the log. If you select the NEXT entry in Ekos (whatever comes after RAW), does it work? Maybe some indexes are out of alignment in libgphoto2.

peppuccio, 2021-03-04:
In Ekos everything goes apparently fine. You have no error message, but if you examine the images then you will notice that the fits viewer doesn't show the last image you take. As I said in a prior message, in Ekos you take a first image and the viewer is showing it, then you take a second image and the viewer is still showing the first, you take a third image and the viewer is showing the second and so on.  You're right it sounds like indexes misaligned or a cache problem.

peppuccio, 2021-03-04:
I finally understand you NEXT entry in EKOS. The next one is RAW+ JPG but it's not allowed. So I put the first one "STANDARD".
Right now only .ARW files are downloaded but still with the shift I describe in my prior message.
[image]

peppuccio, 2021-03-04:
The same capture using bulb mode
[image]

closed 2021-09-28 by knro