indilib/indi-3rdparty #342: RPI-Driver unable to recognise v1 camera

damienh, 2021-03-03:
**Describe the bug**

With a Raspberry Pi v1.3 camera selecting RPI Camera driver results in 

```
2021-03-03T11:13:09: [WARNING] Connect: Unknown camera name: ov5647 
2021-03-03T11:13:09: [INFO] MMAL device connected successfully! 
2021-03-03T11:13:09: [INFO] Debug is enabled. 
```

I noticed in the driver description it should support the v1 camera

> This is an INDI driver for Raspberry Pi High Quality Camera, the V2 camera (imx219 chip) and the V1 camera (omv5647 chip).

If I then select connect the driver crashes and restarts with the issue outlined above. 

**To Reproduce**
Exact steps to reproduce the behaviour.
1. Attach Raspberry Pi v1 camera to the Raspberry Pi
2. Select RPI Driver Indi control panel
3. See warning error in logs. 

**Expected behaviour**

I would assume id see it connected and the dialogue for the RPI Driver.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Astroberry 2.0.3

**Log Files**

```
2021-03-03T11:13:09: [WARNING] Connect: Unknown camera name: ov5647 
2021-03-03T11:13:09: [INFO] MMAL device connected successfully! 
2021-03-03T11:13:09: [INFO] Debug is enabled. 
```

damienh, 2021-03-03:
Worth noting that the camera does work with V4L2 Driver, so the camera itself is fine.

knro, 2021-03-03:
@lboclboc Can you please check this out?

lboclboc, 2021-03-10:
Sorry for being slow, this really should work yes. Ill try to find some time to look into it. I did notice that some system updates can modify the camera, like the HiQ camera changed name from test.c to imx477 at some stage. Please try to update all packages on your rpi.

damienh, 2021-03-11:
I dont have the specific version I am on to hand, but I know I ran `sudo apt update && sudo apt upgrade && sudo apt dist-upgrade` about 10 days ago, so I assume I am pretty up to date. The install itself is only 3 weeks old.

knro, 2021-11-24:
@damienh same issue with recent driver?

knro, 2022-11-16:
We'll be migrating to INDI Libcamera driver so I don't think this driver will be maintained for long.

closed 2022-11-16 by knro