indilib/indi-3rdparty #358: EQMOD Issue

SebNik1, 2021-03-16:
**Describe the bug**
I am experienced difficulties in combination with PHD2. Usually I have trouble during backlash test in south direction, the mount does not seem to respond. Today I tried the star cross test with all debug output enabled. It failed in the fourth movement which is south and I got the following output:
[2021-03-16T18:04:27.910 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] Timed guide South 1000 ms at rate 13.537  "
[2021-03-16T18:04:27.911 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] StartDETracking() : trackspeed = -13.537 arcsecs/s, computed rate = -0.9 "
[2021-03-16T18:04:27.911 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] SetDERate() : rate = -0.9 "
[2021-03-16T18:04:27.912 CET INFO ][           org.kde.kstars.indi] - EQMod Mount :  "[WARNING] Warning: Invalid parameter -> Can not change rate while motor is running (direction differs). "

I set up eqmod to 0.9x guiding speed on both axis while guiding siderial. Eqmod is running in the latest stable version on my indi server. I am connected via Gigabit LAN from my macbook (Mojave) with kstars and phd2. 

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver on Indi server remotely (or local, I did not try that)
2. start guiding and try PHD2 Star cross test
3. if reproducable you will see a failure on the fourth direction, which is the south movement. No trouble on all other three axis

While testing it would be great to perform:
1. PHD2 calibration
2. PHD2 guiding assistant for 120s at least including measurement of declination backlash, which usually failes in south direction
3. if related see a failure while guiding south after doing the backlash pyramid to the north

I did not see an error message there yet, but I focused on the mechanical part on my heq5pro before.

**Expected behavior**
guiding response working in all four directions reliably; Sometimes guiding is working though

**Screenshots**
just the log available

**Desktop (please complete the following information):**
 - OS: Mojave 10.14.6; Kstars 3.5.2 stable; Indiserver on ubuntu mate 2
[log_17-57-37.txt](https://github.com/indilib/indi-3rdparty/files/6152263/log_17-57-37.txt)
0.4.2
 - Version 1.2 Interface 5

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

[log_17-57-37.txt](https://github.com/indilib/indi-3rdparty/files/6152274/log_17-57-37.txt)

SebNik1, 2021-03-19:
After rereading the log I am wondering if the motor status check routine is missing for the faulty direction.

bengal123, 2021-03-25:
I'm having the same issue. A number of other users have reported it here too: https://www.cloudynights.com/topic/730090-heq5-pro-sudden-huge-backlash/

knro, 2021-03-25:
The driver was not updated in quite a bit. I just tried to manually guide pulse to south in EQMod Mount --> Motion Control and it worked OK. I put 1000 at GuideNS --> SOUTH and pressed SET. Change N/S Rate to 0.9

Can you trigger this error from the INDI Control Panel directly?

@geehalel Do you have any opinion on this issue?

bengal123, 2021-03-25:
Looking at this a bit more closely it looks like it could be a bug in PHD2. 

From the log attached above we can see the North steps resulting in a `StartDETracking(13.537)` call, followed by a `StartDETracking(-13.537)` about 100ms later:

[2021-03-16T18:04:26.807 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] Timed guide North 1000 ms at rate 13.537  "
[2021-03-16T18:04:26.807 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] StartDETracking() : trackspeed = 13.537 arcsecs/s, computed rate = 0.9 "
...
[2021-03-16T18:04:27.827 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[MOUNT] StopMotor() : Axis = 2 "
...
[2021-03-16T18:04:27.910 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] Timed guide South 1000 ms at rate 13.537  "
[2021-03-16T18:04:27.911 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] StartDETracking() : trackspeed = -13.537 arcsecs/s, computed rate = -0.9 "
[2021-03-16T18:04:27.911 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] SetDERate() : rate = -0.9 "
[2021-03-16T18:04:27.912 CET INFO ][           org.kde.kstars.indi] - EQMod Mount :  "[WARNING] Warning: Invalid parameter -> Can not change rate while motor is running (direction differs). "
```

This is specifically disallowed by the `INDI::Telescope` class here: https://github.com/indilib/indi/blob/f12c8952697b2aeef846637ffae4aba9cd1e4206/libs/indibase/inditelescope.h#L57

There is a StopMotor command issued between the North and South steps, but it's not clear that the code waits for the Stop to take affect before continuing.

There may also a related bug in PHD2's Guiding Assistant when it is evaluating mount backlash (presumably its doing a similar sequence of North steps followed by South steps). 

Hopefully the same bug isn't present elsewhere?

SebNik1, 2021-03-25:
Hi Jasem,

I only find it reproducible during the star cross test. It might be a problem if several short guide commands come in. Manual guiding works for me, too. 

Besides the star cross test the south part of the  backlash test gives really strange results.

pchev, 2021-03-25:
I guide with phd2 and the eqmod driver for years and never get this messages.

As of phd2 code, it never change tracking directly, all the pulse guide operation are done with the TELESCOPE_TIMED_GUIDE_* properties.

bengal123, 2021-03-26:
>  guide with phd2 and the eqmod driver for years and never get this messages.

I've only seen the issue with the PHD2 star cross test and the PHD2 Guiding Assistant backlash measurement. Otherwise my PHD2 guiding is fine.

I've created an INDI client app which duplicates the error [here](https://gist.github.com/bengal123/4bbe4dcb0f5bd6ba474f7a4590f72655) (based on [tutorial_six](https://github.com/indilib/indi/blob/master/examples/tutorial_six/tutorial_client.cpp)). All it does is issue a TELESCOPE_TIMED_GUIDE_NS north, wait 10ms, then issue a TELESCOPE_TIMED_GUIDE_NS south.

To use it:
1. Start indiserver with the EQMOD driver and a real HEQ5 Pro attached
2. Unpark the mount and start tracking
4. Run the [console app](https://gist.github.com/bengal123/4bbe4dcb0f5bd6ba474f7a4590f72655), and hit enter to start the guide-north-then-south that generates the error.

On my system this gives the error reliably:
```
2021-03-26T07:46:52: [DEBUG] Timed guide North 13 ms at rate 7.52053 (Inverted) 
2021-03-26T07:46:52: [DEBUG] StartDETracking() : trackspeed = -7.52053 arcsecs/s, computed rate = -0.5 
2021-03-26T07:46:52: [DEBUG] SetDERate() : rate = -0.5 
2021-03-26T07:46:52: [MOUNT] SetMotion() : Axis = 2 -- dir=backward mode=slew speedmode=lowspeed 
2021-03-26T07:46:52: [SCOPE] CheckMotorStatus() : Axis = 2 
2021-03-26T07:46:52: [COMM] dispatch_command: ":f2", 4 bytes written 
2021-03-26T07:46:52: [COMM] read_eqmod: "=301", 5 bytes read 
2021-03-26T07:46:52: [MOUNT] SetSpeed() : Axis = 2 -- period=305546 
2021-03-26T07:46:52: [COMM] dispatch_command: ":f2", 4 bytes written 
2021-03-26T07:46:52: [COMM] read_eqmod: "=301", 5 bytes read 
2021-03-26T07:46:52: [COMM] dispatch_command: ":I28AA904", 10 bytes written 
2021-03-26T07:46:52: [COMM] read_eqmod: "=", 2 bytes read 
2021-03-26T07:46:52: [MOUNT] StartMotor() : Axis = 2 
2021-03-26T07:46:52: [COMM] dispatch_command: ":J2", 4 bytes written 
2021-03-26T07:46:52: [COMM] read_eqmod: "=", 2 bytes read 
2021-03-26T07:46:52: [DEBUG] StartDETracking() : trackspeed = -0 arcsecs/s, computed rate = 0 
2021-03-26T07:46:52: [COMM] dispatch_command: ":f2", 4 bytes written 
2021-03-26T07:46:52: [COMM] read_eqmod: "=311", 5 bytes read 
2021-03-26T07:46:52: [MOUNT] StopMotor() : Axis = 2 
2021-03-26T07:46:52: [COMM] dispatch_command: ":K2", 4 bytes written 
2021-03-26T07:46:52: [COMM] read_eqmod: "=", 2 bytes read 
2021-03-26T07:46:52: [DEBUG] End Timed guide North/South 
2021-03-26T07:46:52: [DEBUG] Timed guide South 13 ms at rate 7.52053 (Inverted) 
2021-03-26T07:46:52: [DEBUG] StartDETracking() : trackspeed = 7.52053 arcsecs/s, computed rate = 0.5 
2021-03-26T07:46:52: [DEBUG] SetDERate() : rate = 0.5 
2021-03-26T07:46:52: [WARNING] Warning: Invalid parameter -> Can not change rate while motor is running (direction differs). 
```

The cause of the error appears to be that the motor status is not updated after the call to `StopMotor()` between the two guiding commands, so `DERunning` still indicates the mount is running in the second call to `SetDERate()`, causing the error.

I'm not familiar enough with the codebase to know if this is a valid way of fixing the bug, but inserting a `ReadMotorStatus(Axis2)` in [Skywatcher::SetDERate](https://github.com/indilib/indi-3rdparty/blob/6d7e25f6219783539997368ee3cc46d3fdd24990/indi-eqmod/skywatcher.cpp#L1056) does stop the error from being generated with the indi client app above.

When I get a clear sky I'll see if it also fixes the PHD2 star cross test issue and the Guiding Assistant backlash estimation.

Let me know if this sounds like a valid fix, and if you'd like a PR for it.

knro, 2021-03-26:
@bengal123 Thanks, this indeed demonstrates the issue quite well. I think your fix is appropriate. I think normally this error is not seen because ReadMotorStatus(..) is called every second, but not on very short timescales.

SebNik1, 2021-03-26:
> @bengal123 Thanks, this indeed demonstrates the issue quite well. I think your fix is appropriate. I think normally this error is not seen because ReadMotorStatus(..) is called every second, but not on very short timescales. Please submit a PR when ready.

Thank you all very much for your effort!

xsnrg, 2021-03-26:
Indeed, thank you. I went through a bunch of pain with this as well. I am not sure why I am not having the problem now. It could be because I changed the guide exposure to 2 seconds, allowing more time for everything to settle between pulses.

https://indilib.org/forum/mounts/8171-eqmod-can-not-change-rate-while-motor-is-running-direction-differs.html?start=0

pchev, 2021-03-26:
Very good news, thank you!

This explain I never see the issue because I always guide with 3 seconds exposure and there is a very long time I not run a star cross test or a backlash measurement.

pchev, 2021-03-27:
I get some clear sky tonight to test @beaglebot change.

First I confirmed I get the error using the current version of the driver, with both the star cross test and guiding with very short exposure of 0.1 sec. and high DEC aggressivity, chasing the seeing with quick reversal.

Then I recompile eqmod with the addition of the two ReadMotorStatus. 
Now the star cross test complete successively and the short exposure guide session run without any error message.

Finally I run a guide session using my standard 3 sec. exposure and everything also work fine. 

So for me this can be merged without problem.

beaglebot, 2021-03-27:
Fantastic - thanks @pchev

SebNik1, 2021-03-28:
This sounds great. Thank you!

blemasle, 2021-03-28:
I'm having this problem for months now, and wasn't able to pinpoint the issue precisely.

Looking forward for the updated driver! Thank you very much!

PR #369 (2021-03-28): Fix for "Can not change rate while motor is running (direction differs)" error when guiding
Resolves #358 

(I've made this PR using my @beaglebot github account, rather than the @bengal123 account I used in the issue earlier)

A number of EQMOD users reported that the PHD2 star cross test was failing, and the PHD Guiding Assistant would incorrectly report large DE backlash values (mine was 22000ms). The following message was present in the EQMOD log when these errors occured:
```
 "[WARNING] Warning: Invalid parameter -> Can not change rate while motor is running (direction differs). "
```

T [... cut]
closed 2021-03-28 by knro