indilib/indi-3rdparty #365: RPiCam driver don't run in Ekos

HamHamFonFon, 2021-03-24:
I've installed indi-rpicam driver few months ago, pull few days ago and rebuild. When i launch Ekos, i have a error message  "Unable to establish RPi Camera. Please ensure the device is connected and powered on". It looks like my camera is not detected. I'm sure it's enabled: when i run "raspistill -o test.jpg" i have an image. And camera enabled in rasp-config.

**To Reproduce**
- I launch kstars
- Go to "Toggle ekos"
- In my profile, set "RPI Camera" on CDD device
- Start ekos (ekos detect an other instance running...)
- Log can't establish RPI Camera

**Screenshots**
Some screenshots
[image]
[image]

**Desktop :**
 - Rasperry PI 3+ Model B
 - OS: Ubuntu 18-04 (not raspbian)
 - Version Kstars 3.5.2
 - RPI Camera V2 NoIR

Other question:
When i see updates on github, do i need to rebuild and compile ?
Thank you very much for your help.

**Log Files**
```
[2021-03-24T22:46:25.894 CET DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
[2021-03-24T22:46:25.902 CET WARN ][                       default] - QSqlDatabasePrivate::removeDatabase: connection 'filter_db' is still in use, all queries will cease to work.
[2021-03-24T22:46:25.903 CET WARN ][                       default] - QSqlDatabasePrivate::addDatabase: duplicate connection name 'filter_db', old connection removed.
[2021-03-24T22:46:25.998 CET INFO ][           org.kde.kstars.ekos] - "Starting INDI services..."
[2021-03-24T22:46:26.000 CET DEBG ][           org.kde.kstars.indi] - INDI: Starting local drivers...
[2021-03-24T22:46:26.001 CET DEBG ][           org.kde.kstars.indi] - Starting INDI Server:  ("-v", "-p", "7624", "-m", "300", "-r", "0", "-f", "/tmp/indififob475ca84") -f "/tmp/indififob475ca84"
[2021-03-24T22:46:26.035 CET DEBG ][           org.kde.kstars.indi] - INDI Server Started?  true
[2021-03-24T22:46:26.036 CET DEBG ][           org.kde.kstars.indi] - INDI: INDI Server started locally on port  7624
[2021-03-24T22:46:26.037 CET DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_rpicam"
[2021-03-24T22:46:26.038 CET DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_simulator_guide"
[2021-03-24T22:46:26.038 CET DEBG ][           org.kde.kstars.indi] - Adding managed driver "RPI Camera"
[2021-03-24T22:46:26.039 CET DEBG ][           org.kde.kstars.indi] - Adding managed driver "Guide Simulator"
[2021-03-24T22:46:26.039 CET DEBG ][           org.kde.kstars.indi] - INDIListener: Adding a new client manager to INDI listener..
[2021-03-24T22:46:26.040 CET DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2021-03-24T22:46:26.042 CET DEBG ][           org.kde.kstars.indi] - INDI server connected.
[2021-03-24T22:46:26.043 CET DEBG ][           org.kde.kstars.indi] - Connection to INDI server is successful
[2021-03-24T22:46:26.046 CET INFO ][           org.kde.kstars.ekos] - "INDI services started on port 7624."
[2021-03-24T22:46:26.073 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: startup: /usr/bin/indiserver -v -p 7624 -m 300 -r 0 -f /tmp/indififob475ca84 "
[2021-03-24T22:46:26.075 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: listening to port 7624 on fd 3"
[2021-03-24T22:46:26.076 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_rpicam -n \"RPI Camera\""
[2021-03-24T22:46:26.078 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "With name: RPI Camera"
[2021-03-24T22:46:26.079 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_rpicam"
[2021-03-24T22:46:26.081 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_rpicam: pid=3645 rfd=4 wfd=7 efd=8"
[2021-03-24T22:46:26.082 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_simulator_guide"
[2021-03-24T22:46:26.083 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_simulator_guide"
[2021-03-24T22:46:26.084 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: pid=3646 rfd=5 wfd=10 efd=11"
[2021-03-24T22:46:26.086 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Client 6: new arrival from 127.0.0.1:38656 - welcome!"
[2021-03-24T22:46:26.087 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2021-03-24T22:46:26.115 CET DEBG ][           org.kde.kstars.indi] - Received new device Guide Simulator
[2021-03-24T22:46:26.217 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_rpicam: MMALDriver()"
[2021-03-24T22:46:26.217 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD"
[2021-03-24T22:46:26.218 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on Telescope Simulator.TELESCOPE_INFO"
[2021-03-24T22:46:26.218 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on Telescope Simulator.GEOGRAPHIC_COORD"
[2021-03-24T22:46:26.219 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE"
[2021-03-24T22:46:26.219 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE"
[2021-03-24T22:46:26.219 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on Focuser Simulator.ABS_FOCUS_POSITION"
[2021-03-24T22:46:26.220 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on Focuser Simulator.FOCUS_TEMPERATURE"
[2021-03-24T22:46:26.220 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on CCD Simulator.FILTER_SLOT"
[2021-03-24T22:46:26.220 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on CCD Simulator.FILTER_NAME"
[2021-03-24T22:46:26.220 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on SQM.SKY_QUALITY"
[2021-03-24T22:46:26.221 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_simulator_guide: snooping on Focuser Simulator.FWHM"
[2021-03-24T22:46:26.221 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_rpicam: tvservice-client: Failed to connect to TV service: -1"
[2021-03-24T22:46:26.221 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:46:26: Driver indi_rpicam: MMALDriver() - returning"
[2021-03-24T22:46:26.221 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[... 98 lines cut ...]
[2021-03-24T22:55:07.739 CET WARN ][ default] - QSqlDatabasePrivate::addDatabase: duplicate connection name 'filter_db', old connection removed.
[2021-03-24T22:55:07.849 CET INFO ][org.kde.kstars.ekos] - "Services INDI arrêtés."
[2021-03-24T22:55:35.169 CET DEBG ][org.kde.kstars.indi] - INDI: Starting local drivers...
[2021-03-24T22:55:35.170 CET DEBG ][org.kde.kstars.indi] - Starting INDI Server:  ("-v", "-p", "7624", "-m", "300", "-r", "0", "-f", "/tmp/indififod915b5f0") -f "/tmp/indififod915b5f0"
[2021-03-24T22:55:35.201 CET DEBG ][org.kde.kstars.indi] - INDI Server Started?  true
[2021-03-24T22:55:35.201 CET DEBG ][org.kde.kstars.indi] - INDI: INDI Server started locally on port  7624
[2021-03-24T22:55:35.203 CET DEBG ][org.kde.kstars.indi] - Starting INDI Driver  "indi_rpicam"
[2021-03-24T22:55:35.205 CET DEBG ][org.kde.kstars.indi] - Adding managed driver "RPI Camera"
[2021-03-24T22:55:35.207 CET DEBG ][org.kde.kstars.indi] - INDIListener: Adding a new client manager to INDI listener..
[2021-03-24T22:55:35.208 CET DEBG ][org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2021-03-24T22:55:35.209 CET DEBG ][org.kde.kstars.indi] - INDI server connected.
[2021-03-24T22:55:35.213 CET DEBG ][org.kde.kstars.indi] - Connection to INDI server is successful
[... cut ...]
[2021-03-24T22:55:35.278 CET DEBG ][org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:55:35: Driver indi_rpicam: MMALDriver()"
[2021-03-24T22:55:35.279 CET DEBG ][org.kde.kstars.indi] - INDI Server:  ""
[2021-03-24T22:55:35.294 CET DEBG ][org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:55:35: Driver indi_rpicam: tvservice-client: Failed to connect to TV service: -1"
[2021-03-24T22:55:35.294 CET DEBG ][org.kde.kstars.indi] - INDI Server:  "2021-03-24T21:55:35: Driver indi_rpicam: MMALDriver() - returning"
```

knro, 2021-05-25:
@lboclboc Any idea what's behind this issue?

lboclboc, 2021-05-27:
Not really, if I understand the comment correctly you have recompiled the driver from source? Did you rebuild libindi also then? What commits of both repos did you use in that case.?

HamHamFonFon, 2021-06-09:
Sorry for very late answer. I've changed how to i want to run.
I have indi-server with indi_rpicam installed on a raspberry pi (server side). I'm running it with ssh from client-side : 
```
hamham@astroloutre:~$ indiserver -v indi_rpicam indi_v4l2_ccd
2021-06-09T20:17:01: startup: indiserver -v indi_rpicam indi_v4l2_ccd 
2021-06-09T20:17:01: Driver indi_rpicam: pid=4804 rfd=3 wfd=6 efd=7
2021-06-09T20:17:01: Driver indi_v4l2_ccd: pid=4805 rfd=4 wfd=9 efd=10
2021-06-09T20:17:01: listening to port 7624 on fd 5
2021-06-09T20:17:01: Driver indi_rpicam: MMALDriver()
2021-06-09T20:17:01: Driver indi_rpicam: MMALDriver() - returning
2021-06-09T20:17:01: Driver indi_rpicam: ISGetProperties()
2021-06-09T20:17:01: Driver indi_rpicam: getDefaultName()
2021-06-09T20:17:01: Driver indi_rpicam: initProperties()
2021-06-09T20:17:01: Driver indi_rpicam: getDefaultName()
2021-06-09T20:17:01: Driver indi_rpicam: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2021-06-09T20:17:01: Driver indi_rpicam: snooping on Telescope Simulator.TELESCOPE_INFO
2021-06-09T20:17:01: Driver indi_rpicam: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2021-06-09T20:17:01: Driver indi_rpicam: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2021-06-09T20:17:01: Driver indi_rpicam: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2021-06-09T20:17:01: Driver indi_rpicam: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2021-06-09T20:17:01: Driver indi_rpicam: snooping on Focuser Simulator.FOCUS_TEMPERATURE
2021-06-09T20:17:01: Driver indi_rpicam: snooping on CCD Simulator.FILTER_SLOT
2021-06-09T20:17:01: Driver indi_rpicam: snooping on CCD Simulator.FILTER_NAME
2021-06-09T20:17:01: Driver indi_rpicam: snooping on SQM.SKY_QUALITY
```

Server IP is 10.42.0.1. 
Client IP is 10.42.0.233

Client-side, on a laptop, i have kstars/ekos installed and here i'm a bit confused about how to proceed. When i'm running Kstars, in "Tools > device > device manager > client", i have added my server with a name, adress IP on host and port.
First question, in "Tools > device > Custom driver", there is not "indi_rpicam" Do i need to add it ?

Then, in Ekos tab, i have a profile, in remote mode, connected to my raspberry pi. In CCD, i can't add "indi_rpicam", maybe due to question above ? I'm clicking on start button.
Server side, there is this : 
```
2021-06-09T20:23:47: Client 0: new arrival from 10.42.0.233:40782 - welcome!
2021-06-09T20:23:47: Driver indi_rpicam: CameraControl(/home/hamham/Projects/indi-3rdparty/indi-rpicam/cameracontrol.cpp:34): enter
2021-06-09T20:23:47: Driver indi_rpicam: mmal: mmal_component_create_core: could not find component 'vc.ril.camera'
2021-06-09T20:23:47: Driver indi_rpicam: terminate called after throwing an instance of 'MMALException'
2021-06-09T20:23:47: Driver indi_rpicam:   what():  
2021-06-09T20:23:47: Driver indi_rpicam: stderr EOF
<delProperty device="RPI Camera"/>
2021-06-09T20:23:47: Driver indi_rpicam: restart #1
Child process 4804 died
2021-06-09T20:23:47: Driver indi_rpicam: pid=4818 rfd=3 wfd=8 efd=11
2021-06-09T20:23:47: Driver indi_rpicam: MMALDriver()
2021-06-09T20:23:47: Driver indi_rpicam: MMALDriver() - returning
2021-06-09T20:23:47: Driver indi_rpicam: ISGetProperties()
2021-06-09T20:23:47: Driver indi_rpicam: getDefaultName()
2021-06-09T20:23:47: Driver indi_rpicam: initProperties()
2021-06-09T20:23:47: Driver indi_rpicam: getDefaultName()
2021-06-09T20:23:47: Driver indi_rpicam: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2021-06-09T20:23:47: Driver indi_rpicam: snooping on Telescope Simulator.TELESCOPE_INFO
2021-06-09T20:23:47: Driver indi_rpicam: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2021-06-09T20:23:47: Driver indi_rpicam: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2021-06-09T20:23:47: Driver indi_rpicam: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2021-06-09T20:23:47: Driver indi_rpicam: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2021-06-09T20:23:47: Driver indi_rpicam: snooping on Focuser Simulator.FOCUS_TEMPERATURE
2021-06-09T20:23:47: Driver indi_rpicam: snooping on CCD Simulator.FILTER_SLOT
2021-06-09T20:23:47: Driver indi_rpicam: snooping on CCD Simulator.FILTER_NAME
2021-06-09T20:23:47: Driver indi_rpicam: snooping on SQM.SKY_QUALITY
```
Ekos logs have : 
```
2021-06-09T22:24:02 Unable to establish remote devices:
+ INDI Webcam
+ V4L2 CCD
Please ensure remote device name corresponds to actual device name.
2021-06-09T22:23:47 RPI Camera is offline.
2021-06-09T22:23:47 INDI services started. Connection to remote INDI server is successful. Waiting for devices...
2021-06-09T22:23:47 Connecting to remote INDI server at 10.42.0.1 on port 7 624 ...
2021-06-09T22:23:47 Site location updated to Montpellier, Herault, France.
```
INDI Control Panel window is all grey, without interface...and i can't click on button "Connect".
[image]

I don't know if the problem is from indi_rpicam (and this issue is useful) or in Kstars/Ekos (and i think you are not concerned).

lboclboc, 2021-06-13:
It seems the driver does not find the camera. I have seen this my self when I tried to make crosscompilation work. Did you build the driver your self or is it from the distribution? I think the problem I had was that not all needed libraries was included so that static constructors was missing.

HamHamFonFon, 2021-06-14:
I have build it. In my first messafe, I indicated i'm on Ubuntu, not Raspbian :).
So, did i have to rebuild indi_rpicam ? If yes, could you tell me all files i need to remove ? Or there is an aother way ?

HamHamFonFon, 2021-06-15:
Ok. So, what i am doing ?

lboclboc, 2021-06-15:
If you tell me the exact commands all the way how you cloned and built the driver I can see if I can reproduce. Also I need to know what exact dist and version are running on your rpi, and model of rpi.

lboclboc, 2021-06-17:
Thanks, I'll try that out.

lboclboc, 2021-06-26:
Sorry, its takes long time for me. Ive just got around to install ubuntu-20.04, i could not find a raspi ubuntu 18.04 anywhere.
What packages to you install to get the rapistill  and the userland libraries to work, i have not used plain ubuntu on raspberry before..

HamHamFonFon, 2021-06-28:
Hello, don't worry, it's not a problem if you don't have time: with my one year old baby, i don't have time too ^^.
I've upgraded to ubuntu 20.04 few weeks ago (after my first comment), i've forget to tell it, sorry.
For raspistill, as i remember :
```
sudo bash -c "echo 'start_x=1' >> /boot/config.txt"
sudo bash -c "echo 'gpu_mem=128' >> /boot/config.txt"

sudo rpi-update
sudo raspi-config
#Select 3 Interface Options then P1 to enable the camera
shutdown -r now
```

lboclboc, 2021-07-13:
Sorry, I have tried to make the raspistill work on my pure ubuntu-20.04 work on rpi but have not succeded yet. I get compilation errors for the userland. Ive been following several guides but have not succeded so far.

SebNik1, 2021-08-22:
Hello everyone, today I tried getting my rpi hq cam to run on Ubuntu 20.04.3 arm64 on my rpi4 8gb. It seems like the indi-rpicam driver is no longer included in the full indi 3rd party driver package anymore (I checked both stable and nightly repository directories too). I could not run it via terminal using indiserver indi_rpicam nor indi-rpicam. 

It is also not possible to manually install it via apt-get indi-rpicam.

When I tried to building from source I failed here:

xx@xx ~/Projects/build/indi-rpicam$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-rpicam
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Found INDI: /usr/lib/aarch64-linux-gnu/libindidriver.so (found version "1.9.0") 
-- Found CFITSIO 3.47: /usr/lib/aarch64-linux-gnu/libcfitsio.so
-- Found NOVA: /usr/lib/aarch64-linux-gnu/libnova.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found USB1: /usr/lib/aarch64-linux-gnu/libusb-1.0.so (found version "1.0.23") 
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Found FFTW3: /usr/lib/aarch64-linux-gnu/libfftw3.so
-- Found FFTW3: /usr/lib/aarch64-linux-gnu/libfftw3.so
-- Looking for pthread.h
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/CheckIncludeFile.cmake:80 (message):
  Policy CMP0075 is not set: Include file check macros honor
  CMAKE_REQUIRED_LIBRARIES.  Run "cmake --help-policy CMP0075" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  CMAKE_REQUIRED_LIBRARIES is set to:

    /usr/lib/aarch64-linux-gnu/libnova.so

  For compatibility with CMake 3.11 and below this check is ignoring it.
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindThreads.cmake:128 (CHECK_INCLUDE_FILE)
  CMakeLists.txt:34 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found GTest: /usr/lib/aarch64-linux-gnu/libgtest.a  
-- Found GMock: /usr/lib/aarch64-linux-gnu/libgmock.a  
-- Building unit tests
-- GTEST_BOTH_LIBRARIES /usr/lib/aarch64-linux-gnu/libgtest.a;/usr/lib/aarch64-linux-gnu/libgtest_main.a
-- GTEST_MAIN_LIBRARIES /usr/lib/aarch64-linux-gnu/libgtest_main.a
-- GTEST_LIBRARIES /usr/lib/aarch64-linux-gnu/libgtest.a
-- GTEST_INCLUDE_DIRS /usr/include
-- GMOCK_LIBRARIES /usr/lib/aarch64-linux-gnu/libgmock.a
-- GMOCK_INCLUDE_DIRS /usr/include
-- Could not find hardware support for NEON on this machine.
-- No OMAP3 processor on this machine.
-- No OMAP4 processor on this machine.
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
brcmEGL_LIBRARY
    linked by target "indi_rpicam" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam
    linked by target "test_imx219" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam/test
    linked by target "test_imx477" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam/test
brcmGLESv2_LIBRARY
    linked by target "indi_rpicam" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam
    linked by target "test_imx219" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam/test
    linked by target "test_imx477" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam/test

-- Configuring incomplete, errors occurred!
See also "/home/sn/Projects/build/indi-rpicam/CMakeFiles/CMakeOutput.log".
See also "/home/sn/Projects/build/indi-rpicam/CMakeFiles/CMakeError.log".

I'd be happy to join forces here to get the rpicam to work. Please let me know what additional information would help. 

Thank you very much in advance!
Sebastian

HamHamFonFon, 2021-08-24:
I have trying to use indi_rpicam with only indi as server and [Astrophotoplus](https://github.com/GuLinux/AstroPhoto-Plus) as client and camera not detected (USB camera not detected too).
I will retry a full installation of indi and 3rd drivers likes indi_rpicam because i don't think the error come from driver. Response in few days (or weeks...)

HamHamFonFon, 2021-08-28:
Yes, the package in repository doesn't exist anymore. That's wht indi_rpicam can only be installed manually. Here's my yesterday installation : 
```
# Install indi
sudo apt-get install indi-full gsc
cd ~/Projects

# install libraries
sudo apt-get -y install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libtiff-dev libfftw3-dev libftdi-dev libgps-dev libraw-dev libdc1394-22-dev libgphoto2-dev libboost-dev libboost-regex-dev librtlsdr-dev liblimesuite-dev libftdi1-dev libavcodec-dev libavdevice-dev

# Clone repository
git clone --depth=1 https://github.com/indilib/indi-3rdparty
mkdir -p ~/Projects/build/indi-rpicam
cd ~/Projects/build/indi-rpicam
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-rpicam

# Got an error, so install forgotten package
sudo apt-get -y install libindi-dev
cd ..

# Delete previous installation
rm -rf *
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-rpicam
make -j4
sudo make install

# Check
ls -al /usr/bin/ | grep rpicam
-rwxr-xr-x  1 root root      960288 août  27 14:38 indi_rpicam*
```
It seems you got a problem with C++ package "libnova". Try : 
```
sudo apt-get install libnova-dev
```

SebNik1, 2021-08-29:
Thank you very much for sharing your steps! I tried reproduce everything step by step but it seems my error is not due to missing packages or libraries. Everything was already installed and up to date.

cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-rpicam
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Found INDI: /usr/lib/aarch64-linux-gnu/libindidriver.so (found version "1.9.1") 
-- Found CFITSIO 3.47: /usr/lib/aarch64-linux-gnu/libcfitsio.so
-- Found NOVA: /usr/lib/aarch64-linux-gnu/libnova.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found USB1: /usr/lib/aarch64-linux-gnu/libusb-1.0.so (found version "1.0.23") 
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Found FFTW3: /usr/lib/aarch64-linux-gnu/libfftw3.so
-- Found FFTW3: /usr/lib/aarch64-linux-gnu/libfftw3.so
-- Looking for pthread.h
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/CheckIncludeFile.cmake:80 (message):
  Policy CMP0075 is not set: Include file check macros honor
  CMAKE_REQUIRED_LIBRARIES.  Run "cmake --help-policy CMP0075" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  CMAKE_REQUIRED_LIBRARIES is set to:

    /usr/lib/aarch64-linux-gnu/libnova.so

  For compatibility with CMake 3.11 and below this check is ignoring it.
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindThreads.cmake:128 (CHECK_INCLUDE_FILE)
  CMakeLists.txt:34 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found GTest: /usr/lib/aarch64-linux-gnu/libgtest.a  
-- Found GMock: /usr/lib/aarch64-linux-gnu/libgmock.a  
-- Building unit tests
-- GTEST_BOTH_LIBRARIES /usr/lib/aarch64-linux-gnu/libgtest.a;/usr/lib/aarch64-linux-gnu/libgtest_main.a
-- GTEST_MAIN_LIBRARIES /usr/lib/aarch64-linux-gnu/libgtest_main.a
-- GTEST_LIBRARIES /usr/lib/aarch64-linux-gnu/libgtest.a
-- GTEST_INCLUDE_DIRS /usr/include
-- GMOCK_LIBRARIES /usr/lib/aarch64-linux-gnu/libgmock.a
-- GMOCK_INCLUDE_DIRS /usr/include
-- Could not find hardware support for NEON on this machine.
-- No OMAP3 processor on this machine.
-- No OMAP4 processor on this machine.
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
brcmEGL_LIBRARY
    linked by target "indi_rpicam" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam
    linked by target "test_imx219" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam/test
    linked by target "test_imx477" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam/test
brcmGLESv2_LIBRARY
    linked by target "indi_rpicam" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam
    linked by target "test_imx219" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam/test
    linked by target "test_imx477" in directory /home/sn/Projects/indi-3rdparty/indi-rpicam/test

-- Configuring incomplete, errors occurred!
See also "/home/sn/Projects/build/indi-rpicam/CMakeFiles/CMakeOutput.log".
See also "/home/sn/Projects/build/indi-rpicam/CMakeFiles/CMakeError.log".

and regarding libnova-dev:
$ sudo apt-get install libnova-dev
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
libnova-dev ist schon die neueste Version (0.16-4).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.

So I still cannot compile...

knro, 2021-09-28:
The issue is not libnova (that was just warning), it's with brcmEGL_LIBRARY

It appears these might be only available on Raspbian OS?

closed 2021-11-24 by knro

simonachmueller, 2022-06-22:
Hi, I have same issue compiling latest indi_rpicam:
```
pi@raspberrypi:~/src/indi-3rdparty/build/indi-rpicam $ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ../../indi-rpicam/
-- Found CFITSIO: /usr/lib/arm-linux-gnueabihf/libcfitsio.so
-- Found libnova: /usr/lib/arm-linux-gnueabihf/libnova.so
-- Found FFTW3: /usr/lib/arm-linux-gnueabihf/libfftw3.so
-- Found FFTW3: /usr/lib/arm-linux-gnueabihf/libfftw3.so
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY)
-- Could NOT find GMock (missing: GMOCK_LIBRARY GMOCK_INCLUDE_DIR)
-- GTEST not found, not building unit tests
-- No OMAP3 processor on this machine.
-- No OMAP4 processor on this machine.
-- Neon found with compiler flag : -mfpu=neon -D__NEON__
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
brcmEGL_LIBRARY
    linked by target "indi_rpicam" in directory /home/pi/src/indi-3rdparty/indi-rpicam
brcmGLESv2_LIBRARY
    linked by target "indi_rpicam" in directory /home/pi/src/indi-3rdparty/indi-rpicam

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```
also
```
pi@raspberrypi:~/src/indi-3rdparty/build/indi-rpicam $  ldconfig -p | grep libGLESv2
        libGLESv2.so.2 (libc6,hard-float) => /lib/arm-linux-gnueabihf/libGLESv2.so.2
        libGLESv2.so (libc6,hard-float) => /lib/arm-linux-gnueabihf/libGLESv2.so
```

What do I miss to (pre-)install?

knro, 2022-06-22:
bullseye not supported, they changed the API to libcamera. We need to migrate to libcamera.

simonachmueller, 2022-06-22:
Here is my CMake output
```
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ../../indi-rpicam/
-- The CXX compiler identification is GNU 10.2.1
-- The C compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Found INDI: /usr/lib/arm-linux-gnueabihf/libindidriver.so (found version "1.9.7")
-- Found CFITSIO 3.49: /usr/lib/arm-linux-gnueabihf/libcfitsio.so
-- Found NOVA: /usr/lib/arm-linux-gnueabihf/libnova.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2")
-- Found USB1: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so (found version "1.0.24")
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Found FFTW3: /usr/lib/arm-linux-gnueabihf/libfftw3.so
-- Found FFTW3: /usr/lib/arm-linux-gnueabihf/libfftw3.so
-- Looking for pthread.h
CMake Warning (dev) at /usr/share/cmake-3.18/Modules/CheckIncludeFile.cmake:80 (message):
  Policy CMP0075 is not set: Include file check macros honor
  CMAKE_REQUIRED_LIBRARIES.  Run "cmake --help-policy CMP0075" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  CMAKE_REQUIRED_LIBRARIES is set to:

    /usr/lib/arm-linux-gnueabihf/libnova.so

  For compatibility with CMake 3.11 and below this check is ignoring it.
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindThreads.cmake:142 (CHECK_INCLUDE_FILE)
  CMakeLists.txt:34 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
-- Found GTest: /usr/lib/arm-linux-gnueabihf/libgtest.a
-- Found GMock: /usr/lib/arm-linux-gnueabihf/libgmock.a
-- Building unit tests
-- GTEST_BOTH_LIBRARIES /usr/lib/arm-linux-gnueabihf/libgtest.a;/usr/lib/arm-linux-gnueabihf/libgtest_main.a
-- GTEST_MAIN_LIBRARIES /usr/lib/arm-linux-gnueabihf/libgtest_main.a
-- GTEST_LIBRARIES /usr/lib/arm-linux-gnueabihf/libgtest.a
-- GTEST_INCLUDE_DIRS /usr/include
-- GMOCK_LIBRARIES /usr/lib/arm-linux-gnueabihf/libgmock.a
-- GMOCK_INCLUDE_DIRS /usr/include
-- No OMAP3 processor on this machine.
-- No OMAP4 processor on this machine.
-- Neon found with compiler flag : -mfpu=neon -D__NEON__
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
brcmEGL_LIBRARY
    linked by target "indi_rpicam" in directory /home/pi/src/indi-3rdparty/indi-rpicam
    linked by target "test_imx219" in directory /home/pi/src/indi-3rdparty/indi-rpicam/test
    linked by target "test_imx477" in directory /home/pi/src/indi-3rdparty/indi-rpicam/test
brcmGLESv2_LIBRARY
    linked by target "indi_rpicam" in directory /home/pi/src/indi-3rdparty/indi-rpicam
    linked by target "test_imx219" in directory /home/pi/src/indi-3rdparty/indi-rpicam/test
    linked by target "test_imx477" in directory /home/pi/src/indi-3rdparty/indi-rpicam/test

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```