indilib/indi-3rdparty #368: rpi-cam not possible to make on astroberry

vopalem, 2021-03-26:
Hello, 
I am not able to compile the indi-rpicam driver on Astroberry running on raspi 4B. It is not ubuntu, so I could not proceed as described here https://indilib.org/raspberry-pi/raspberry-pi-camera.html 
Instead, I followed these instructions https://github.com/indilib/indi-3rdparty , making the clone as 

git clone --depth=1 https://github.com/indilib/indi-3rdparty

I am able to perform cmake:

astroberry@astroberry:~/Projects/build/indi-rpicam $ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debut ~/Projects/indi-3rdparty/indi-rpicam
-- Found CFITSIO: /usr/lib/arm-linux-gnueabihf/libcfitsio.so
-- Found libnova: /usr/lib/arm-linux-gnueabihf/libnova.so
-- Found FFTW3: /usr/lib/arm-linux-gnueabihf/libfftw3.so
-- Found FFTW3: /usr/lib/arm-linux-gnueabihf/libfftw3.so
-- Building unit tests
-- GTEST_BOTH_LIBRARIES /usr/lib/arm-linux-gnueabihf/libgtest.a;/usr/lib/arm-linux-gnueabihf/libgtest_main.a
-- GTEST_MAIN_LIBRARIES /usr/lib/arm-linux-gnueabihf/libgtest_main.a
-- GTEST_LIBRARIES /usr/lib/arm-linux-gnueabihf/libgtest.a
-- GTEST_INCLUDE_DIRS /usr/include
-- GMOCK_LIBRARIES /usr/lib/arm-linux-gnueabihf/libgmock.a
-- GMOCK_INCLUDE_DIRS /usr/include
-- No OMAP3 processor on this machine.
-- No OMAP4 processor on this machine.
-- Neon found with compiler flag : -mfpu=neon -D__NEON__
-- Configuring done
-- Generating done
-- Build files have been written to: /home/astroberry/Projects/build/indi-rpicam

But after $make -j4, I get errors in the code of mmaldriver.cpp:

astroberry@astroberry:~/Projects/build/indi-rpicam $ make -j4
[  6%] Building CXX object CMakeFiles/rpicam.dir/broadcompipeline.cpp.o
[  6%] Building CXX object CMakeFiles/rpicam.dir/mmaldriver.cpp.o
[  6%] Building CXX object CMakeFiles/rpicam.dir/jpegpipeline.cpp.o
[  6%] Building CXX object CMakeFiles/rpicam.dir/pipetee.cpp.o
/home/astroberry/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp: In member function ‘virtual bool MMALDriver::Connect()’:
/home/astroberry/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp:139:14: error: ‘getCurrentPollingPeriod’ was not declared in this scope
     SetTimer(getCurrentPollingPeriod());
              ^~~~~~~~~~~~~~~~~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp:139:14: note: suggested alternative: ‘setDefaultPollingPeriod’
     SetTimer(getCurrentPollingPeriod());
              ^~~~~~~~~~~~~~~~~~~~~~~
              setDefaultPollingPeriod
/home/astroberry/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp: In member function ‘virtual bool MMALDriver::updateProperties()’:
/home/astroberry/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp:276:9: error: ‘defineProperty’ was not declared in this scope
         defineProperty(&mGainNP);
         ^~~~~~~~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp:276:9: note: suggested alternative: ‘deleteProperty’
         defineProperty(&mGainNP);
         ^~~~~~~~~~~~~~
         deleteProperty
/home/astroberry/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp: In member function ‘virtual void MMALDriver::TimerHit()’:
/home/astroberry/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp:442:26: error: ‘getCurrentPollingPeriod’ was not declared in this scope
     uint32_t nextTimer = getCurrentPollingPeriod();
                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp:442:26: note: suggested alternative: ‘setDefaultPollingPeriod’
     uint32_t nextTimer = getCurrentPollingPeriod();
                          ^~~~~~~~~~~~~~~~~~~~~~~
                          setDefaultPollingPeriod
make[2]: *** [CMakeFiles/rpicam.dir/build.make:76: CMakeFiles/rpicam.dir/mmaldriver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/rpicam.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

It is possible to connect the camera with the V4L2 driver, but with limited resolution. 

Can anyone help me, please? I have to remark that I am totally not a Linux guy.

Thank you very much.

knro, 2021-03-26:
Compile and install indi first.

vopalem, 2021-03-26:
@knro:
Ah, I thought indi would be preinstalled properly on astroberry. Anyways, to reinstall it, is the procedure below correct?

---sorry for my dummy questions---

sudo apt-get update
sudo apt-get install libnova-dev libcfitsio3-dev libusb-1.0-0-dev zlib1g-dev libgsl0-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev
mkdir ~/Projects 
cd ~/Projects 
git clone https://github.com/indilib/indi.git 
mkdir -p build/libindi 
cd build/libindi 
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi/libindi 
sudo make install

closed 2021-04-02 by knro