indilib/indi-3rdparty #373: Issues - ZWO Cameras

pawel-soja, 2021-04-01:
Hi!

I created this topic to summarize and confirm the problems.  I try to fix issues with ZWO cameras.
- No possibility to run more than one camera.
By analyzing code sources (reverse) ASICamera2, I came across a static variable that can do a conflict with several cameras (fixed in ASICameraBoost)
- Long exposure time - I got information from ZWO that this was fixed, please confirm.
- Bandwidth 100% / HighSpeedMode 1 - on Raspberrypi 4 - small number of frames. The problem is the implementation of the ASICamera2 library and slow RAM.

Please confirm and report problems using the latest INDI / INDI-3rd Party compilation.

pawel-soja, 2021-04-01:
I also invite you to test ASICameraBoost.
I updated the swap method - I hope it's simple.
[image]
https://github.com/pawel-soja/AsiCamera

knro, 2021-04-02:
Thanks! I will check it out. Btw, how come you can't run more than one camera? the driver been running multiple cameras for years now and I can confirm this along with many other users. 

There is a limitation in INDI server since it is single threaded. Once it receives a big blob from the main camera, then it needs to finish sending that first to the client before any other blobs (e.g. guide camera) can make it out.

knro, 2021-04-02:
Just tested with this:

`LD_PRELOAD=~/Projects/build/AsiCamera/lib/libASICamera2Boost.so indiserver -v indi_asi_ccd indi_simulator_telescope
`

On first capture it crashes:

```
2021-04-02T07:46:04: Client 0: new arrival from 192.168.1.10:53910 - welcome!
2021-04-02T07:46:05: Client 8: new arrival from 192.168.1.10:53912 - welcome!
2021-04-02T07:46:06: Client 11: new arrival from 192.168.1.10:53914 - welcome!
2021-04-02T07:46:12: Driver indi_asi_ccd: stderr EOF
<delProperty device="ZWO CCD ASI120MC"/>
Child process 5803 died
<delProperty device="ZWO CCD ASI178MC-Cool"/>
2021-04-02T07:46:12: Driver indi_asi_ccd: restart #1
2021-04-02T07:46:12: Driver indi_asi_ccd: pid=5866 rfd=3 wfd=12 efd=13
```

Does it need to be linked in a different way?

pawel-soja, 2021-04-02:
Linking OK, it couldn't be simpler :) I fixed crash issue, please confirm.

I never had 2 cameras working at once. I thought this was a known issue.
I found something like this in the libASICamera2 code:
```cpp
#include "CCameraFX3.h"
#include <algorithm>

S_MUTEX_UNIX mtx_usb;
libusb_context *g_usb_context = NULL;

/* BEGIN disaster */
static int len_get = 0;
static int lin_XferLen = 0;
static bool lin_XferCallbacked = false;
static int XferErr;
/* END disaster */

bool bMtxInit = false;

void callbackUSBTransferComplete(libusb_transfer *transfer)
{
  XferErr = transfer->status;
  if (XferErr == 0)
  {
    lin_XferLen = transfer->actual_length;
    len_get += lin_XferLen;
    lin_XferCallbacked = 1;
  }

  if (XferErr > -1 && XferErr < 7)
  {
    lin_XferLen = -1;
    lin_XferCallbacked = 1;
  }
}

void async_xfer_cb(libusb_transfer *transfer)
{
  *(int *)transfer->user_data = 1;
}
```
Comparison of ASICamera2 and ASICamera2Boost libraries.
https://youtu.be/X8sXbmgUNII

knro, 2021-04-02:
Impressive video! Unfortunately, it still crashes here :(

```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
2021-04-02T09:36:17: startup: /usr/bin/indiserver -r 0 -v indi_asi_ccd indi_simulator_telescope 
[Attaching after Thread 0xb6ff7ef0 (LWP 17401) fork to child process 17404]
[New inferior 2 (process 17404)]
[Detaching after fork from parent process 17401]
2021-04-02T09:36:17: Driver indi_asi_ccd: pid=17404 rfd=3 wfd=6 efd=7
2021-04-02T09:36:17: Driver indi_simulator_telescope: pid=17405 rfd=4 wfd=9 efd=10
2021-04-02T09:36:17: listening to port 7624 on fd 5
[Inferior 1 (process 17401) detached]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
process 17404 is executing new program: /usr/bin/indi_asi_ccd
2021-04-02T09:36:17: Driver indi_simulator_telescope: HaAxis: TrackRate 1, trackingRateDegSec 15.041067 arcsec
2021-04-02T09:36:17: Driver indi_simulator_telescope: snooping on GPS Simulator.GEOGRAPHIC_COORD
2021-04-02T09:36:17: Driver indi_simulator_telescope: snooping on GPS Simulator.TIME_UTC
2021-04-02T09:36:17: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_PARK
2021-04-02T09:36:17: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_SHUTTER
2021-04-02T09:36:18: Client 0: new arrival from 192.168.1.10:55972 - welcome!
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb4f97220 (LWP 17406)]
[Thread 0xb4f97220 (LWP 17406) exited]
[New Thread 0xb4f97220 (LWP 17407)]
[Thread 0xb4f97220 (LWP 17407) exited]
[New Thread 0xb4f97220 (LWP 17408)]
[Thread 0xb4f97220 (LWP 17408) exited]
[New Thread 0xb4f97220 (LWP 17409)]
[New Thread 0xb4796220 (LWP 17410)]
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on Telescope Simulator.TELESCOPE_INFO
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on Focuser Simulator.FOCUS_TEMPERATURE
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on CCD Simulator.FILTER_SLOT
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on CCD Simulator.FILTER_NAME
2021-04-02T09:36:21: Driver indi_asi_ccd: snooping on SQM.SKY_QUALITY
[New Thread 0xb3f95220 (LWP 17411)]
[New Thread 0xb35ff220 (LWP 17412)]
[New Thread 0xb2dfe220 (LWP 17413)]
[New Thread 0xb23ff220 (LWP 17414)]
[New Thread 0xb1bfe220 (LWP 17415)]
[New Thread 0xb0cf1220 (LWP 17416)]
2021-04-02T09:36:22: Client 8: new arrival from 192.168.1.10:55976 - welcome!
[New Thread 0xad7dc220 (LWP 17418)]
[New Thread 0xaab00220 (LWP 17419)]
2021-04-02T09:36:23: Client 11: new arrival from 192.168.1.10:55978 - welcome!
[New Thread 0xa92ff220 (LWP 17421)]

Thread 2.15 "indi_asi_ccd" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xa92ff220 (LWP 17421)]
0xb6e2a1d4 in CameraBoost::releaseAsyncXfer() () from /home/stellarmate/Projects/build/AsiCamera/lib/libASICamera2Boost.so
#0  0xb6e2a1d4 in CameraBoost::releaseAsyncXfer() () from /home/stellarmate/Projects/build/AsiCamera/lib/libASICamera2Boost.so
#1  0xb6ea57a8 in WorkingFunc () from /home/stellarmate/Projects/build/AsiCamera/lib/libASICamera2Boost.so
#2  0xb6f3865c in MyThr(void*) () from /home/stellarmate/Projects/build/AsiCamera/lib/libASICamera2Boost.so
#3  0xb6264494 in start_thread (arg=0xa92ff220) at pthread_create.c:486
#4  0xb61e7578 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/arm-linux-gnueabihf/libc.so.6
2021-04-02T09:36:29: Driver indi_asi_ccd: stderr EOF
<delProperty device="ZWO CCD ASI120MC"/>
<delProperty device="ZWO CCD ASI178MC-Cool"/>
2021-04-02T09:36:29: Driver indi_asi_ccd: Terminated after #0 restarts.
Child process 17404 died
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

```

pawel-soja, 2021-04-02:
Thanks a lot for the tests. I need to add a few checks.
Probably a wrong configuration popped up. Is it RPi4?
Can you show me the screen of the "Controls" tab?

knro, 2021-04-02:
Yes rpi4, check screenshot.

[image]

knro, 2021-04-02:
I ran another test with debug enabled, here is the log file.

[crash_20210402_125902.txt](https://github.com/indilib/indi-3rdparty/files/6248983/crash_20210402_125902.txt)

pawel-soja, 2021-04-02:
Streaming is crashing too?

knro, 2021-04-02:
Progress! Now I can capture with the 178, but the 120 always fails. When trying with the regular driver, 120 captures fine. Log attached.
[log.txt](https://github.com/indilib/indi-3rdparty/files/6250062/log.txt)

pawel-soja, 2021-04-02:
Please try to run the library from branch:
https://github.com/pawel-soja/AsiCamera/tree/fix-old-camera
There are unnecessarily different implementations of the same for different cameras in the original library.
My ASI120MM Mini camera belongs to the same implementation as ASI178. Your implementation of the ASI120MC camera is strange (I mean an implementation in original ASICamera2 library).
Remember to turn off "AutoExpGain" and "AutoWhiBal".

Please also confirm that you can start streaming from 2 cameras at the same time on the original library - As I mentioned, it never worked for me and that's not a problem at INDI.

pawel-soja, 2021-04-05:
@knro, Can you check?

knro, 2021-04-06:
@pawel-soja Sure just did. Now both cameras don't capture. I'm attaching the logs. Btw, there was a SIGPIPE crash in KStars when I terminated remote INDI server and then reconnected. It appears that BaseClient still trying to send to the old FD even in the case of abnormal disconnection which is probably not handled well now. Can you see where this could be?

[kstars_sigpipe.txt](https://github.com/indilib/indi-3rdparty/files/6263796/kstars_sigpipe.txt)
[asilog.txt](https://github.com/indilib/indi-3rdparty/files/6263797/asilog.txt)

pawel-soja, 2021-04-06:
@knro Sure, I'll take a look. In the meantime, try this:
https://github.com/pawel-soja/AsiCamera/tree/new-way

knro, 2021-04-07:
Just tested, same issue. Log attached.
[newway.txt](https://github.com/indilib/indi-3rdparty/files/6270733/newway.txt)

pawel-soja, 2021-04-07:
I think when this is solved, most of the cameras will work. It would be easiest for me to buy this camera...
Thanks for your patience and testing. Now I can carefully analyze the overused amount of buffers in the original library.

pawel-soja, 2021-04-26:
@knro Please confirm, is this the camera? https://astronomy-imaging-camera.com/product/asi120mc-s

pawel-soja, 2021-04-27:
I have already ordered this camera. I hope it's this one.

knro, 2021-04-27:
Yes it's this camera.. though I'm using the older USB 2.0 version.

pawel-soja, 2021-04-27:
This https://astronomy-imaging-camera.com/product/asi120mc-color ?
That's why the codes for it are so strange.
Well, I will make monitoring on the balcony ;)
I have to look for someone to sell.

closed 2021-09-28 by knro