indilib/indi-3rdparty #379: ASI ccd driver compiling error

ZhangStoneRoy, 2021-04-21:
**Describe the bug**
Error message when running 'make install':
fatal error: indipropertyswitch.h: No such file or directory
 #include "indipropertyswitch.h"
          ^~~~~~~~~~~~~~~~~~~~~~

**To Reproduce**
1. git clone and follow the README file
2. sudo make install
3. getting the error message

**Desktop (please complete the following information):**
 - OS: Raspberry Pi OS
 - Version: latest

knro, 2021-04-21:
Please install libindi first.

ZhangStoneRoy, 2021-04-21:
> Please install libindi first.

I did built and installed libasi from source first and 'whereis' command returns the correct path, so I assume that's not what caused the problem?

ZhangStoneRoy, 2021-04-21:
> Please install libindi first.

I have indi-full installed from the stellarmate repo, is that what you meant by libindi? I will try to install it from source and see if the error still occurs.

knro, 2021-04-21:
From source it should work OK until the repo is updated.

ZhangStoneRoy, 2021-04-21:
> From source it should work OK until the repo is updated.

I installed it from source and now it works good. Thanks!

closed 2021-04-21 by ZhangStoneRoy