indilib/indi-3rdparty #380: libpigpiod consumes too much CPU

knro, 2021-04-24:
Just noticed that this pigpiod takes about 7% to 10% on raspberry pi CPU while not used. @ken-self Can you please check what's going on? Since this causes the performance of the PI to degrade substantially.

ken-self, 2021-04-25:
I saw this early on and looked it up. According to the author joan2937 this is only on one of the 4 CPUs so actually 2% overall.
https://github.com/joan2937/pigpio/issues/29
It is possible to reduce the sample rate with switch "-s 10" on the daemon and/or with switch "-m".
I'll give those a try to see if they help

ken-self, 2021-04-25:
The -m option reduces the idle CPU to zero BUT stops the Timed GPIO port option in the driver from working. I assume the sampling is needed by the watchdog timer. This driver function is primarily for triggering DSLR bulb exposures. So we could default to -m and advise DSLR users to remove the switch in the service. This should only affect those DSLR users not using libgphoto. It would also affect users using the timer function for other purposes like flashing LEDs

knro, 2021-04-25:
Thanks, but even this small CPU percentage was causing the GUI to be a bit unresponsive in Raspberry PI. I suppose it's not possible to programatically trigger this on/off after starting the server somehow? INDI uses FIFO for this purpose for example.

ken-self, 2021-04-26:
No, it cannot be switched programatically. I'll look around for an alternative timing mechanism. I just used the built-in pigpiod hardware timer because it was convenient. But if I can find an alternativethat should work just as well and allow the -m switch to be used. Worst case I might need to restrict the timing function to just one pin.
Do you have any preferred timer? Ideally something that triggers a callback when the timer expires.  I've seen some examples that use boost::asio and boost::chrono that might do the job

knro, 2021-04-26:
Recently @pawel-soja added timers to INDI, but not sure if they're suitable for this kind of work?

pawel-soja, 2021-04-26:
That's right, but I wrote timers based on those already existing in INDI Core. The class provides an easy-to-use interface.
In the future, it will also be a precise timer. Current disadvantages:
- works on the main thread (no change possible)
- event loop is not woken up when adding a new timer - this causes a slight delay in the first callback call.

https://github.com/indilib/indi/blob/master/libs/indibase/timer/inditimer.h
Example:
```cpp
#include "inditimer.h"
// ... //
INDI::Timer timer;
timer.callOnTimeout([](){
  // Hello!
});
timer.setSingleShot(false); // periodic timer - constant frequency of events
timer.start(10); // 10 ms
```
I will offer RPi support (optimization) as soon as I deal with the libASICamera2Boost - Currently, I can help, but I need the exact spot of the problem.

ken-self, 2021-04-27:
Thanks @pawel-soja I think I can manage on my own using inditimer

ken-self, 2021-05-04:
@knro do I need to do anything to add libpigpiod to the nightlies? indi-asi-power and indi-rpi-gpio have a dependency on it.
Latest PR adds the -m option which reduces idle CPU to near zero %. The INDI timer works well with about 1 ms error per 1000 ms which is negligible.

closed 2021-05-06 by knro