indilib/indi-3rdparty #382: Can't connect with either latest indi_atik_wheel or indi_qhy_ccd

grantbrown, 2021-04-24:
Apologies if this is the wrong place, the indilib forum appears to be broken, at least for new user accounts. I tried two machines and four browsers, but kept getting:

    WHOOOPS! ERROR!
    Looks like something went wrong. Sorry!
    0 Call to undefined method KunenaUser::getName() 

Anyway, I use Indi via Ekos/Kstars on Fedora, but I've hit two issues today that I'm having trouble resolving. I recently picked up a QHY-268M, and an Atik EFW 2.2, but can't get either of them to work. Both devices seem to connect and function fine on Windows using the software from the manufacturers.

### Device/Software Info:

**Machine:** Fedora 33 64 bit
**Indi:** built from source, latest (8cb8dd5ce88be)
**Kstars:** built from source, latest (3d1d568d894)
**Indi-3rdparty components:** built from source, latest (60009a959bdd)

**Hardware setup:**

The devices were each connected to the computer in turn with nothing else attached, and used in a profile in Ekos with simulators for all other devices. Two USB cables were used. Both devices show up correctly via the "lsusb" command:

Atik:
Bus 001 Device 024: ID 04d8:003f Microchip Technology, Inc. Filter Wheel 2

QHY:
Bus 001 Device 023: ID 1618:c268 Cypress WestBridge

## Problem 1: Atik EFW2.2

I've installed libatik and indi-atik, but the driver in ekos crashes. When I run indiserver -v, I get the following result:

```
bash-5.0$ indiserver -v indi_atik_wheel
2021-04-24T15:12:11: startup: indiserver -v indi_atik_wheel  
2021-04-24T15:12:11: Driver indi_atik_wheel: pid=18045 rfd=3 wfd=6 efd=7
2021-04-24T15:12:11: listening to port 7624 on fd 4
Child process 18045 died
2021-04-24T15:12:12: Driver indi_atik_wheel: stderr EOF
2021-04-24T15:12:12: Driver indi_atik_wheel: restart #1
2021-04-24T15:12:12: Driver indi_atik_wheel: pid=18053 rfd=0 wfd=6 efd=7
2021-04-24T15:12:12: Driver indi_atik_wheel: stderr EOF
Child process 18053 died
2021-04-24T15:12:12: Driver indi_atik_wheel: restart #2
2021-04-24T15:12:12: Driver indi_atik_wheel: pid=18070 rfd=0 wfd=6 efd=7
2021-04-24T15:12:12: Driver indi_atik_wheel: stderr EOF
Child process 18070 died
2021-04-24T15:12:12: Driver indi_atik_wheel: restart #3
2021-04-24T15:12:12: Driver indi_atik_wheel: pid=18079 rfd=0 wfd=6 efd=7
2021-04-24T15:12:13: Driver indi_atik_wheel: stderr EOF
Child process 18079 died
2021-04-24T15:12:13: Driver indi_atik_wheel: restart #4
2021-04-24T15:12:13: Driver indi_atik_wheel: pid=18093 rfd=0 wfd=6 efd=7
2021-04-24T15:12:13: Driver indi_atik_wheel: stderr EOF
Child process 18093 died
2021-04-24T15:12:13: Driver indi_atik_wheel: restart #5
2021-04-24T15:12:13: Driver indi_atik_wheel: pid=18105 rfd=0 wfd=6 efd=7
2021-04-24T15:12:13: Driver indi_atik_wheel: stderr EOF
Child process 18105 died
2021-04-24T15:12:13: Driver indi_atik_wheel: restart #6
2021-04-24T15:12:13: Driver indi_atik_wheel: pid=18121 rfd=0 wfd=6 efd=7
2021-04-24T15:12:13: Driver indi_atik_wheel: stderr EOF
Child process 18121 died
2021-04-24T15:12:13: Driver indi_atik_wheel: restart #7
2021-04-24T15:12:13: Driver indi_atik_wheel: pid=18133 rfd=0 wfd=6 efd=7
Child process 18133 died
2021-04-24T15:12:14: Driver indi_atik_wheel: restart #8
2021-04-24T15:12:14: Driver indi_atik_wheel: pid=18147 rfd=0 wfd=6 efd=7
Child process 18147 died
2021-04-24T15:12:14: Driver indi_atik_wheel: restart #9
2021-04-24T15:12:14: Driver indi_atik_wheel: pid=18160 rfd=0 wfd=6 efd=7
Child process 18160 died
2021-04-24T15:12:14: Driver indi_atik_wheel: restart #10
2021-04-24T15:12:14: Driver indi_atik_wheel: pid=18171 rfd=0 wfd=6 efd=7
Child process 18171 died
2021-04-24T15:12:14: Driver indi_atik_wheel: stderr EOF
2021-04-24T15:12:14: Driver indi_atik_wheel: Terminated after #10 restarts.
2021-04-24T15:12:14: good bye
```

Some topics on the forum seem to indicate that I should get the Atik software from cloudmakers, but it seems older (and requires an antiquated version of libnova and won't install by default).

## Problem 2: QHY268M

Despite everything seeming to install correctly, the camera doesn't seem to be detected at all. Running qhy_ccd_test, I get:

`
QHY Test CCD using SingleFrameMode, Version: 1.00
-- InitQHYCCDResource param
QHYCCD|QHYCCD.CPP|InitQHYCCDResource()|START
QHYCCD|QHYCCD.CPP|InitQHYCCDResource|auto_detect_camera:false,call InitQHYCCDResourceInside
QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|START
QHYCCD|QHYCCD.CPP|libusb_version 1.0.24.11584
QHYCCD|QHYCCD.CPP|libusb_init(libqhyccd_context) called...
QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|numdev set to 0
QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|END
************************** config file path 21.4.15.16 svn: 10823 ************************************
QHYCCD|QHYCCD.CPP|InitQHYCCDResource|Load ini filePath = /home/USERNAME/devel/indi-astro/indi-3rdparty/indi-qhy/build fileName = qhyccd.ini
SDK resources initialized.
No QHYCCD camera found, please check USB or power.
`

Has anyone experienced either of these issues, or know what else I should try? Any help is greatly appreciated.

grantbrown, 2021-04-24:
I'm closing this, since based on some experiments with Gnome Boxes, it looks like things work correctly on Ubuntu, at least for 18.04. I'm going to install an up to date 21.04 system and see how things work. In case others are experiencing these issues, I'll report my findings back here.

grantbrown, 2021-04-24:
As an update, I was able to resolve the issue with the QHY 268M by installing the SDK directly from QHY, rather than the bundled libqhy in this repository. 

I'm still at a loss on the EFW. It worked when running Ubuntu 18.04 through gnome Boxes, but the issue has recurred in a native 21.04 install, giving the same result whether the EFW is present or not, and no matter which version of the Atik SDK I install. This (unresolved) [issue looks related](https://www.indilib.org/forum/focusers-filter-wheels/8574-atik-filter-wheel-driver-crashes.html).

grantbrown, 2021-04-25:
After continued experimentation, it looks like the issue is that the Atik driver needs to be rebuilt, possibly due to a change in glibc.

grantbrown, 2021-09-24:
[Re-opening](https://www.indilib.org/forum/focusers-filter-wheels/8574-atik-filter-wheel-driver-crashes.html?start=12#75905), as this is still a thing and Atik doesn't appear to have released a fix. Any ideas about a workaround?

grantbrown, 2021-09-27:
After some help in [this issue](https://github.com/indilib/indi-3rdparty/issues/292), it looks like this should be fixable on Ubuntu 20.04 by installing the following system packages in addition to those identified elsewhere in the build process:

    libgcc1
    ib64stdc++6

TallFurryMan, 2022-04-28:
Let's try to review and fix this issue again, if possible. Do you still own the EFW2? With the latest indi/indi-3rdparty drivers (I suggest the ppa drivers first), under any flavor of Linux that would suit you, what is the output of:

- `cat /etc/os-release`
- `apt search libindi1 && apt info libindi1` if the INDI core binaries are from the ppa, else the git HEAD of `indilib/indi` if built from source
- `apt search indi-atik && apt info indi-atik` if the INDI drivers are from the ppa, else the git HEAD of `indlib/indi-3rdparty' if build from source
- `ldd $(which indi_atik_wheel)`
- `indiserver -v indi_atik_wheel` with no device plugged
- `lsusb -t` with the EFW2 device plugged
- `indiserver -v indi_atik_wheel` with the EFW2 device plugged

If the driver misses no library but fails, I will try to add logs progressively in order to nail the issue down. It would be easier if I had a EFW2 at hand, but these things are quite expensive :) Maybe at some point it would be possible to debug further if a StellarMate with the device is accessible, but let's start like this first.

grantbrown, 2022-04-29:
Appreciate it! 

I just fired up a new 20.04 VM (looks like not everything is available through the ppa yet on 22.04) and installed indi, kstars, and the full suite of indi-3rdparty libraries. And... no crash. I was able to address the filter wheel no problem. 

I haven't attempted to build the whole thing from source in a while, but it seems like some dependency is getting met that wasn't previously when building from source. That's promising enough that I'm going to take the plunge and reimage my primary astronomy laptop and see if the good luck holds. 

Would it be helpful to grab that info from the working config, or alternatively to try a clean build from source and see if it's still happening that way?

TallFurryMan, 2022-04-30:
Thanks for reporting, and glad it connects. The information output by the commands above is usually a good start to debug such an issue. If the EFW2 works, I suppose a few functional and stability tests should be a better option on your setup.

grantbrown, 2022-06-24:
Ok, so I feel like I might be losing my mind a little bit. Sorry for the delay, I'm just now getting around to re-trying this on an actual installation of 20.04 (and 21.04, and 22.04). While things seem to work in that original VM I spun up, I'm getting errors when installing indi-full and kstars-bleeding from the ppa. Here's the output from the commands you highlight. 

```
> cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.4 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.4 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

So far so good. 

```
> apt search libindi1 && apt info libindi1
Sorting... Done
Full Text Search... Done
libindi1/focal,now 1.9.6+202205211403~ubuntu20.04.1 amd64 [installed,automatic]
  Instrument-Neutral Device Interface library -- shared library

Package: libindi1
Version: 1.9.6+202205211403~ubuntu20.04.1
Priority: optional
Section: libs
Source: libindi
Maintainer: Jasem Mutlaq <mutlaqja@ikarustech.com>
Installed-Size: 1,773 kB
Depends: libc6 (>= 2.29), libcfitsio8 (>= 3.470~), libfftw3-double3 (>= 3.3.5), libgcc-s1 (>= 3.0), libgsl23 (>= 2.5), libgslcblas0 (>= 2.4), libjpeg8 (>= 8c), libnova-0.16-0, libogg0 (>= 1.0rc3), libstdc++6 (>= 6), libtheora0 (>= 1.0), libusb-1.0-0 (>= 2:1.0.9), zlib1g (>= 1:1.1.4), libindi-data (>= 1.9.6+202205211403~ubuntu20.04.1), indi-bin (>= 1.9.6+202205211403~ubuntu20.04.1)
Conflicts: libindi-plugins, libindi0, libindialignmentdriver1, libindidriver1
Replaces: libindi-plugins, libindi0, libindialignmentdriver1, libindidriver1
Download-Size: 454 kB
APT-Manual-Installed: no
APT-Sources: http://ppa.launchpad.net/mutlaqja/ppa/ubuntu focal/main amd64 Packages
Description: Instrument-Neutral Device Interface library -- shared library
 INDI (Instrument-Neutral Device Interface) is a distributed XML-based
 control protocol designed to operate astronomical instrumentation.
 INDI is small, flexible, easy to parse, scalable, and stateless.
 It supports common DCS functions such as remote control, data acquisition,
 monitoring, and a lot more.
 .
 This package contains the shared library.
```

Seems ok

```
> apt search indi-atik && apt info indi-atik
Sorting... Done
Full Text Search... Done
indi-atik/focal,now 2.6+202205230947~ubuntu20.04.1 amd64 [installed]
  INDI Driver for Atik cameras and filter wheels.

indi-atik-dbg/focal 2.6+202205230947~ubuntu20.04.1 amd64
  INDI Driver for Atik cameras and filter wheels.

Package: indi-atik
Version: 2.6+202205230947~ubuntu20.04.1
Priority: extra
Section: science
Maintainer: Jasem Mutlaq <mutlaqja@ikarustech.com>
Installed-Size: 125 kB
Depends: libc6 (>= 2.14), libcfitsio8 (>= 3.470~), libgcc-s1 (>= 3.0), libindi1, libstdc++6 (>= 5.2), libatik
Download-Size: 30.1 kB
[... 217 lines cut ...]
2022-06-24T18:52:22: Driver indi_atik_wheel: stderr EOF
Child process 5072 died
2022-06-24T18:52:22: Driver indi_atik_wheel: restart #6
2022-06-24T18:52:22: Driver indi_atik_wheel: pid=5081 rfd=0 wfd=6 efd=7
Child process 5081 died
2022-06-24T18:52:24: Driver indi_atik_wheel: restart #7
2022-06-24T18:52:24: Driver indi_atik_wheel: pid=5090 rfd=0 wfd=6 efd=7
2022-06-24T18:52:25: Driver indi_atik_wheel: stderr EOF
Child process 5090 died
2022-06-24T18:52:25: Driver indi_atik_wheel: restart #8
2022-06-24T18:52:25: Driver indi_atik_wheel: pid=5099 rfd=0 wfd=6 efd=7
2022-06-24T18:52:26: Driver indi_atik_wheel: stderr EOF
2022-06-24T18:52:26: Driver indi_atik_wheel: restart #9
Child process 5099 died
2022-06-24T18:52:26: Driver indi_atik_wheel: pid=5108 rfd=0 wfd=6 efd=7
Child process 5108 died
2022-06-24T18:52:28: Driver indi_atik_wheel: restart #10
2022-06-24T18:52:28: Driver indi_atik_wheel: pid=5117 rfd=0 wfd=6 efd=7
2022-06-24T18:52:29: Driver indi_atik_wheel: stderr EOF
Child process 5117 died
2022-06-24T18:52:29: Driver indi_atik_wheel: Terminated after #10 restarts.
2022-06-24T18:52:29: good bye
```

Any thoughts? I can repeat these under the VM where things are mysteriously working if that would be helpful.

grantbrown, 2022-06-25:
Here's some stack trace information 

```
Package: indi-atik 2.6+202205230947~ubuntu20.04.1 [origin: LP-PPA-mutlaqja]
PackageArchitecture: amd64
ProcCpuinfoMinimal:
 processor	: 7
 vendor_id	: AuthenticAMD
 cpu family	: 23
 model		: 24
 model name	: AMD Ryzen 7 3700U with Radeon Vega Mobile Gfx
 stepping	: 1
 microcode	: 0x8108109
 cpu MHz		: 1700.000
 cache size	: 512 KB
 physical id	: 0
 siblings	: 8
 core id		: 3
 cpu cores	: 4
 apicid		: 7
 initial apicid	: 7
 fpu		: yes
 fpu_exception	: yes
 cpuid level	: 13
 wp		: yes
 flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb hw_pstate ssbd ibpb vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves clzero irperf xsaveerptr arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov succor smca sme sev sev_es
 bugs		: sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
 bogomips	: 4591.21
 TLB size	: 2560 4K pages
 clflush size	: 64
 cache_alignment	: 64
 address sizes	: 43 bits physical, 48 bits virtual
 power management: ts ttp tm hwpstate eff_freq_ro [13] [14]
ProcVersionSignature: Ubuntu 5.13.0-51.58~20.04.1-generic 5.13.19
Registers:
 rax            0xeed               3821
 rbx            0x0                 0
 rcx            0x0                 0
 rdx            0x0                 0
 rsi            0x0                 0
 rdi            0x7f6baf4b2eed      140100479168237
 rbp            0x7f6baf6f7028      0x7f6baf6f7028 <AtikCore::DebugHelper::app>
 rsp            0x7fff41f184b8      0x7fff41f184b8
 r8             0x7f6bae459350      140100462023504
 r9             0x73656369766564    32481099967260004
 r10            0x7                 7
 r11            0x246               582
 r12            0x55e7db56ab60      94454305696608
 r13            0x55e7db508b00      94454305295104
 r14            0x55e7db5025e0      94454305269216
 r15            0x7f6baf448c60      140100478733408
 rip            0x7f6baf106bf2      0x7f6baf106bf2 <__strcmp_avx2+34>
 eflags         0x10283             [ CF SF IF RF ]
 cs             0x33                51
 ss             0x2b                43
 ds             0x0                 0
 es             0x0                 0
 fs             0x0                 0
 gs             0x0                 0
SegvAnalysis:
 Segfault happened at: 0x7f6baf106bf2 <__strcmp_avx2+34>:	vpcmpeqb (%rsi),%ymm1,%ymm0
 PC (0x7f6baf106bf2) ok
 source "(%rsi)" (0x00000000) not located in a known VMA region (needed readable region)!
 destination "%ymm1" ok
SegvReason: reading NULL VMA
SourcePackage: indi-atik
Stacktrace:
 #0  __strcmp_avx2 () at ../sysdeps/x86_64/multiarch/strcmp-avx2.S:102
 No locals.
 #1  0x00007f6baf49ff09 in AtikCore::StringHelper::AreTheSame(char const*, char const*) () from /lib/x86_64-linux-gnu/libatikcameras.so
[... 95 lines cut ...]
 #1  0x00007f6baf49ff09 in AtikCore::StringHelper::AreTheSame(char const*, char const*) () from /lib/x86_64-linux-gnu/libatikcameras.so
 No symbol table info available.
 #2  0x00007f6baf47e7f0 in AtikCore::ExternalFilterWheelManager::RefreshDevicesLinux() () from /lib/x86_64-linux-gnu/libatikcameras.so
 No symbol table info available.
 #3  0x00007f6baf47ea3c in AtikCore::ExternalFilterWheelManager::Refresh() () from /lib/x86_64-linux-gnu/libatikcameras.so
 No symbol table info available.
 #4  0x00007f6baf47eaa4 in AtikCore::ExternalFilterWheelManager::LockFW(int) () from /lib/x86_64-linux-gnu/libatikcameras.so
 No symbol table info available.
 #5  0x00007f6baf42cca7 in AtikCore::ArtemisDLL::EFWIsPresent(int) () from /lib/x86_64-linux-gnu/libatikcameras.so
 No symbol table info available.
 #6  0x00007f6baf42aba1 in ArtemisEFWIsPresent () from /lib/x86_64-linux-gnu/libatikcameras.so
 No symbol table info available.
 #7  0x000055e7da08dae4 in Loader::Loader() ()
 No symbol table info available.
 #8  0x000055e7da08cefa in ?? ()
 No symbol table info available.
 #9  0x000055e7da08e63d in __libc_csu_init ()
 No symbol table info available.
 #10 0x00007f6baefa7010 in __libc_start_main (main=0x7f6bb041b3a3 <main>, argc=1, argv=0x7fff41f188f8, init=0x55e7da08e5f0 <__libc_csu_init>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fff41f188e8) at ../csu/libc-start.c:264
         result = <optimized out>
         unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140100475765480, 140100474019367, 140100496564576, 140100496412672, 140734299736328, 140100496394352, 1, 140100496009700}, mask_was_saved = -1336801184}}, priv = {pad = {0x7f6bb05bab9a <call_init+154>, 0x0, 0x1, 0x7fff41f188f8}, data = {prev = 0x7f6bb05bab9a <call_init+154>, cleanup = 0x0, canceltype = 1}}}
         not_first_call = <optimized out>
 #11 0x000055e7da08c5ae in _start ()
 No symbol table info available.
```

grantbrown, 2022-06-28:
Getting closer. I pulled down the repo and am playing with ae0a4412, since it seems that the next most recent commit may have broken compilation. 

Compiling and running just `./indi_atik_wheel` immediately segfaults, so it seems an EFW isn't needed to debug this issue. Here's the valgrind output:

```
==10449== Memcheck, a memory error detector
==10449== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==10449== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==10449== Command: ./indi_atik_wheel
==10449== 
==10449== Conditional jump or move depends on uninitialised value(s)
==10449==    at 0x483EF49: strlen (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==10449==    by 0x4B5DF28: AtikCore::StringHelper::Length(char const*) (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4B5DF40: AtikCore::StringHelper::Copy(char*, char const*) (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4B1A090: AtikCore::AtikCameraManager::AtikCameraManager(AtikCore::IArtemisHandleGenerator*) (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4AF41FD: AtikCore::ArtemisDLLDeviceManager::ArtemisDLLDeviceManager(AtikCore::IUSBDetector*, AtikCore::ILibUSBDeviceManager*, AtikCore::IFlyCaptureDeviceManager*, AtikCore::IUSB1DeviceManager*, AtikCore::IUSB1Wrapper*, AtikCore::IAtikAirDeviceManager*, AtikCore::IAtikAirWrapper*, AtikCore::IHIDDeviceManager*, AtikCore::IFTDIDeviceManager*) (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4AF1A55: AtikCore::ArtemisDLL::ArtemisDLL(AtikCore::ILibUSB*, AtikCore::IUSBDetector*, AtikCore::IFlyCaptureWrapperManager*, AtikCore::IUSB1Wrapper*, AtikCore::IAtikAirWrapper*, AtikCore::IHIDDeviceWrapper*, AtikCore::IFTDIDeviceWrapper*) (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4ADFD56: ??? (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4011B99: call_init.part.0 (dl-init.c:72)
==10449==    by 0x4011CA0: call_init (dl-init.c:30)
==10449==    by 0x4011CA0: _dl_init (dl-init.c:119)
==10449==    by 0x4001139: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==10449==  Uninitialised value was created by a heap allocation
==10449==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==10449==    by 0x4AF1A18: AtikCore::ArtemisDLL::ArtemisDLL(AtikCore::ILibUSB*, AtikCore::IUSBDetector*, AtikCore::IFlyCaptureWrapperManager*, AtikCore::IUSB1Wrapper*, AtikCore::IAtikAirWrapper*, AtikCore::IHIDDeviceWrapper*, AtikCore::IFTDIDeviceWrapper*) (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4ADFD56: ??? (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4011B99: call_init.part.0 (dl-init.c:72)
==10449==    by 0x4011CA0: call_init (dl-init.c:30)
==10449==    by 0x4011CA0: _dl_init (dl-init.c:119)
==10449==    by 0x4001139: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==10449== 
==10449== Syscall param openat(filename) points to uninitialised byte(s)
==10449==    at 0x5D8BDB4: open (open64.c:48)
==10449==    by 0x5D0E015: _IO_file_open (fileops.c:189)
==10449==    by 0x5D0E2D9: _IO_file_fopen@@GLIBC_2.2.5 (fileops.c:281)
==10449==    by 0x5D0098D: __fopen_internal (iofopen.c:75)
==10449==    by 0x5D0098D: fopen@@GLIBC_2.2.5 (iofopen.c:86)
==10449==    by 0x5B4D633: std::__basic_file<char>::open(char const*, std::_Ios_Openmode, int) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==10449==    by 0x5B90F4E: std::basic_filebuf<char, std::char_traits<char> >::open(char const*, std::_Ios_Openmode) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==10449==    by 0x4AF1BC0: AtikCore::ArtemisDLL::ArtemisDLL(AtikCore::ILibUSB*, AtikCore::IUSBDetector*, AtikCore::IFlyCaptureWrapperManager*, AtikCore::IUSB1Wrapper*, AtikCore::IAtikAirWrapper*, AtikCore::IHIDDeviceWrapper*, AtikCore::IFTDIDeviceWrapper*) (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4ADFD56: ??? (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449==    by 0x4011B99: call_init.part.0 (dl-init.c:72)
==10449==    by 0x4011CA0: call_init (dl-init.c:30)
==10449==    by 0x4011CA0: _dl_init (dl-init.c:119)
==10449==    by 0x4001139: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==10449==  Address 0x1ffefffcc0 is on thread 1's stack
==10449==  in frame #6, created by AtikCore::ArtemisDLL::ArtemisDLL(AtikCore::ILibUSB*, AtikCore::IUSBDetector*, AtikCore::IFlyCaptureWrapperManager*, AtikCore::IUSB1Wrapper*, AtikCore::IAtikAirWrapper*, AtikCore::IHIDDeviceWrapper*, AtikCore::IFTDIDeviceWrapper*) (???:)
==10449==  Uninitialised value was created by a stack allocation
==10449==    at 0x4AF17CC: AtikCore::ArtemisDLL::ArtemisDLL(AtikCore::ILibUSB*, AtikCore::IUSBDetector*, AtikCore::IFlyCaptureWrapperManager*, AtikCore::IUSB1Wrapper*, AtikCore::IAtikAirWrapper*, AtikCore::IHIDDeviceWrapper*, AtikCore::IFTDIDeviceWrapper*) (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
==10449== 
==10449== Invalid read of size 1
==10449==    at 0x483FED7: strcmp (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==10449==    by 0x4B5DF08: AtikCore::StringHelper::AreTheSame(char const*, char const*) (in /usr/lib/x86_64-linux-gnu/libatikcameras.so.2.2.10)
[... cut ...]
==10449==    definitely lost: 192 bytes in 2 blocks
==10449==    indirectly lost: 3,932 bytes in 58 blocks
==10449==      possibly lost: 10,632 bytes in 26 blocks
==10449==    still reachable: 279,464 bytes in 2,385 blocks
==10449==         suppressed: 0 bytes in 0 blocks
==10449== Rerun with --leak-check=full to see details of leaked memory
==10449== 
==10449== For lists of detected and suppressed errors, rerun with: -s
==10449== ERROR SUMMARY: 4 errors from 3 contexts (suppressed: 0 from 0)
```

knro, 2022-06-29:
The crash is happening inside the SDK library itself. @TallFurryMan Any updates to the library or was this crash reported to Atik?

TallFurryMan, 2022-06-29:
Let me analyse all this information and contact Atik. Thanks @grantbrown !

grantbrown, 2022-06-29:
@TallFurryMan No worries at all, happy to continue to run some tests. I'd recommend trying to replicate it in your environment too just to be sure there's not something weird that I'm doing. The output was generated with just

```
valgrind --trace-children=yes --track-origins=yes ./indi_atik_wheel
```

and I don't think it requires the EFW at all, since it's failing in the library just while trying to figure out where the EFW is and if it's ready (both with one plugged in and not).

grantbrown, 2022-07-03:
As a brief update, I've successfully used `virt-manager` instead of `boxes` to interface with the wheel, exposing it over the local network to Ekos as a remote resource. I still don't know why it seems to work inside a virtual machine and not natively. I also now don't have a working configuration for [my CCD](https://github.com/indilib/indi-3rdparty/issues/594), so more experimentation is required.

grantbrown, 2022-07-05:
Excellent, I'll try that out this week. Thanks!

grantbrown, 2022-07-09:
Brief status update. I dumped the files from the beta SDK in [here](https://github.com/grantbrown/indi-3rdparty/commit/1aadb4bf945e3ec83e19d33d23688814be72d26e) and also [made](https://github.com/grantbrown/indi-3rdparty/commit/4a60f4d500dc57719eb0b1579fbce59049a9094e) a minor typo fix and logging change. I haven't tested it extensively, and so far have only looked at 64 bit Linux, but I was able to interface with the EFW without resorting to a VM. 

It seems like there might be some [other fragile things](https://gist.github.com/grantbrown/544c5ca68a81d2c6e0a0d88b9d78c541) happening inside the SDK, but overall progress!

TallFurryMan, 2022-07-25:
I'll try to integrate this SDK this week. There's also the other issue about the executable stack that I need to ping Atik with.

PR #635 (2022-08-20): Upgrade Atik SDK to 2022.07.14.1005.
This PR upgrades the Atik SDK to version 2022.07.14.1005 (Beta version).

There is an additional Beta SDK available, 2022.08.16, with a fix for Apx60, which I may upgrade to later.

It is probable 2022.07.13 fixes #382.

**2022.07.15 -  DLL v2022.07.14.1005**

- First relase containing firmware for the TE-47.

**2022.07.13 - DLL v2022.07.13.1003**

- Linux bug fix
  - Fixed issue with iterating through connected udev devices.

**2022.07.06 - DLL v2022.07.06.999**

- Apx Series
  - Updated the Apx26/60’s cooling algorithm so that it sits tighter on the setpo [... cut]
closed 2022-08-20 by knro