indilib/indi-3rdparty #387: Is there still a wayward IS function?

JerryBlack, 2021-04-30:
sorry my 1st post to github...
**Describe the bug**

Compile error 
Building CXX object indi-toupbase/CMakeFiles/indi_nncam_ccd.dir/indi_toupbase.cpp.o
In file included from /home/odroid/media/odroid/T5_1T_SSD/Projects/indi/indi-3rdparty/indi-toupbase/indi_toupbase.cpp:30:
/home/odroid/media/odroid/T5_1T_SSD/Projects/indi/indi-3rdparty/indi-toupbase/indi_toupbase.cpp: In member function ‘virtual bool ToupBase::ISNewSwitch(const char*, const char*, ISState*, char**, int)’:
/home/odroid/media/odroid/T5_1T_SSD/Projects/indi/indi-3rdparty/indi-toupbase/indi_toupbase.h:57:15: error: ‘Nncam_AwbOnce’ was not declared in this scope; did you mean ‘Nncam_AwbInit’?
   57 | #define FP(x) Nncam_##x
      |               ^~~~~~

**Desktop (please complete the following information):**
 Ubuntu 20.04.2 LTS

knro, 2021-04-30:
Install libnncam-dev first?

JerryBlack, 2021-04-30:
Ah, thank you. that must be it. libnncam-dev was missing from my compile script  apt-get install.

I'm still lost a somewhat as  an apt-get install can't find it on Ubuntu 20.04.2
sudo apt-get install libnncam-dev
  Reading package lists... Done
  Building dependency tree       
  Reading state information... Done
E: Unable to locate package libnncam-dev

I see https://github.com/openastroproject/libnncam-pkg is there to provide a pkg, but I'm out of my league in seeing an automatic connection between that project and apt-get.

openastroproject, 2021-04-30:
Given the relevant upstream files, https://github.com/openastroproject/libnncam-pkg would probably work for you, but it's not related to indilib and might perhaps serve more to confuse than illuminate in this instance.  I just made the files available in case they were useful to someone else.

I think building indilib will build a libnncam-dev package somewhere along the way, but Jasem might have to help you out with that one.

Of course, what would really make everyone's life simpler would be for all the vendors to supply the packages and save us all the trouble of having to make them ourselves.  Like that's going to happen :)

pawel-soja, 2021-04-30:
You have old libraries installed (nncam).
If you work on sources, install libraries from indi-3rdparty (this repo), e.g.

```bash
for driver in libtoupcam libaltaircam libstarshootg libnncam libmallincam
do
    (mkdir -p build/$driver && cd build/$driver && cmake ~/Projects/indi-3rdparty/$driver && make && sudo make install)
done
```

pawel-soja, 2021-05-07:
@JerryBlack Is there anything else we can help you with?

JerryBlack, 2021-05-07:
@pawel-soja **Thank you**, I hope I'm good now. I'll be able to test tonight.

closed 2021-05-09 by JerryBlack