indilib/indi-3rdparty #389: Moravian USB initialization issue

knro, 2021-05-02:
After the changes in INDI v1.9.0, there is an issue when initializing Moravian USB cameras (didn't test Ethernet).

gxccd_initialize_usb(...) always returns nullptr. When switching back to using the ISXXX functions, the issue goes away and it returns a valid handle. I can't see what would be causing such an issue, since it's affecting the USB enumeration of the library. Is there some order of initialization that got messed up?

gxccd_enumerate_usb works fine in both versions and detects camera ID 30031.

@pawel-soja @sWski Any ideas on what's going on here?

pawel-soja, 2021-05-02:
One difference I can see is that gxccd_initialize_xxx is now called during gxccd_enumerate_xxx.
I will throw in a fix for testing.

sWski, 2021-05-03:
Looking through the library code, it's not possible to call gxccd_initialize_usb() in gxccd_enumerate_usb(), as the library claims the interface with libusb and another libusb_open() fails because of that. @pawel-soja's fix should be sufficient. However there is no reason not to release the interface before calling the callback function, so I can fix this in next version.

knro, 2021-05-25:
It's working now fine. Thanks @pawel-soja for the fix. Closing.

closed 2021-05-25 by knro