indilib/indi-3rdparty #396: indi-bresserexos2 - Errors During Install

tstevens83, 2021-05-10:
I'm running Astroberry (updated today) and trying to install indi-bresserexos2 drivers.  I'm getting errors duing the "cmake --build ." command, see below:

astroberry@astroberry:~/indi-bresserexos2/build $ cmake ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/astroberry/indi-bresserexos2/build

astroberry@astroberry:~/indi-bresserexos2/build $ cmake --build .

Scanning dependencies of target SerialDeviceControl
[ 60%] Building CXX object SerialDeviceControl/CMakeFiles/SerialDeviceControl.dir/SerialCommand.cpp.o
[ 60%] Linking CXX static library libSerialDeviceControl.a
[ 60%] Built target SerialDeviceControl
Scanning dependencies of target indi_bresserexos2
[ 60%] Building CXX object CMakeFiles/indi_bresserexos2.dir/BresserExosIIGoToDriver.cpp.o
/home/astroberry/indi-bresserexos2/BresserExosIIGoToDriver.cpp: In member function ‘virtual bool GoToDriver::BresserExosIIDriver::initProperties()’:
/home/astroberry/indi-bresserexos2/BresserExosIIGoToDriver.cpp:92:2: error: ‘defineProperty’ was not declared in this scope
  defineProperty(&SourceCodeRepositoryURLTP);
  ^~~~~~~~~~~~~~
/home/astroberry/indi-bresserexos2/BresserExosIIGoToDriver.cpp:92:2: note: suggested alternative: ‘deleteProperty’
  defineProperty(&SourceCodeRepositoryURLTP);
  ^~~~~~~~~~~~~~
  deleteProperty
make[2]: *** [CMakeFiles/indi_bresserexos2.dir/build.make:63: CMakeFiles/indi_bresserexos2.dir/BresserExosIIGoToDriver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:96: CMakeFiles/indi_bresserexos2.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

tstevens83, 2021-05-10:
yes and thought the comments I received were ok but I could be mistaken...

astroberry@astroberry:~ $ sudo apt-get install libindi1 libindidriver1 libindi-data libindi-dev libindi-plugins libindialignmentdriver1 libnova-0.16-0 libnova-dev libnova-dev-bin
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libnova-0.16-0 is already the newest version (0.16-4).
libnova-0.16-0 set to manually installed.
libnova-dev is already the newest version (0.16-4).
libnova-dev set to manually installed.
libnova-dev-bin is already the newest version (0.16-4).
libnova-dev-bin set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libindi1 : Conflicts: libindi-plugins but 1.7.5+dfsg-1 is to be installed
            Conflicts: libindialignmentdriver1 but 1.7.5+dfsg-1 is to be installed
            Conflicts: libindidriver1 but 1.7.5+dfsg-1 is to be installed
E: Unable to correct problems, you have held broken packages.

knro, 2021-05-25:
libindi-plugin ..etc are installed from Ubuntu/Debian sources and they conflict with our packages. Remove them first.

tstevens83, 2021-05-25:
thank you for your response... let me make sure i get this straight.
remove libindi and then insall the bresserexos2 drivers/software.  then
reinstall libindi?

knro, 2021-09-28:
Purge all INDI and then reinstall. I cannot reproduce this issue.

closed 2021-09-28 by knro