indilib/indi-3rdparty #401: indi_asi_ccd stop recording

nadvornik, 2021-05-13:
**Describe the bug**

I am testing indi_asi_ccd from current git on RPI4 with asi1600.

I have found that stopping recording does not work correctly - it prints the statistics and closes the file but the cpu load stays high, gdb shows that ASICCD::workerStreamVideo is still running and changing properties like Binning sometimes cause segfaults.

Further debugging shows that ASICCD::StopStreaming is called from the ASICCD::workerStreamVideo thread, so the lock in INDI::SingleThreadPool does nothing. Then isAboutToQuit is set to true and immediately back to false and ASICCD::workerStreamVideo does not notice it and continues running.

BTW, "Expose Abort" during recording has opposite effect - it stops ASICCD::workerStreamVideo but the recording continues.

**To Reproduce**
1. Connect the camera
2. Set Video Record On
3. Set Video Record Off
4. watch the high cpu load 

#0  ASICCD::StopStreaming (this=0x558fead410) at /root/indi-3rdparty/indi-asi/asi_ccd.cpp:1063
#1  0x0000007fa89ce29c in INDI::StreamManagerPrivate::stopRecording (this=this@entry=0x558fec5030, force=false)
    at /root/indi/libs/stream/streammanager.cpp:728
#2  0x0000007fa89d1d78 in INDI::StreamManagerPrivate::newFrame (this=0x558fec5030, buffer=buffer@entry=0x7f8c46e010 "\340",
    nbytes=nbytes@entry=32778240) at /root/indi/libs/stream/streammanager.cpp:270
#3  0x0000007fa89d239c in INDI::StreamManager::newFrame (this=<optimized out>, buffer=buffer@entry=0x7f8c46e010 "\340",
    nbytes=nbytes@entry=32778240) at /root/indi/libs/stream/streammanager.cpp:345
#4  0x0000005573caaa0c in ASICCD::workerStreamVideo (this=0x558fead410, isAboutToQuit=...) at /usr/include/c++/8/bits/unique_ptr.h:342

pawel-soja, 2021-05-13:
Thank you for reporting the problem with a detailed description. The changes have been made to the master branch.

Fixed: https://github.com/indilib/indi/commit/3a7ea5f6cdd8c87067cc6479f6bb371ce008a08a

Please rebuild [INDI Core](https://github.com/indilib/indi) project (and optional INDI 3rdParty).

pawel-soja, 2021-05-13:
> BTW, "Expose Abort" during recording has opposite effect - it stops ASICCD::workerStreamVideo but the recording continues.

That's right, StreamManager is not getting information about streaming interrupted.
The fix will be soon.

nadvornik, 2021-05-14:
The original problem is fixed, but I have added debug output everywhere and it seems that the behavior is not fully correct.
Steps to reproduce:

1. Connect the camera
2. set stream exposure to higher value, like 5s
3. Set Video Record On
4. Set Video Record Off
5. Wait until the driver replies with Video Record Off and state Idle
6. Set CCD_BINNING, CCD_VIDEO_FORMAT or CCD_STREAM_FRAME

If the client is fast enough, the methods ASICCD::UpdateCCDBin, ASICCD::setVideoFormat, ASICCD::UpdateCCDFrame are executed while the ASICCD::workerStreamVideo thread is in ASIStopVideoCapture. It did not crash, but I think it is just luck.

I can see these problems:
1. Driver replies to  Video Record Off too fast. It should wait until the recording is really stopped, otherwise the client has no chance to detect if changing the parameters is safe.
2. CCD_VIDEO_FORMAT is protected with Streamer->isBusy() but it does not work
3. other parameters are probably not protected at all

pawel-soja, 2021-05-14:
Thanks for your in-depth analysis.
I understand that the problems you have detected do not affect the performance and usability of the driver.
There should be a condition to check that threads are terminated before changing parameters.
I plan to review all the code because some of the code was left unchanged on the last refactoring. I will pay attention to the tests you described.

knro, 2021-05-14:
Could this be related as well? https://www.indilib.org/forum/ccds-dslrs/9641-dual-camera-driver-issue-for-qhy-confused-state-of-camera.html#71308

knro, 2021-05-25:
Hello, just wanted to check if there is any update on this issue?

pawel-soja, 2021-05-25:
I was unable to reproduce significant functional problems. I will fix minor issues when CI/Tests works again.

closed 2021-11-24 by knro