indilib/indi-3rdparty #431: Libraries are missing soname

paolostivanin, 2021-08-10:
Version: all released versions, included latest stable `1.9.1`

The following libraries are missing a `SONAME` (`DT_SONAME`) field, which is unfortunate. Programs built with them will not receive proper library requires (`DT_NEEDED`):

```
libaltaircam.so
libASICamera2.so
libEFWFilter.so
libUSB2ST4Conv.so
libEAFFocuser.so
libatikcameras.so
libflycapture.so
libmallincam.so
libnncam.so
libpktriggercord.so
libqsiapi.so
libRicohCameraSDKCpp.so
libmtpricoh.so
libsbig.so
libstarshootg.so
libSVBCameraSDK.so
```

Unfortunately, this is something that only the vendors (ZWO, QHY, etc) can fix.

jengelh, 2021-08-10:
As a workaround, `patchelf --set-soname libstarshootg.so.1` on the packager side is possible.

knro, 2021-08-10:
Binary blobs provided by the camera manufacturers. Can't really do much about it besides using patchelf as suggested above.

paolostivanin, 2021-08-19:
Thanks!
That worked for almost all libs. I have one last issue with `libatikcameras`, because it requires `libflycapture.so.2`:

```
$ patchelf --debug --print-needed indi-3rdparty-1.9.1/libatik/x64/libatikcameras.bin 
...
libflycapture.so.2
...
```

I'm using `patchelf --set-soname 2  libflycapture.so.2`, and it works fine because `DT_SONAME` is correctly set, but when trying to install `libatikcameras2` I get:

```
unresolvable: nothing provides libflycapture.so.2()(64bit) needed by libatikcameras2
```

Any idea how I can sovle this @jengelh ?

paolostivanin, 2021-08-19:
libflycapture has the correct soname set, so I don't understand why OBS is not picking up the lib correctly (https://build.opensuse.org/package/binary/home:polslinux:teststuff/libindi-3rdparty-libs/openSUSE_Tumbleweed/x86_64/libatikcameras2-1.9.1-17.1.x86_64.rpm)
 
```
$ objdump -p indi-3rdparty-1.9.1/libatik/x64/libflycapture.bin |grep 'SONAME'
  SONAME               libflycapture.so.2
```

paolostivanin, 2021-08-19:
found and fixed the packaging issue :) (as always the bug was between the chair and the keyboard)

closed 2021-08-19 by paolostivanin

jengelh, 2021-08-19:
When you (re-)set the SONAME on one library, it may be necessary to do --replace-needed in another.

>Binary blobs provided by the camera manufacturers. Can't really do much about it besides using patchelf as suggested above.

@knro, would it be possible/feasible that indi-3rdparty does the patchelf (perhaps as part of `make install` or something)? Then all distributions would benefit.

paolostivanin, 2021-08-19:
In case it might be helpful, that's what I've used to find missing/wrong soname:
```
for f in $(find lib* -type f -name *bin | grep -E 'x64|64' | sed '/arm64/d' | xargs); do echo "=> Checking $f" ; patchelf --debug --print-soname $f; echo "------"; done
```