indilib/indi-3rdparty #434: ZWO ASI Camera unable to use minimum exposure

aaronwmorris, 2021-08-20:
**Describe the bug**
I am not sure if this is a python, ZWO, Indi, C, or ARM bug.  The minimum exposure for the ASI290MM is 0.000032 but if that exposure is set, the camera returns an error.  This was working in Indi 1.8.9.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Set ASI290MM minimum exposure to 0.000032

**Desktop (please complete the following information):**
 - OS:  Raspbian GNU/Linux 10 (buster)
 - Raspberry Pi 3
 - Version: 1.9.1
 - Python 3.7.3
 - pyindi-client  (git: 68803a1c51fb5aa271302a748e012cf0cee309b2)
 - Tested camera:  ASI290MM
 

**Log Files**
```
new Message 2021-08-20T15:41:02: [ERROR] Requested exposure value (3.2e-05) seconds out of bounds [3.2e-05,2000].
```

```
'CCD_EXPOSURE': {
  'CCD_EXPOSURE_VALUE': 
    {'current': 0.0,
     'format': '%5.2f',
     'max': 2000.0,
     'min': 3.2e-05,
     'step': 1.0}
},
```

knro, 2021-08-23:
indi_setprop works? Ekos?

aaronwmorris, 2021-08-23:
Both indi_setprop and Ekos accept the 0.000032 value and the exposure is properly processed.

knro, 2021-08-23:
Well that takes care of "INDI/C/ARM" and might be just some python issue?

aaronwmorris, 2021-08-23:
It could be some floating point conversion error between python and swig.  I will reopen this bug under the pyindi-client repo.

closed 2021-08-23 by aaronwmorris