indilib/indi-3rdparty #436: ZWO ASI 294MM-PRO camera fails to download with 1x1 binning in Ekos

gsleap, 2021-08-23:
**Describe the bug**
Using Ekos via Kstars v3.5.4 stable on a Raspberry Pi4, the ZWO ASI294MM-Pro camera fails to successfully work in 1x1 bin mode. 2x2 mode is fine. When using 1x1 bin mode, Ekos reports it is capturing, then downloading, but "Downloading" gets hung and after a few retries, Ekos aborts the capture.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start Kstars/Ekos. Start and connect devices.
2. Ensure H and V binning are set to 1 (should be the default) and change gain to 120 and offset to 30 (unity gain). As a side note, Ekos does defaults gain and offset to "--" for this camera...
3. Set up a capture, but setting exposure to anything you like, and count to whatever you want then click  "Add job to sequence queue", then "Start sequence"
4. Ekos will report it is capturing, then will get stuck on "Downloading" for several minutes until it finally aborts the capture.

**Expected behavior**
Ekos should capture and download the image.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Astroberry (Raspbian GNU/Linux 10 (buster) Linux astroberry 5.10.59-v7l+ #1447 SMP Thu Aug 19 12:25:41 BST 2021 armv7l GNU/Linux
 - Version: KStars/Ekos 3.5.4 stable
 - indi-asi                                  1.9-5
 - indi-bin                                  1.9.1-1
 - indi-full                                 1.9.1
 - libindi-dev                               1.9.1-1
 - libindi1:armhf                            1.9.1-1
 - libindi-data                              1.9.1-1

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[log_21-40-14.txt](https://github.com/indilib/indi-3rdparty/files/7032627/log_21-40-14.txt)

Key part of log (near end of log):
```
[2021-08-23T21:44:50.051 AWST DEBG ][   org.kde.kstars.ekos.capture] - Preparing capture job "/home/astroberry/Pictures/ekos/Light/Light_1_secs" for execution.
[2021-08-23T21:44:50.070 AWST INFO ][   org.kde.kstars.ekos.capture] - "Job requires 1.000-second  images, has 0/1 frames captured and will be processed."
[2021-08-23T21:44:50.073 AWST WARN ][   org.kde.kstars.ekos.capture] - Skipping flat/dark cap since it is not connected.
[2021-08-23T21:44:50.076 AWST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1.000-second  image..."
[2021-08-23T21:44:50.134 AWST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI294MM Pro : "[DEBUG] StartExposure->setexp : 1.000s "
[2021-08-23T21:44:53.507 AWST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI294MM Pro : "[DEBUG] Uploading file. Ext: fits, Size: 93559680, sendImage? Yes, saveImage? No "
[2021-08-23T21:44:55.337 AWST DEBG ][           org.kde.kstars.indi] - INDI server disconnected from BLOB manager for Device: "ZWO CCD ASI294MM Pro" Property: "CCD1" Exit code: -1
[2021-08-23T21:44:55.361 AWST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI294MM Pro : "[DEBUG] BLOB transfer took 1.35486 seconds "
[2021-08-23T21:44:55.361 AWST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI294MM Pro : "[DEBUG] Upload complete "
```

knro, 2021-09-28:
Did you check with the latest INDI release? perhaps this is an Astroberry specific issue as well, did you check on PC or other platforms?

knro, 2021-11-24:
If using the same driver works fine with CCDCiel then it's an Ekos issue, not INDI.

closed 2021-11-24 by knro