indilib/indi-3rdparty #437: Problem with ZWO ASI6200MM camera coexisting with ASI290MM

pleiades-coram, 2021-08-23:
SUMMARY
Raspberry (Native Raspbian version)
KStar 3.5.5 beta
KStar Build: 2021-07-28T15:01:33Z

My ZWO ASI6200MM doesn't seem to coexist with my ZWO ASI290MM guide camera.  I plugged my guide camera to the camera USB hub or to the Raspberry USB ports but nothing.  However, both cameras are connecting well when plugged at the same time in using ekos but the main camera is unable to produce image.  Connecting each camera one by one is working well. The problem only happen to my main camera (ASI6200MM) and I needed to power it down to regain control if it. I can say for sure that both cameras can be connected and working altogether when using Windows 7. It seems that's a driver conflict.   

Regards,
Yves

knro, 2021-08-23:
It appears there is some limitation when the SDK is invoked with multiple cameras. If a single executable does this like what we do in INDI then multiple camera operation may not work fully. The only solution to this issue is to run individual driver per camera. The current ZWO INDI driver architecture does not work like this.

So it would take some work to convert the multi-camera single-driver architecture to single-camera single-driver.

pleiades-coram, 2021-08-23:
Hi Jasem,Why asking me to send my problem report here if you are the same person who already responded to me on the other forum? Is someone going to do something about this in a near future?Regards,Yves

knro, 2021-08-23:
Because this is the correct place to report this issue. It's not a KStars issue. Let's see if any of the INDI developers, perhaps @jpaana can suggest a proper solution for this.

pleiades-coram, 2021-08-24:
Okay Jasem, I'm going to wait for his response.Regards,Yves

jpaana, 2021-08-24:
Unfortunately I haven't actually even tried running multiple ASI cameras on one machine as currently one of the two I have is in my allsky camera (120MM-S) and the other (178MC) is in my remote observatory. Both have small sensors and neither of them have hubs either so it's quite different from your equipment in any case.

Running one driver per camera would sound quite logical thing to try as that would minimize the influence they have on each other. But if the other camera doesn't show up at all even with lsusb command, then that wouldn't help. Running things off the Pi directly is often power sensitive and running under powered does cause all sorts of weird issues. Then again running the other camera off the other's (hopefully powered when the 12V is connected?) should also eliminate that...

pleiades-coram, 2021-08-25:
Hi Jarno,

Both cameras are able to connect at the same time with ekos. I can use the guide camera but not the main camera. In using the main camera, the guide camera no longer worked.

Below is the listing when using lsusb command. As you can see, Device 007: ID 03c3:290f and Device 005: ID 03c3:620b have no labels but are related to ZWO cameras.

***@***.***:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 001 Device 004: ID 04ca:0022 Lite-On Technology Corp. 
Bus 001 Device 007: ID 03c3:290f 
Bus 001 Device 005: ID 03c3:620b 
Bus 001 Device 003: ID 04b4:6572 Cypress Semiconductor Corp. 
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Is there a way to run one driver for each camera?

I also tried to connect my main camera with a short USB 2.0 cable. The one provided by them is a flat USB 3.0 cable and it seems fragile. However it didn't help !

Another thing, both cameras use their power from the USB, The only 12V DC port (optionnal) on the ASI6200 is to power the cooling system.

Regards,
Yves

knro, 2021-08-25:
I submitted PR #438 that could fix this issue.

knro, 2021-09-28:
Can you test with "ZWO Camera 1" and "ZWO Camera 2" drivers?

closed 2021-11-24 by knro