indilib/indi-3rdparty #439: Inconsistency in naming .so library fo libsbig

rkaczorek, 2021-08-28:
I have been heavily playing with cross-compilation and Debian packaging for arm architecture on x86_64 platform to overcome Raspberry Pi resource limitations. I believe I can call it success after quite a few nights to make it running properly for all INDI components, libs and drivers. However...

I have noticed that packaging indi-sbig driver renders an error:
`dpkg-shlibdeps: error: cannot find library libsbigudrv.so.1 needed by debian/indi-sbig/usr/bin/indi_sbig_ccd (ELF format: 'elf32-littlearm' abi: '0101002800000000'; RPATH: '')
`
After inspecting libsbig sources (indi-sbig dependens on it) I found out that .so library uses inconsistent naming:
```
$ objdump -p /usr/lib/arm-linux-gnueabihf/libsbig.so.4.9.9 | grep SONAME
  SONAME               libsbigudrv.so.1
```
Apparently it comes directly from binary library provided by libsbig:
```
astroberry@5fe4efd80a22:~/DEV/BUILD$ objdump -p indi-3rdparty/libsbig/libsbig_armhf.bin |grep SONAME
  SONAME               libsbigudrv.so.1
```

Some time ago libsbigudrv2 was replaced by libsbig, mainly for naming convention of indi drivers. The content is libsbig is same though. The tricky part is that original libsbig_armhf.bin comes as a binary from SBIG SDK so it cannot be changed.

Manual solution is to link missing libsbigudrv.so.1 to libsbig.so:
`ln -s /usr/lib/arm-linux-gnueabihf/libsbig.so /usr/lib/arm-linux-gnueabihf/libsbigudrv.so.1`

Persistent solution should be at compile time by using cmake symlink to create symbolic link.
@knro could you confirm that? Does the issue exist while compiling on Ubuntu?
I guess that adding cmake symlink to indi-3rdparty/libsbig/CMakeLists.txt should be totally safe unless libsbigudrv.so.1 is provided by a package I'm not aware of. In any case I can prepare PR for this issue if needed.

knro, 2021-08-29:
I also cross compile but this was never an issue. Does this continue to build a debian package? or it aborts here?

rkaczorek, 2021-08-29:
Yes, it does compile after symbolic link added. It actually compiles also without symbolic link but packaging does not work. I guess because dpkg-shlibdeps picks .so name from library and does not rely on naming provided by a filename. What do you use for building a debian package? It could be the case that it depends on packaging tools.

knro, 2021-08-30:
Just the usual tools like debuild

rkaczorek, 2021-08-30:
So there is no difference in tools used. I think I know why there is a difference in building this library on both sides ;)
```
$ apt-file find libsbigudrv
libsbigudrv2: /usr/lib/libsbigudrv.so.2   
libsbigudrv2: /usr/lib/libsbigudrv.so.2.0.0
libsbigudrv2: /usr/share/doc/libsbigudrv2/changelog.Debian.gz
libsbigudrv2: /usr/share/doc/libsbigudrv2/copyright
libsbigudrv2-dev: /usr/lib/libsbigudrv.so
libsbigudrv2-dev: /usr/share/doc/libsbigudrv2-dev/changelog.Debian.gz
libsbigudrv2-dev: /usr/share/doc/libsbigudrv2-dev/copyright
```
So there are 2 packages related, namely libsbigudrv2 and libsbigudrv2-dev.
When you inspect these packages you get these results:
```
$ apt show libsbigudrv2
Package: libsbigudrv2
Version: 2.0.0-0ubuntu1
Priority: extra
Section: multiverse/libs
Source: sbig
Origin: Ubuntu
Maintainer: Ubuntu MOTU Developers <ubuntu-motu@lists.ubuntu.com>
Original-Maintainer: Jasem Mutlaq <mutlaqja@ikarustech.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 319 kB
Depends: libc6 (>= 2.4), libusb-0.1-4 (>= 2:0.1.12), fxload
Download-Size: 91,9 kB
APT-Sources: http://pl.archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages
Description: Santa Barbra Instrument Group (SBIG) Universal Library Driver
 The Official Santa Barbra Instrument Group (SBIG) Universal Library
 Driver is used by applications to access and control the USB,
 Parallel Port, and Ethernet versions of the CCD cameras and filter
 wheels including SBIG ST-7/8/9/10/2K cameras, ST-L cameras, and
 ST-402/1603/3200 cameras. The firmware is included for all classes of
 cameras.
```
```
$ apt show libsbigudrv2-dev
Package: libsbigudrv2-dev
Version: 2.0.0-0ubuntu1
Priority: extra
Section: multiverse/libdevel
Source: sbig
Origin: Ubuntu
Maintainer: Ubuntu MOTU Developers <ubuntu-motu@lists.ubuntu.com>
Original-Maintainer: Jasem Mutlaq <mutlaqja@ikarustech.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 53,2 kB
Depends: libsbigudrv2 (= 2.0.0-0ubuntu1)
Conflicts: libsbigudrv-dev
Replaces: libsbigudrv-dev
Download-Size: 8 066 B
APT-Sources: http://pl.archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages
Description: Santa Barbra Instrument Group (SBIG) Universal Library Driver
 The Official Santa Barbra Instrument Group (SBIG) Universal Library
 Driver is used by applications to access and control the USB,
 Parallel Port, and Ethernet versions of the CCD cameras and filter
 wheels including SBIG ST-7/8/9/10/2K cameras, ST-L cameras, and
 ST-402/1603/3200 cameras. The firmware is included for all classes of
 cameras.
 .
 This package contains the development files.
```
So these are obsolete packages still on your repository. Currently replaced by libsbig. libsbig debian/control file says:
```
Conflicts: libsbigudrv2
Replaces: libsbigudrv2
```
I bet that libsbigudrv2 is installed automatically on your side at compilation time, as debuild check build dependencies. Even though it is obsolete now it provides missing libsbigudrv.so file. As the result libsbig builds without issues on your side.

Please check if you have libsbigudrv2 installed on your system and if possible try compiling libsbig without it.

closed 2021-09-28 by knro

rkaczorek, 2021-09-29:
Can be related to #24