indilib/indi-3rdparty #450: celestron_aux alignment accuracy

luzik, 2021-09-05:
Alignment accuracy depends on first step. Before syncing any point using plate solving user is obligated to setup mount to the North leveled with alt=0. Sometimes it is hard to do this accurately. Any other software I used, do not require such step.

To reproduce, make an error in leveling with +20 deg in ALT.

1. Reboot software and mount, let it go with fresh run.
2. Set physically mount to point 20deg Alt, 0 deg Az North
3. Using Kstars mount control sync it to that point (change to az -> enter values AZ:0 and AL:20 -> press sync)
4. Using Kstars mount control only left arrow move the mount 90 deg left to AZ 270 West
5. notice that mount physically is pointing AZ:270 ALT:20, Kstars report it as AZ:270 ALT:0
6. Using Kstars mount control sync to point where mount is physically pointing (enter values AZ:270 AL:20 -> press sync)
7. Using Kstars mount control goto back to AZ: 0 North ALT:20
8. this is looking all right
9. Using Kstars mount control goto AZ:90 East ALT:20

boooom!!!! My mount physically going to AZ:90 ALT:60 !! 

Please rewrite alignment code to not base on this initial position requirements.

New celestron_aux users switching from Celestron Hand Controller, SkySafari, SkyPortal and other software like me have to be confused while trying use this driver just trying to align to first star on sky like they use to do

@jochym please go back to Your child :)

knro, 2021-09-05:
Recently, I added "Nearest Alignment" plugin. Please try this plugin and report back if it fares any better.

jochym, 2021-09-07:
@luzik I did not abandon this project. I am just very busy, so work is going slowly...
I just ported the driver to the 1.9.1 release (PR coming up) and run a test session in "slew to target" and "sync" modes. It was not perfect, but good enough to run 50x30s imaging job with decent results. @knro is the new plugin in 1.9.1 or only in recent commits?

knro, 2021-09-07:
@jochym latest in GIT master, not released yet.

jochym, 2021-09-07:
Ok. Will test.

jochym, 2021-09-28:
I have tested under the sky, and it worked ok. I could not properly reproduce the test case. I am working (slowly) on a better simulator for the scope. So no fundamental updates yet. Keep the issue open - it may be valid and may need fixing.

closed 2021-09-28 by knro