indilib/indi-3rdparty #453: indi_toupcam_ccd: Can't connect with ATR3CMOS26000KMA

hangpark, 2021-09-09:
**Describe the bug**

With both the bleeding edge version and the latest released version of `indi-toupbase` (and `libtoupcam`),
my [ATR3CMOS26000KMA](http://www.touptek.com/product/showproduct.php?lang=en&id=298) can't connect to indi server,
with an error log: https://github.com/indilib/indi-3rdparty/blob/master/indi-toupbase/indi_toupbase.cpp#L115

Below is my USB log:

[image]

And I found that https://github.com/indilib/indi-3rdparty/blob/master/indi-toupbase/oem_cameras.cpp#L638 does not contain ATR3CMOS26000KMA. Is this why my camera couldn't connect?

[image]

@knro could you please address this problem? thank you :)

**Desktop**
 - OS: RPi4/Ubuntu20.04

knro, 2021-09-09:
@bgilsrud Does OEM cameras inclusion affect in anyway the detection of regular Touptek cameras? From Loader class it doesn't appear to affect native touptek detection.

hangpark, 2021-09-09:
@knro actually, this problem also happened before OEM supports were included

bgilsrud, 2021-09-09:
@knro the intent was that OEM camera detection would not affect regular Touptek camera detection. However, I don't have a Touptek camera so wasn't able to test that very rigorously.

bgilsrud, 2021-09-09:
@hangpark I thought that RisingCam was supported by the libnncam library? Have you tried the indi_nncam_ccd driver?

hangpark, 2021-09-09:
@bgilsrud still doesn't work.. (`No Toupcam detected. Power on?`)

bgilsrud, 2021-09-09:
@hangpark are you certain that you have the latest toupcam libraries installed? The library that @knro added with his commit on 6/15 appears to support your camera (using the indi_toupcam_ccd driver). I can check the usb IDs later to make sure they match your device, but just running strings is enough to suggest it should work:
`bgilsrud@lnxdev:~/indi-3rdparty/libtoupcam/arm64$ strings libtoupcam.bin | grep ATR3CMOS26000KMA
7PLHlperI17CATR3CMOS26000KMAE
17CATR3CMOS26000KMA
11ToupcamImplI17CATR3CMOS26000KMA5eSizeE
20ImplATR3CMOS26000KMA
ATR3CMOS26000KMA
ATR3CMOS26000KMA(USB2.0)`

hangpark, 2021-09-10:
@bgilsrud thanks for your help 😃 

I think my camera is included in the library, but can't be detected on.

[image]

Please let me know how can I get information about the connection issue.

bgilsrud, 2021-09-10:
You'll want to check the library that indi_toupcam_ccd is linked against (rather than the library from the git repo). It's probably in /usr/lib/aarch* somewhere. You can find it by running "ldd $(which indi_toupcam_ccd) | grep libtoup"

Another idea would be to verify that you have permission to access the device. You can do this by doing "ls -alh /dev/bus/usb/${BUS}/ and looking for the address of your device. In the lsusb example that you posted, the BUS would be 001 and the address would be 004. I don't know why the permissions wouldn't be setup correctly, but it's easy to check.

hangpark, 2021-09-10:
@bgilsrud I think everything is well configured..

[image]

hangpark, 2021-09-10:
hmm... I found that my `libtoupcam` library's `*.so` file is located at `/usr/lib/*` but `ldd` targets `/lib/*`. Could this be a root cause?

[image]

(but both `/usr/lib/*.so` and `/lib/*.so` contain my camera as well 😢)

[image]

hangpark, 2021-09-13:
@bgilsrud any further approaches...? 😿

bgilsrud, 2021-09-14:
Does Touptek provide their own imaging client? If so, it might be worth trying your camera with that.

bgilsrud, 2021-09-17:
I extracted the camera table from the 1.49.1 version of libtoupcam and I can see that the ATR3CMOS26000KMA camera is there, but it shows a different USB PID. This explains why your camera is not found:
`[TOUPCAM_MODEL_ATR3CMOS26000KMA      ] = { 0x12bf, { "ATR3CMOS26000KMA            ", 0x00000180876b84d9, 2, 3, 0, 1, 0, 3.760000, 3.760000, {{6224, 4168},{3104, 2084},{2064, 1386},}}},
[TOUPCAM_MODEL_ATR3CMOS26000KMA_USB2 ] = { 0x12be, { "ATR3CMOS26000KMA(USB2.0)    ", 0x00000180876b8599, 2, 3, 0, 1, 0, 3.760000, 3.760000, {{6224, 4168},{3104, 2084},{2064, 1386},}}},`

My guess is that the Touptek library just doesn't support your camera yet. I would suggest that you contact Touptek (or Risingcam or whatever brand the camera was sold under) and ask for a library that supports this camera.

bgilsrud, 2021-09-20:
We could use the toupcam OEM support functionality to add support for this camera, but the code will need to be improved so that it doesn't interfere with devices that are supported by libtoupcam. I'll do this in the next day or so.

hangpark, 2021-09-22:
@bgilsrud Thanks a lot! I'll try with your fix. Please let me know if you need any extra information about my camera.

bgilsrud, 2021-09-24:
@hangpark I've pushed some changes to https://github.com/bgilsrud/indi-3rdparty.git in the ATR3CMOS26000KMA branch. Would you be able to test them? Also, it would be helpful if you could provide the output of lsusb when the camera is plugged into a USB 2.0 port since the PID will be different (probably either 0x12c1 or 0x12bf) than it is when the camera is plugged into a USB 3.0 port.

jens-hennig, 2021-09-24:
I have tested my ATR3CMOS26000KMA on a Rasperry Pi 4. The camera reports PID 0x12c0 when pugged into a USB 2.0 port and 0x12bf when connecting to a USB 3.0 port. For testing purposes I have added the ATR3CMOS26000KMA with PID 0x12bf to oem_cameras.cpp with the same parameters than the ATR3CMOS26000KPA and this worked.

bgilsrud, 2021-09-24:
@hangpark my changes won't work as-is since I have pid 12c0 mapped as the USB3 pid. I'll fix them later today.

bgilsrud, 2021-09-25:
@hangpark I pushed some changes that I think should work.

@jens-hennig, had you tried the latest touptek library (1.49.1) with the camera plugged into a USB 3.0 port? I would expect that should work since the 0x12bf pid is supported by that library. It's interesting that library shows 0x12be for the touptek ATR3CMOS26000KMA in USB 2.0 mode. Maybe the difference is that your cameras are sold under the RisingCam brand? It would be too pid if the pids weren't unique across the different OEMs.

hangpark, 2021-09-25:
@bgilsrud i'm on the way to testing! it's very first time to build & install from the source code, sonce i've installed via apt. does following instruction on the below doc for indi-toupcam work and override my pre-installed apt version well? or should uninstall apt driver before running cmake & make install?

https://github.com/indilib/indi-3rdparty#building-individual-3rd-party-drivers-typically-recommended

jens-hennig, 2021-09-25:
I can confirm, that I have tested with library version 1.49.1. On the USB 3.0 it shows 0x12bf.  I am not shure if I have a Touptek, Rinsingcam or something else... They all look the same, I think? There is also an Austrian Vendor that sells this camera under his brand "LACERTA". The label on the package is attached..
[image]
.

jens-hennig, 2021-09-25:
@hangpark  The compilation procedure that you mentioned worked fine for me. The driver was overwritten. You may face some problems if you have an older version of the indi library. In this case you need to download, compile and install the indi library too. 
[https://github.com/indilib/indi](url)

bgilsrud, 2021-09-26:
@hangpark the only difference is that you'll have to checkout my branch:
git clone https://github.com/bgilsrud/indi-3rdparty.git
cd indi-3rdparty
git checkout ATR3CMOS26000KMA

hangpark, 2021-09-26:
@bgilsrud i finally connected my camera successfully! but, it fails capturing.. :(

The error log shows "Camera timed out multiple times. Exposure failed". @jens-hennig does your camera capture well?

jens-hennig, 2021-09-26:
Yes, I can capture images. but I have problems when I use a longer (>1m) USB cables. The camera connects but the image download fails. I have orderd an active USB cable. I hope this gives me a bit more length. There is also a setting for the USB speed. This has to be set to zero.

hangpark, 2021-09-26:
@jens-hennig I can't find a setting for USB speed. Can you share your indi control panel settings?

jens-hennig, 2021-09-26:
@hangpark  I can be found under "CONTROLS" The parameter is simply named "speed"

hangpark, 2021-09-26:
@jens-hennig Oh, it is set to zero already.

@bgilsrud My camera connected successfully, and I can adjust settings well, such as turn on/off fan and cooler, but capturing images always fails with timeout. Can my USB cable cause this problem? hmmm....

jens-hennig, 2021-09-26:
@hangpark : I have just tested it. 
1 m cable works. 
1 m cable with 1m extension failed. 
1 m cable with a USB 3.0 Hub connected by 1m cable works. This will be my final configuration. 

Do you use USB 3.0 or 2.0. I have only tested 3.0 on a Rasperry Pi 4

bgilsrud, 2021-09-26:
Its possible that i messed up the change and didnt associate the pid with the correct camera. @jens-hennig can you share the change that you made?

jens-hennig, 2021-09-26:
@bgilsrud  I don't have much experience with github... so i attach my changes to this post. I want to learn how to use use git correctly... I think the "correct" way of sharing would be: I clone the repository to my home folder on github... ,ake a branch for testing.  Commit the chamnges... make a Pull request...  am I right? So please apologize for this
[oem_cameras_cpp.txt](https://github.com/indilib/indi-3rdparty/files/7231523/oem_cameras_cpp.txt)

I did it in a quite naive way... just adding the PID with the same parameters that the similar to the colour cam.  
I have tested the following points up to now: 
- connect disconnet OK
- capture images with durations from 0.1 to 40 seconds OK
- fan on/off OK
- Cooler on/off OK
- Cooler control seems to be OK, Temperature constant within 2 or 3 deg.

I'll continue testing. Please let me know when you need specific tests.

bgilsrud, 2021-09-26:
@jens-hennig yes, that's pretty much it for the github process. The process of copying the repo into your account is called forking. Everything else is spot on.

I looked at the changes and can't see how they would have any impact. The toupcam_models array is just a table of cameras that were dumped from a previous version of libtoupcam (SDK v48.18042.2020.1128). We use this table since we need to associate OEM (non-touptek branded cameras) cameras with a camera that's natively supported by libtoupcam in order to make them work. We make this association in the oem_cameras array. You can see a bunch of Meade cameras there, which is how I was able to get my Meade-branded toupcams to work.

I'm pretty sure that the x86-64 version of libtoupcam 1.49.1 supports this camera. In fact, I dumped the supported camera table from 1.49.1 and added them to the toupcam_models table for my change (https://github.com/bgilsrud/indi-3rdparty/blob/ATR3CMOS26000KMA/indi-toupbase/oem_cameras.cpp#L1238). You can see the association that I made in the oem_cameras table at: https://github.com/bgilsrud/indi-3rdparty/blob/ATR3CMOS26000KMA/indi-toupbase/oem_cameras.cpp#L1238. Of course, it doesn't make much sense to have an oem association with the same PID as the libtoupcam supported camera...but it does make some sense for the camera in USB 2.0 mode since the PID you reported is different from the USB 2.0 pid in the libtoupcam library.

The fact that @hangpark was able to connect to the camera with my changes, tells me that my changes are at least working but it's likely the association is wrong or maybe that the associated camera isn't present in his library. 

Based on the thread at https://indilib.org/forum/ccds-dslrs/9117-the-very-new-touptek-s-mono-camera-atr3cmos26000kma-doesn-t-work-at-kstars-indi.html, I suspect these cameras are RisingCam branded cameras and are supported in libnncam. It might be worthwhile to get the latest SDK from RisingCam and pull that into the repo.

jens-hennig, 2021-09-26:
@bgilsrud  You are right... it seems that my camera evaluated by the indi driver function FP(EnumV2(pCameraInfo)); and not by OEMCamEnum(&pCameraInfo[iConnectedCamerasCount], iCamInfosLeft);   I had a lot of problems with the USB connection. Maybe that it coincidentally worked after I made this uncomplete change. Sorry for the confusion.

hangpark, 2021-09-27:
@bgilsrud actually I tested with usb2.0. it might work if I do with usb3.0. I'll test in few days...!

hangpark, 2021-10-01:
@bgilsrud @jens-hennig I finally could capture an image after change USB 2.0 to 3.0!!

closed 2021-11-24 by knro