indilib/indi-3rdparty #458: Meade DSI IV Color Exposure failed with indi_toupcam_ccd

noldi20, 2021-09-16:
Meade DSI IV Color is connected to StellarMate (1.6.0) and controlled with Kstars and driver Indi_toupcam_ccd V0.7. All settings can be done and read out. 

Reproduced error:
1. Ekos Camera: Exposure 1.0 sec, rest on default
2. Add job to sequence "+"
3. Start Sequence
4. Exposure starts with countdown
Error: Camera timed out multiple time. Exposure failed

Loggfile:
...
[2021-09-15T19:31:58.718 CEST INFO ][   org.kde.kstars.ekos.capture] - "Job requires 1.000-second  images, has 0/1 frames captured and will be processed."
[2021-09-15T19:31:58.729 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1.000-second  image..."
[2021-09-15T19:31:58.765 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Starting exposure: 1000000 us @ 4640 x 3506 "
[2021-09-15T19:31:58.766 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Event 0X01 "
[2021-09-15T19:32:05.771 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Capture timed out, restarting exposure... "
[2021-09-15T19:32:12.776 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Capture timed out, restarting exposure... "
[2021-09-15T19:32:19.774 CEST INFO ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color :  "[ERROR] Camera timed out multiple times. Exposure failed. "
[2021-09-15T19:32:19.784 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture failed. Check INDI Control Panel for details."
[2021-09-15T19:32:19.796 CEST INFO ][   org.kde.kstars.ekos.capture] - "Restarting capture attempt #1"
[2021-09-15T19:32:19.807 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1.000-second  image..."
[2021-09-15T19:32:19.851 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Starting exposure: 1000000 us @ 4640 x 3506 "
[2021-09-15T19:32:26.860 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Capture timed out, restarting exposure... "
[2021-09-15T19:32:33.841 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Capture timed out, restarting exposure... "
[2021-09-15T19:32:40.842 CEST INFO ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color :  "[ERROR] Camera timed out multiple times. Exposure failed. "
[2021-09-15T19:32:40.851 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture failed. Check INDI Control Panel for details."
[2021-09-15T19:32:40.860 CEST INFO ][   org.kde.kstars.ekos.capture] - "Restarting capture attempt #2"
[2021-09-15T19:32:40.870 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1.000-second  image..."
[2021-09-15T19:32:40.943 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Starting exposure: 1000000 us @ 4640 x 3506 "
[2021-09-15T19:32:47.940 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Capture timed out, restarting exposure... "
[2021-09-15T19:32:54.944 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Capture timed out, restarting exposure... "
[2021-09-15T19:33:01.946 CEST INFO ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color :  "[ERROR] Camera timed out multiple times. Exposure failed. "
[2021-09-15T19:33:01.957 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capture failed. Check INDI Control Panel for details."
[2021-09-15T19:33:01.967 CEST INFO ][   org.kde.kstars.ekos.capture] - "CCD capture aborted"
[2021-09-15T19:33:01.977 CEST DEBG ][   org.kde.kstars.ekos.capture] - setMeridianFlipStage:  "MF_READY"
[2021-09-15T19:33:02.088 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Event 0X01 "
[2021-09-15T19:33:02.088 CEST DEBG ][           org.kde.kstars.indi] - Toupcam Meade DSI IV Color : "[DEBUG] Event 0X01 "

knro, 2021-09-16:
@bgilsrud is this camera supported by the generic oem toupcam driver?

bgilsrud, 2021-09-16:
Yes, it should be. It looks like the camera was detected and connected correctly. @noldi20 can you provide the USB device ID for the camera? Running lsusb while the camera is connected will have the info I'm looking for.

noldi20, 2021-09-16:
here is the usb device ID:

tellarmate@stellarmate:~ $ lsusb
Bus 002 Device 002: ID 0547:e077 Anchor Chips, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

bgilsrud, 2021-09-16:
Thanks. The driver incorrectly thinks that camera doesn't have a DDR buffer. That might explain why image capture doesn't work. I'll submit a PR in a little bit.

bgilsrud, 2021-09-17:
@noldi20 the change has been merged. I'm not sure how/when this will be available in Stellarmate, but you could manually build it to see if it fixes your problem.

noldi20, 2021-09-17:
@bgilsrud great thanks a lot for your quick fix! I will test it.

noldi20, 2021-09-20:
> @noldi20 the change has been merged. I'm not sure how/when this will be available in Stellarmate, but you could manually build it to see if it fixes your problem.

@bgilsrud it's working now and already deployed with StellarMate. many thanks for the fast fix.

closed 2021-09-22 by knro