indilib/indi-3rdparty #463: Exposure timeouts with Toupcam USB2 cameras

bgilsrud, 2021-09-27:
I'm seeing exposure timeouts when autoguiding using a Meade LPI-GM camera. When the timeouts occur, it can take around 30s before a new exposure is requested. This extended delay will cause the guide accuracy to drop.

It would be nice if these timeouts could be avoided, but the next best thing would be to reduce the time it takes to recognize that the exposure isn't going to complete and start a new exposure.

bgilsrud, 2021-09-27:
I'm happy to work on this since I can easily reproduce the issue.

knro, 2021-09-27:
Sure go ahead, I've tried to deal with timeouts before as you can see in the code.

knro, 2021-11-24:
@bgilsrud I don't recall if this was already fixed or not?

wlatanowicz, 2022-05-06:
`m_DownloadEstimation` can become negative. This PR fixes the issue: https://github.com/indilib/indi-3rdparty/pull/576

closed 2022-09-21 by knro