indilib/indi-3rdparty #464: indi-gige does not compile with current version of Aravis

Jul3k, 2021-09-27:
The current implementation of  indi-gige requires to be compiled with Aravis < 0.8.0. as function calls were changed to include an [error parameter](https://github.com/AravisProject/aravis/commit/36d23d9a3f3e6e4118e3f8e80c5bfed000632d0d). I do not know how mature this driver is. I was trying to get a USB3 Vision camera working which is supported by Aravis.

**Reproduce**
1. Install Aravis 0.8.18
2. Follow build instructions
3. Compile error:
```
    indi-3rdparty-1.9.1/indi-gige/src/ArvGeneric.cpp:198:53: error: too few arguments to function ‘const char* arv_camera_get_device_id(ArvCamera*, GError**)’
    this->cam.device_id   = arv_camera_get_device_id(camera);
                            ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
```

**Expected behavior**
Should compile with current version of Aravis.

**Desktop:**
 - Arch Linux

knro, 2021-09-28:
I don't think driver is currently well maintained. Can you please submit a patch to fix the issue?

Jul3k, 2021-09-28:
The driver compiles correctly with the submitted patch. It does however not show in Ekos because it says no camera is connected:
`"Unable to establish:\n+ GigE CCD\nPlease ensure the device is connected and powered on."`
This is strange because I can see in the Debugger, that a camera is found and initialized. Do you know where this error message comes from? I cannot find anything similar in the source code.

knro, 2021-09-28:
You can safely ignore this if the devices indeed shows up in Ekos. What's the name of the device that shows up exactly?

Jul3k, 2021-09-28:
The problem is that no device tab shows up in the Indi Control Panel. I can select a `GigE CCD` in the profile editor, but when I start the server, no new device tab is shown in the indi control panel. The log shows the following:

`[2021-09-28T14:30:14.801 CEST DEBG ][           org.kde.kstars.indi] - Received new device U3-380xCP-C
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - < U3-380xCP-C >: < CONNECTION >
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - < U3-380xCP-C >: < DRIVER_INFO >
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - < U3-380xCP-C >: < POLLING_PERIOD >
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - < U3-380xCP-C >: < CONFIG_PROCESS >
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - < U3-380xCP-C >: < DEBUG >
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - < U3-380xCP-C >: < ACTIVE_DEVICES >
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: Found Camera: U3-380xCP-C"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on Telescope Simulator.TELESCOPE_INFO"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on Focuser Simulator.ABS_FOCUS_POSITION"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on Focuser Simulator.FOCUS_TEMPERATURE"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on CCD Simulator.FILTER_SLOT"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on CCD Simulator.FILTER_NAME"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-09-28T12:30:14: Driver indi_gige_ccd: snooping on SQM.SKY_QUALITY"
[2021-09-28T14:30:14.806 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2021-09-28T14:30:24.242 CEST INFO ][           org.kde.kstars.ekos] - "Unable to establish:\n+ GigE CCD\nPlease ensure the device is connected and powered on."`

knro, 2021-09-29:
I think you need to patch the driver so that getDefaultName always return "GigE CCD", and in the constructor perhaps try to rename the driver to something like "GigE CCD %s" where %s is camera->name so that the final name is "GigE CCD U3-380xCP-C". You can also drop the whole "CCD" part if you like, it's old tradition but no longer applicable today.

Let me know if you need help in this.

Jul3k, 2021-09-29:
Thank you. That solves the problem. The driver is now displayed in the Control Tab. It crashes however when a frame is captured and the properties are not correctly read from the camera. There seem to be more problems with this driver. As far for the compilation error this can be closed.

closed 2021-09-29 by Jul3k