indilib/indi-3rdparty #481: std::abs used instead of std::fabs

sgromoll, 2021-10-30:
**Describe the bug**
In `asi_base.cpp`, `std::abs` is used in `ASIBase::SetTemperature` to compare the current temperature difference to a floating point value like `TEMP_THRESHOLD` (0.25).  `std::abs` takes (and returns) and int/long, so presumably `std::fabs` should be used here since you are comparing to a float.

This occurs in several places in the file.

**To Reproduce**
N/A

**Expected behavior**
N/A

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2021-10-30:
it actually returns double. Check the definition in std_abs.h

```
#ifndef __CORRECT_ISO_CPP_MATH_H_PROTO
  inline _GLIBCXX_CONSTEXPR double
  abs(double __x)
  { return __builtin_fabs(__x); }

  inline _GLIBCXX_CONSTEXPR float
  abs(float __x)
  { return __builtin_fabsf(__x); }

  inline _GLIBCXX_CONSTEXPR long double
  abs(long double __x)
  { return __builtin_fabsl(__x); }
#endif
```

sgromoll, 2021-11-05:
Yep, thanks for clarification.

closed 2021-11-05 by sgromoll