indilib/indi-3rdparty #484: Can't load QHY firmware QHY600.img

treaves, 2021-11-14:
**Describe the bug**
I'd installed Astroberry several months ago.  Everything in my rig was seen and worked.  Until I ran apt upgrade.  This originally broke several things.  In the subsequent runs, most have started working again, except my QHY camera.  I am now building everything from source, and I have the same issue.

I can see my attached QHY camera with **lsusb**: 
`Bus 001 Device 013: ID 1618:c600 Cypress WestBridge`

However, qhy_ccd_test fails to see it:
```
QHY Test CCD using SingleFrameMode, Version: 1.00
-- qhyccd.cpp param
QHYCCD|QHYCCD.CPP|InitQHYCCDResource()|START
QHYCCD|QHYCCD.CPP|InitQHYCCDResource|auto_detect_camera:false,call InitQHYCCDResourceInside
QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|START
QHYCCD|QHYCCD.CPP|libusb_version 1.0.24.11584
QHYCCD|QHYCCD.CPP|libusb_init(libqhyccd_context) called...
QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|numdev set to 0
QHYCCD|QHYCCD.CPP|InitQHYCCDResourceInside|END
************************** config file path  21.10.26.22 svn: 11577  ************************************
QHYCCD|QHYCCD.CPP|InitQHYCCDResource|Load ini filePath = /usr/lib/firmware/qhy  fileName = qhyccd.ini
SDK resources initialized.
No QHYCCD camera found, please check USB or power.
```

In trying to figure out what was going on, I found the following thanks to **journalctl**:
```
Nov 14 12:05:42 astropi kernel: usb 1-1.1.1.3: USB disconnect, device number 12
Nov 14 12:05:46 astropi kernel: usb 1-1.1.1.3: new high-speed USB device number 13 using xhci_hcd
Nov 14 12:05:46 astropi kernel: usb 1-1.1.1.3: New USB device found, idVendor=1618, idProduct=c600, bcdDevice= 1.00
Nov 14 12:05:46 astropi kernel: usb 1-1.1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 14 12:05:46 astropi kernel: usb 1-1.1.1.3: Product: WestBridge
Nov 14 12:05:46 astropi kernel: usb 1-1.1.1.3: Manufacturer: Cypress
Nov 14 12:05:46 astropi kernel: usb 1-1.1.1.3: SerialNumber: 0000000004BE
Nov 14 12:05:46 astropi mtp-probe[1794]: checking bus 1, device 13: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1.1/1-1.1.1.3"
Nov 14 12:05:46 astropi mtp-probe[1794]: bus: 1, device: 13 was not an MTP device
Nov 14 12:05:46 astropi systemd-udevd[1793]: 1-1.1.1.3: Process '/sbin/fxload -t fx3 -I /lib/firmware/qhy/QHY600.img -D /dev/bus/usb/001/013' failed with exit code 255.
Nov 14 12:05:46 astropi systemd-udevd[1793]: 1-1.1.1.3:1.0: Process '/sbin/fxload -t fx3 -I /lib/firmware/qhy/QHY600.img -D ' failed with exit code 255.
Nov 14 12:05:46 astropi mtp-probe[1801]: checking bus 1, device 13: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1.1/1-1.1.1.3"
Nov 14 12:05:46 astropi mtp-probe[1801]: bus: 1, device: 13 was not an MTP device

```

**To Reproduce**
Exact steps to reproduce the behavior.
1. plug QHY camera in
2. run journalctl
3. See error

**Expected behavior**
The firmware successfully loaded

**Desktop (please complete the following information):**
 - OS: Raspbian / Raspberry Pi OS
 - Version: both Buster and Bullseye

treaves, 2021-11-14:
Well, some more digging has lead me to [this](https://github.com/qhyccd-lzr/QHYCCD_Linux/issues/29) report of it being an **fxload** issue.  The linked version does work (meaning both Ekos & qhy_ccd_test work).  Not sure what to do from here to get this properly resolved.

paolostivanin, 2021-11-15:
In order to upload a firmware on recent astro cameras (usb3 devices), the distro needs fxload version `2013_01_03`.
Most distros ship with `2008_10_13`, a version that does not support those new devices.
The codebase of the new fxload is the same as the old fxload but with some small changes made by Steven J. Magnani that add support for Cypress FX3 USB 3.0 devices.

TL;DR: you have 3 options:
1. switch to openSUSE Leap 15.3 and/or Tumbleweed which has this new package
2. compile and install the new fxload yourself
3. ask your distro's maintainers to update the package

sonny486, 2021-11-17:
I have seen this happen many times when I am using a USB3 hub that does not communicate with the QHY product properly, OR the USB3 cable is too long.

I would make sure the power to the camera and the USB3 hub also has the proper power to link properly.

If you can watch your raspberry PI and see when you plug it in if you get the FXload error with tail -f /var/log/syslog

knro, 2021-11-24:
Closing as this is a downstream issue.

closed 2021-11-24 by knro