indilib/indi-3rdparty #487: Minimal exposure time is 1/10 for Sony DSLR in INDI and in EKOS (astroberry 2.0.4 setup)

ilkanaev, 2021-11-17:
**Describe the bug**

When using INDI/EKOS, there is no any option to use exposure lower than 1/10 sec.
i am running with **astroberry 2.0.4 on raspbery pi 4 (4Gb)**.
i have **Sony a6500 DSLR** as my main camera. The camera set to Manual -> BLUB mode.
The camera connected to D-LINK USB3 hub (with few other devices), and the hub connected to the USB3 port of rasbery.

**To Reproduce**
If i swith the camera to any but not BLUB exposure, the exposure defined directly on my camera will be used, and no any way to change it from INDI/EKOS.

If i switch the camera to BLUB, I can enter/select any vaues for the exposure in INDI [DSLR tab]->[Main Control]->[Presets or Expose], but the actual exposure time will not go down than 1/10.

If i disable the BLUB mode in INDI, and select any exposure, it looks like during taking the picture the Sony DSLR driver used by INDI is crashing and the software loosing connection to the camera (perhaps camera doing reset, i was not able to understand that ) . The only way to recover from this situation is by clicking on 'restart driver' option next to the camera name in EKOS.

i saw several posts on this (mostly dealing with Flat frames, and enabling/disabling BLUB settings in INDI), but noone of these helped me to get the actual exposure time controled by INDI and lower than 1/10.

**Expected behavior**
When i am using gphoto2 (isntalled directly on the same setup), the exposure on my camera is fully controlled by the gphoto2 . The gphoto2  actualy updating the exposure on the camera itself and waiting for the image to be received.
Even if the camera set to BLUB mode, the gphoto2 will update this setting to the right exposure value.

In case of INDI, looks like the only option to control the exposure is using BLUB mode.  And the actual minimal exposure is alwasy limited by 1/10 sec. I can enter/select any value lower than 1/10 but the actual exposure will not be less than 1/10.

**Desktop (please complete the following information):**
i am running with **astroberry 2.0.4 on raspbery pi 4 (4Gb)**.
i have **Sony a6500 DSLR** as my main camera. The camera set to Manual -> BLUB mode.
The camera connected to D-LINK USB3 hub (with few other devices), and the hub connected to the USB3 port of rasbery.

rkaczorek, 2021-11-18:
Please provide output from `gphoto2 --version`

ilkanaev, 2021-11-19:
astroberry@astroberry:~ $ gphoto2 --version
gphoto2 2.5.20

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
astroberry@astroberry:~ $

rkaczorek, 2021-11-20:
This a standard libgphoto version so it must be indi-gphoto driver indeed.

knro, 2021-11-20:
Driver log?

ilkanaev, 2021-11-20:
Hi,
Please trell me how to get the log you need and i will provide it.
Thanks,
Yossi

ilkanaev, 2021-11-22:
Hi @knro ,
I would really appreciate your help here, how to get the logs you need?

Thank you in advance,
Yossi

knro, 2021-11-22:
Please check here: https://indilib.org/individuals/logs-howto.html

ilkanaev, 2021-11-22:
hi
please find enclosed the logs
at the begining the camera was in BLUB mode.
i took one image with 1sec exposure
then multiple images with multiple exposures using INDI controller and using EKOS
then set of three images with different exposure from EKOS controller 
then moved the camera out of blub, to 1/2000 sec
and took set of three images as before
[log_21-42-48.txt.gz](https://github.com/indilib/indi-3rdparty/files/7583985/log_21-42-48.txt.gz)

knro, 2021-11-23:
I see all taken with bulb exposure mode so this is why you can't get very low. If you disable bulb exposure in INDI --> Options then  you said it crashes? I need to see log of what happens and preferably a backtrace of the crash.

ilkanaev, 2021-11-23:
Hi,
please find enclosed the new lo
[log_22-02-07.txt.gz](https://github.com/indilib/indi-3rdparty/files/7588713/log_22-02-07.txt.gz)
g
the scenario is enable logs

1. connect
2. change Force BLUB to disabled
3. in EKOS take a picture with 0.0001 sec exposure

I did it few times, including one time witth 1sec exposure before 0.0001sec exposure (crashed as well).

Please also note - i found some workeround for this issue that sometimes work -
If in step '3' above, before taking the pictures in EKOS, you will take multiple pictures with different exposures directly in INDI Control pannel, sometimes it will start working in EKOS as well...

ilkanaev, 2021-11-25:
When Force BLUB disabled in INDI, I saw the crash happening on various exposures and not only on these lower than 1sec. so it is not related to the pecific exposures....

knro, 2021-12-03:
I tested with Canon 600D (Force Bulb enabled) Nikon D3500 (Force Bulb disabled) and I can get 1/1000 and 30+ second image just fine with both cameras. Not sure what's going on with your particular model.

ilkanaev, 2021-12-06:
is there anythyng i can help you to help me :-)?

rkaczorek, 2021-12-07:
@knro the strange thing is that tests performed by @ilkanaev with binary gphoto2 give expected behavior and indi driver, using same libgphoto2 does not reproduce this. See here https://github.com/rkaczorek/astroberry-server/issues/165#issuecomment-971657432

knro, 2021-12-07:
@rkaczorek Maybe set Force bulb to OFF in Options? or that doesn't help?

ilkanaev, 2021-12-07:
That doesnt help

londonc, 2022-04-22:
Can confirm behavior in my A6300 as well when using Native format. Trying to use BULB and setting exposure to 0.00025 seconds will actually result in a 1/10 sec photo. Disabling BULB and using preset 1/4000 in INDI Control Panel also delivers a 1/10 second photo. 

Even tried with higher values like 1/100. Same result. FITS allegedly shows the correct exposure length using presets IF you take a single photo. If you try to schedule multiple you get something like this;

> 2022-04-22T06:46:43: [ERROR] Rapid exposure not possible since upload time is 5.08 seconds while exposure time is 0.00 seconds. 

Hope this helps.

Mas0821, 2024-02-18:
@londonc Find some informations in Sony Camera Remote SDK document, explanation of API ```CrDeviceProperty_ShutterSpeed```.

> The real value of shutter speed (Upper two bytes:  numerator, Lower two bytes: denominator)
> In the case of the shutter speed is displayed as "Real Number" on the camera, the denominator is fixed 0x000A.
> e.g.) 0x000F000A: 0x000F (means 15) / 0x000A (means 10) = 1.5"
> In the case of the shutter speed is displayed as "Fraction Number" on the camera, the numerator is fixed 0x0001.
> e.g.) 0x000103E8: 0x0001 (means 1) / 0x03E8 (means 1000) = 1/1000

Maybe that's why always get 1/10s cus set data=0x0001000A is the minimum.

But still not know how to set or check shutter speed in "Real Number" or "Fraction Number" yet, I'll keep trying.

ps: sorry for English mistakes if it has