indilib/indi-3rdparty #489: Bullseye installation on Raspberry pi 4b fails

dabateman, 2021-11-19:
I was able to successfully install Indi and Phd2 on the recent Bullseye Raspberry pi OS.
However I can't connect to my Lodestar x2c camera in Phd2.  I can see it using lsusd.
When testing if the Sx ccd was there, it wasn't. 
But I can't seem to install these 3rd party drivers. 
I get a bunch of library issues.  

Will Bullseye support be added soon or am I missing something simple. 
Thank you.

knro, 2021-11-19:
What issues with 3rd party install?

dabateman, 2021-11-19:
**This is my error:**

pi@raspberrypi:~ $ cd ~/Projects/build/indi-sx
pi@raspberrypi:~/Projects/build/indi-sx $ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-sx
-- Found CFITSIO: /usr/lib/arm-linux-gnueabihf/libcfitsio.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Projects/indi-3rdparty/indi-sx
pi@raspberrypi:~/Projects/build/indi-sx $ make -j4
[ 40%] Built target sx_ccd_test
[ 40%] Linking CXX executable indi_sx_wheel
[ 40%] Building CXX object CMakeFiles/indi_sx_ao.dir/sxao.cpp.o
[ 40%] Building CXX object CMakeFiles/indi_sx_ccd.dir/sxccd.cpp.o
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::~SXWHEEL()':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:51: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:52: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::Connect()':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:68: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:70: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::Disconnect()':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:85: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:86: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::SendWheelMessage(int, int)':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:106: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:108: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:116: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:121: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:124: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:129: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o:/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:132: more undefined references to `INDI::BaseDevice::getDeviceName()' follow
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISNewText'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISNewSwitch'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISNewBLOB'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISNewNumber'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISGetProperties'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISSnoopDevice'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/indi_sx_wheel.dir/build.make:106: indi_sx_wheel] Error 1
make[1]: *** [CMakeFiles/Makefile2:155: CMakeFiles/indi_sx_wheel.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/pi/Projects/indi-3rdparty/indi-sx/sxao.cpp: In member function ‘virtual bool SXAO::updateProperties()’:
/home/pi/Projects/indi-3rdparty/indi-sx/sxao.cpp:159:9: error: ‘defineProperty’ was not declared in this scope; did you mean ‘deleteProperty’?
  159 |         defineProperty(&GuideNSNP);
      |         ^~~~~~~~~~~~~~
      |         deleteProperty
make[2]: *** [CMakeFiles/indi_sx_ao.dir/build.make:82: CMakeFiles/indi_sx_ao.dir/sxao.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:128: CMakeFiles/indi_sx_ao.dir/all] Error 2
/home/pi/Projects/indi-3rdparty/indi-sx/sxccd.cpp: In member function ‘virtual bool SXCCD::updateProperties()’:
/home/pi/Projects/indi-3rdparty/indi-sx/sxccd.cpp:173:13: error: ‘defineProperty’ was not declared in this scope; did you mean ‘deleteProperty’?
  173 |             defineProperty(&CoolerSP);
      |             ^~~~~~~~~~~~~~
      |             deleteProperty
/home/pi/Projects/indi-3rdparty/indi-sx/sxccd.cpp:175:13: error: ‘defineProperty’ was not declared in this scope; did you mean ‘deleteProperty’?
  175 |             defineProperty(&ShutterSP);
      |             ^~~~~~~~~~~~~~
      |             deleteProperty
make[2]: *** [CMakeFiles/indi_sx_ccd.dir/build.make:82: CMakeFiles/indi_sx_ccd.dir/sxccd.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:182: CMakeFiles/indi_sx_ccd.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
pi@raspberrypi:~/Projects/build/indi-sx $

knro, 2021-11-19:
Looks like you didn't install libindi first from Git?

dabateman, 2021-11-19:
**I think I did.  I installed Indi first and all the libraries from the 3rd party.

I have also tried this with the following error:**
pi@raspberrypi:~ $ sudo apt install indi-full gsc

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 indi-apogee : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-asi : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-atik : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-bin : Depends: libcfitsio7 (>= 3.450~) but it is not installable
            Depends: libgsl23 (>= 2.5) but it is not installable
 indi-celestronaux : Depends: libgsl23 (>= 2.5) but it is not installable
 indi-dsi : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-eqmod : Depends: libgsl23 (>= 2.5) but it is not installable
 indi-ffmv : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-fishcamp : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-fli : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-gphoto : Depends: libcfitsio7 (>= 3.450~) but it is not installable
               Depends: libraw19 (>= 0.19.0) but it is not installable
 indi-gpsd : Depends: libgps23 (>= 3.3) but it is not installable
 indi-mi : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-nightscape : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-pentax : Depends: libcfitsio7 (>= 3.450~) but it is not installable
               Depends: libraw19 (>= 0.16.0) but it is not installable
 indi-playerone : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-qhy : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-qsi : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-sbig : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-sv305 : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 indi-toupbase : Depends: libcfitsio7 (>= 3.450~) but it is not installable
 libindi1 : Depends: libcfitsio7 (>= 3.450~) but it is not installable
            Depends: libgsl23 (>= 2.5) but it is not installable
E: Unable to correct problems, you have held broken packages.

dabateman, 2021-11-19:
Oh I see running it gave errors:

pi@raspberrypi:~ $ sudo apt-get -y install libindi-dev

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libindi1 : Depends: libcfitsio7 (>= 3.450~) but it is not installable
            Depends: libgsl23 (>= 2.5) but it is not installable
            Depends: indi-bin (>= 1.9.2-1) but it is not going to be installed
            Conflicts: libindi-plugins but 1.8.8+dfsg-1 is to be installed
            Conflicts: libindidriver1 but 1.8.8+dfsg-1 is to be installed
E: Unable to correct problems, you have held broken packages.

knro, 2021-11-19:
You need to install libindi from GIT, not sudo apt-get since that version in the Debian repo is older.

dabateman, 2021-11-19:
Where is that located?

dabateman, 2021-11-26:
Still getting errors when trying. I did update to newest libindi-dev

pi@raspberrypi:~ $ cd ~/Projects/build/indi-sx

pi@raspberrypi:~/Projects/build/indi-sx $ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-sx

-- Found CFITSIO: /usr/lib/arm-linux-gnueabihf/libcfitsio.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Projects/indi-3rdparty/indi-sx

pi@raspberrypi:~/Projects/build/indi-sx $ make -j4

[ 40%] Built target sx_ccd_test
[ 40%] Building CXX object CMakeFiles/indi_sx_ao.dir/sxao.cpp.o
[ 40%] Linking CXX executable indi_sx_wheel
[ 40%] Building CXX object CMakeFiles/indi_sx_ccd.dir/sxccd.cpp.o
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::~SXWHEEL()':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:51: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:52: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::Connect()':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:68: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:70: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::Disconnect()':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:85: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:86: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::SendWheelMessage(int, int)':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:106: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:108: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:116: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:121: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:124: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:129: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o:/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:132: more undefined references to `INDI::BaseDevice::getDeviceName()' follow
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISNewText'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISNewSwitch'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISNewBLOB'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISNewNumber'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISGetProperties'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../arm-linux-gnueabihf/libindidriver.so: undefined reference to `ISSnoopDevice'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/indi_sx_wheel.dir/build.make:106: indi_sx_wheel] Error 1
make[1]: *** [CMakeFiles/Makefile2:155: CMakeFiles/indi_sx_wheel.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/pi/Projects/indi-3rdparty/indi-sx/sxao.cpp: In member function ‘virtual bool SXAO::updateProperties()’:
/home/pi/Projects/indi-3rdparty/indi-sx/sxao.cpp:159:9: error: ‘defineProperty’ was not declared in this scope; did you mean ‘deleteProperty’?
  159 |         defineProperty(&GuideNSNP);
      |         ^~~~~~~~~~~~~~
      |         deleteProperty
make[2]: *** [CMakeFiles/indi_sx_ao.dir/build.make:82: CMakeFiles/indi_sx_ao.dir/sxao.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:128: CMakeFiles/indi_sx_ao.dir/all] Error 2
/home/pi/Projects/indi-3rdparty/indi-sx/sxccd.cpp: In member function ‘virtual bool SXCCD::updateProperties()’:
/home/pi/Projects/indi-3rdparty/indi-sx/sxccd.cpp:173:13: error: ‘defineProperty’ was not declared in this scope; did you mean ‘deleteProperty’?
  173 |             defineProperty(&CoolerSP);
      |             ^~~~~~~~~~~~~~
      |             deleteProperty
/home/pi/Projects/indi-3rdparty/indi-sx/sxccd.cpp:175:13: error: ‘defineProperty’ was not declared in this scope; did you mean ‘deleteProperty’?
  175 |             defineProperty(&ShutterSP);
      |             ^~~~~~~~~~~~~~
      |             deleteProperty
make[2]: *** [CMakeFiles/indi_sx_ccd.dir/build.make:82: CMakeFiles/indi_sx_ccd.dir/sxccd.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:182: CMakeFiles/indi_sx_ccd.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
pi@raspberrypi:~/Projects/build/indi-sx $ sudo make install
[ 30%] Built target sx_ccd_test
[ 40%] Building CXX object CMakeFiles/indi_sx_ao.dir/sxao.cpp.o
/home/pi/Projects/indi-3rdparty/indi-sx/sxao.cpp: In member function ‘virtual bool SXAO::updateProperties()’:
/home/pi/Projects/indi-3rdparty/indi-sx/sxao.cpp:159:9: error: ‘defineProperty’ was not declared in this scope; did you mean ‘deleteProperty’?
  159 |         defineProperty(&GuideNSNP);
      |         ^~~~~~~~~~~~~~
      |         deleteProperty
make[2]: *** [CMakeFiles/indi_sx_ao.dir/build.make:82: CMakeFiles/indi_sx_ao.dir/sxao.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:128: CMakeFiles/indi_sx_ao.dir/all] Error 2
make: *** [Makefile:171: all] Error 2

knro, 2021-11-26:
Compile INDI library first: https://github.com/indilib/indi

closed 2021-11-26 by knro