indilib/indi-3rdparty #493: Raspberry pi installation on Bullseye fails!

dabateman, 2021-11-26:
Ok fresh version of Bullseye added to a micro SD card.
Indi sucessfully installs from Github, just as done in the past. But this time did it with full verison, not just with depth 1.

Was able to install all 3rdparty libraries succesfully.

But can't install all 3rdparty drivers.
Also can't install the SX driver for the lodestar X2C camera that I need.

Following was the output:

pi:~/Projects/build/indi-3rdparty $ mkdir -p ~/Projects/build/indi-sx

pi@raspberrypi:~/Projects/build/indi-3rdparty $ cd ~/Projects/build/indi-sx

pi@raspberrypi:~/Projects/build/indi-sx $ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-sx

-- The CXX compiler identification is GNU 10.2.1
-- The C compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found CFITSIO 3.49: /usr/lib/aarch64-linux-gnu/libcfitsio.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Found USB1: /usr/lib/aarch64-linux-gnu/libusb-1.0.so (found version "1.0.24") 
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Found INDI: /usr/lib/aarch64-linux-gnu/libindidriver.so;/usr/lib/aarch64-linux-gnu/libindiAlignmentDriver.so (found version "1.8.8") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Projects/build/indi-sx

pi@raspberrypi:~/Projects/build/indi-sx $ make -j4

Scanning dependencies of target sx_ccd_test
Scanning dependencies of target indi_sx_wheel
Scanning dependencies of target indi_sx_ao
Scanning dependencies of target indi_sx_ccd
[ 10%] Building CXX object CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o
[ 20%] Building CXX object CMakeFiles/indi_sx_ao.dir/sxao.cpp.o
[ 30%] Building CXX object CMakeFiles/sx_ccd_test.dir/sxccdtest.cpp.o
[ 40%] Building CXX object CMakeFiles/indi_sx_ccd.dir/sxccd.cpp.o
[ 50%] Building CXX object CMakeFiles/sx_ccd_test.dir/sxccdusb.cpp.o
/home/pi/Projects/indi-3rdparty/indi-sx/sxao.cpp: In member function ‘virtual bool SXAO::updateProperties()’:
/home/pi/Projects/indi-3rdparty/indi-sx/sxao.cpp:159:9: error: ‘defineProperty’ was not declared in this scope; did you mean ‘deleteProperty’?
  159 |         defineProperty(&GuideNSNP);
      |         ^~~~~~~~~~~~~~
      |         deleteProperty
make[2]: *** [CMakeFiles/indi_sx_ao.dir/build.make:82: CMakeFiles/indi_sx_ao.dir/sxao.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:155: CMakeFiles/indi_sx_ao.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 60%] Building CXX object CMakeFiles/indi_sx_ccd.dir/sxccdusb.cpp.o
[ 70%] Linking CXX executable indi_sx_wheel
[... 35 lines cut ...]
pi@raspberrypi:~/Projects/build/indi-sx $ sudo make install

[ 30%] Built target sx_ccd_test
[ 40%] Linking CXX executable indi_sx_wheel
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::~SXWHEEL()':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:51: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:52: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::Connect()':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:68: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:70: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::Disconnect()':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:85: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:86: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o: in function `SXWHEEL::SendWheelMessage(int, int)':
/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:106: undefined reference to `INDI::DefaultDevice::isSimulation()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:108: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:116: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:121: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:124: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: /home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:129: undefined reference to `INDI::BaseDevice::getDeviceName()'
/usr/bin/ld: CMakeFiles/indi_sx_wheel.dir/sxwheel.cpp.o:/home/pi/Projects/indi-3rdparty/indi-sx/sxwheel.cpp:132: more undefined references to `INDI::BaseDevice::getDeviceName()' follow
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/indi_sx_wheel.dir/build.make:106: indi_sx_wheel] Error 1
make[1]: *** [CMakeFiles/Makefile2:128: CMakeFiles/indi_sx_wheel.dir/all] Error 2
make: *** [Makefile:171: all] Error 2

dabateman, 2021-11-26:
Also this time I have my camera attached to the pi, just incase that makes a difference. Obviously not as still will not install. 
Must still be a driver problem that I can't see.

k-ross, 2021-11-28:
This is looking like a version conflict between indi and indi-3rdparty. Did you install indi via the repo, or did you build indi from source? If you're going to build indi-3rdparty from source, you also need to build indi from source (after removing any indi packages that were installed).

dabateman, 2021-11-28:
I download the version of indi that was currently up here and followed all instructions: 
https://github.com/indilib/indi
Then downloaded 3rd party apps here and built moments later following the instructions. 
And these are the errors. 
Indi installed without errors. 

Today I downloaded and installed Phd2 and it installed without any errors.  But can't see my Lodestar camera.  I definitely need these Sx ccd drivers.

closed 2022-11-16 by knro