indilib/indi-3rdparty #506: Player One Camera Image in indi_playerone_ccd

YasuNishimura, 2021-12-24:
**Describe the bug**
I use a Player One camera(Ceres-C).
However, when I take an image with the camera using EKOS, the image looks strange. Two similar images are lined up side by side.

**To Reproduce**
1. Run driver :  indi_playerone_ccd ver0.2
2. Perform action : The Format setting was "Raw16". I take an image with Player One Camera(Ceres-C).
3. See error : Two similar images are lined up side by side. Please refer to the figure below.

**Expected behavior**
When the setting is ”Raw8”, there is no problem with the image.

**Desktop (please complete the following information):**
 - SW: Kstars(ver. 3.5.6) on RPi4

**Screenshots**
Raw16: not correct
[image]

Raw8: correct
[image]

knro, 2021-12-25:
@hiro3110i Can you please check this issue?

hiro3110i, 2021-12-25:
Sorry for inconvenience, it has already fixed by v0.3 which was included in indi v1.9.3.
Could you please use the latest version of KStars/Ekos?
Or, RAW8 has no problem even in the older version.

Merry Christmas!

YasuNishimura, 2021-12-31:
Hi, Knro-san and hiro3110i-san,

Thank you for your response.
My first bug report was wrong. I used KStars(v3.5.5) on RPi4.
I tried to update it to the latest version(v3.5.6). But I couldn't update it with "apt".
It seems that the RPi's repository does not support v3.5.6.
I will check after the repository supports it.

hiro3110i, 2022-01-03:
Dear YasuNishimura-san and Jasem-san,

Today, I could upgrade KStars to v3.5.6 by apt command on Astroberry v2.0.4.
Thanks for preparing the latest version of KStars on Raspberry Pi.
PlayerOne camera driver was also upgraded to v0.3 (bug fixed version).
Could you please try to apt upgrade on your Raspberry Pi?

My new year greetings was captured by PlayerOne Neptune-C II (16bits) through indi_playerone_ccd v0.3.
[image]

YasuNishimura, 2022-01-04:
Hi, hiro3110i-san and Jasem-san,

I checked the new KStars(v3.5.6) on RPi4. The bug is fixed.
Thank you for your cooperation

The greeting card is good. I also will take pictures like this with the new KStars.

closed 2022-01-04 by YasuNishimura