indilib/indi-3rdparty #517: EQMod Mount crashes when updating mount location.

setnes, 2022-01-03:
Running latest Astroberry updates that include fresh INDI drivers.  I can no longer set my mount's location.

EQMod Mount Version: 1.2
Interface: 5

You can reproduce by connecting EQMod Mount in simulation mode.  Then try to set the mount location either via the Site Management tab in an INDI control panel, or by using indi_setprop.

```
indi_setprop "EQMod Mount.GEOGRAPHIC_COORD.LAT;LONG;ELEV=44.8103;-93.9398;300"
```

This will cause the EQMod Mount driver to crash.  I can find the following kernel errors in /var/log/syslog.  I think this suggests some sort of memory pointer issue.  Is it possible this is only trapped by ARM processors?

```
Jan  2 19:12:38 astroberry kernel: [17633.673572] Alignment trap: not handling instruction ed917b06 at [<00451e4c>]
Jan  2 19:12:38 astroberry kernel: [17633.673591] 8<--- cut here ---
Jan  2 19:12:38 astroberry kernel: [17633.673620] Unhandled fault: alignment exception (0x221) at 0xb6e0c6ed
Jan  2 19:12:38 astroberry kernel: [17633.673644] pgd = e4e6a26c
Jan  2 19:12:38 astroberry kernel: [17633.673666] [b6e0c6ed] *pgd=06768003, *pmd=f7073003
```

arun289, 2022-01-03:
I am having the exact same problem in my Raspberry PI after Astroberry update today.
```
[  103.349143] Alignment trap: not handling instruction ed917b06 at [<00431e4c>]
[  103.349152] 8<--- cut here ---
[  103.349165] Unhandled fault: alignment exception (0x221) at 0xb6e7a6ed
[  103.349176] pgd = 326a05e4
[  103.349185] [b6e7a6ed] *pgd=06121003, *pmd=7cd45003
```

setnes, 2022-01-04:
This is more than just the location.  I can get crashes at different times too.  If I turn on or off tracking, for example, the driver crashes.  Not all updates crash the driver, but enough do to make it unusable.

rkaczorek, 2022-01-04:
The issue has been fixed. The root cause of the issue was an error in pushing latest indi libs to the astroberry repository.
Please update your system and retest.

arun289, 2022-01-04:
Thanks @rkaczorek for the quick update. It works now.

TallFurryMan, 2022-06-19:
@setnes Please close if it works for you too.

setnes, 2022-06-19:
Yes, this was fixed in January.

closed 2022-06-19 by setnes