indilib/indi-3rdparty #519: [Help] Android support

daoudeddy, 2022-01-03:
I've been working on building INDI Lib/3rd party libraries on Android and i have successfully build 100% of indilib and 80% of 3rd party libraries.

The building system is the Termux App, which provides similar functionalities to Linux but limited by Android system.

The only problem I'm facing is that all the pre-built shared libraries provided by "Closed Source" minded companies can't run on Termux, you can read more [here](https://wiki.termux.com/wiki/Differences_from_Linux)

I've created a Github [repo](https://github.com/daoudeddy/termux-indilib) that provide a script with all the needed changes to install the needed libraries, patching and building both indilib and indi-3rdparty from source, and it contains a list of what is supported and what is not, I don't have all the equipments to test all the libraries but im sure the community will test and provide help. I hope this will become official at some point.

Thank  you.

knro, 2022-01-04:
Thank you for this effort. So how would someone connect the devices to an INDI phone/tablet? using some sort of USB adapter? can you use USB hub? I checked your patch and it looks good. A couple of points:
1. Use if Android when possible in all places. I noticed your 3rd party patch turns off by default some drivers, this should only be turned off on Android and not globally.
2. udev rules should also have an if not android or something similar to all the files. I saw you only commented qhy but left the others?

daoudeddy, 2022-01-04:
I've tested on an old USB otg device and on a new device using Type C hub both worked well.

As you can see in indilib i did the android check but not yet in the 3rd party, i will do it for sure but i needed some help from the community for more tests