indilib/indi-3rdparty #531: indi-mi: turning off cooling does not work as expected

peter-englmaier, 2022-01-24:
**Describe the bug**
When I select cooler 'Off' in the indi control panel, the cooling is going to maximum instead of cooling down the chip. I can turn off cooling only by setting temperature to +50 Celsius. 

**To Reproduce**
Start Indi
Cool to some temperature below ambient.
Turn off cooling by clicking on cooler "OFF".
Temperature goes to maximum instead of turning off.

**Expected behavior**
Temperature should rise regulated to ambient temperature and then cooling should stop.

I'm using the latest version (bleeding).

peter-englmaier, 2022-01-24:
This happens with my camera "C4-160000EC". 

Perhaps the following setting (indi_mi.cpp) is too high for this camera?
```
#define TEMP_COOLER_OFF 100  /* High enough temperature for the camera cooler to turn off (°C) */
```

peter-englmaier, 2022-03-26:
The bug description was confusing, sorry for that. I think it is now more readable.

I also want to comment, that I have two cameras attached MI C1-5000A as guiding cam and MI C4-16000EC as imaging cam.

sWski, 2022-08-01:
Hello Peter,

this was fixed in library by #175, however you have CMOS camera and there is same problem in camera firmware. Some have negative compensation value in their EEPROM and by setting high temperature it overflows. This can be fixed by updating camera firmware, which should be release in upcoming weeks.

knro, 2022-08-04:
@peter-englmaier Can you please confirm the fix?

peter-englmaier, 2022-08-10:
Wonderful!
Yes, I will test as soon as firmware has been released. So, far I see only firmware released May 16th, 2022 on this page:
https://www.gxccd.com/cat?id=201&lang=409
@sWski : would you please be so kind and post firmware version number here, once it is released?

peter-englmaier, 2022-09-19:
Hi again, sorry for my long silence. I just upgraded the camera to the latest firmware and tested with kstars. Now, when I turn cooling off in kstars, the camera cools down as expected.
Problem solved!
Best, Peter

closed 2022-09-19 by peter-englmaier