indilib/indi-3rdparty #543: celestron-aux Capture and solve always slew to target while allowing taking pictures

luzik, 2022-02-12:
I don't know this is a bug or a feature, but with new driver, when I set action to sync and press "capture and solve" it is syncing and  moving to target (fixing its error). This can be a good thing (not sure yet), but user is able to take photos while scope is moving, so I got lot of star trails, and don't know when this move ends.
When using goto command, ekos block ability to take pictures and this is good thing.

jochym, 2022-02-12:
This is how it worked before. The problem may be with the blocking mechanism. It should be a normal goto, so it should block capture.

luzik, 2022-02-12:
I believe that old behavior with selected 'sync' action did not causes mount to change its ra/dec, now it's moving. "Slew to target" action did, and do with proper blocking capture.

knro, 2022-02-13:
This looks like an Ekos issue and not INDI.

knro, 2022-02-13:
But while we're on this topic, @jochym do you know how we can support Celestron on a Wedge? While I was working on fixing some issues I found (probably due to my lack of understanding), I changed the Alignment subsystem so that equatorial mounts use direct TDV <--> EQ Coord conversion, not passing via any Alt-Az conversion phase. This made EQMod and other mount finally work perfectly OK with the alignment subsystem.

However, I'm not sure how to handle Wedge case, and even equatorial mode is not even *tested* yet.