indilib/indi-3rdparty #544: celestron-aux tracking and parking in same time is allowed

luzik, 2022-02-12:
Is this ok to park and track in the same time ?

[image]

jochym, 2022-02-12:
Definitely not. Is the mount actually tracking (moving)?

luzik, 2022-02-12:
I did't check that. Will do.

luzik, 2022-02-13:
When  I press tracking ON button while mount is parked, AZ, ALT and DEC values are changing. RA stays still

knro, 2022-03-22:
If the mount is parked, it's not possible to toggle tracking ON. When mount is parked, RA values should be changing and AZ/AL are static. Please include mount log so we can check what's going on.