indilib/indi-3rdparty #545: celestron-aux Missed move command in one axis

luzik, 2022-02-13:
While testing caux using inbuilt math plugin, align was as bad as usual (10 sec max exposure time), and then I parked in sweet spot of alt ~0, az ~0 and cleared old sync points. My first point solver error was only ~200arcsec (this mean that park position was almost perfect), second and third great  too. With this math set up I was able to take 60seconds pictures with stars as dots!
This also confirm how correct park position 0,0 affect alignment with 'inbuilt math' plugin.

The BUG:
I wanted add one more align point, set goto (it was a vector from actual location 10deg up, ~30 deg left) and my mount did not move in AZ at all, but KStars showed this move on map and plate solve give me over 90000arcsec error. I parked and park position physically also changed by about 30deg AZ.

Something similar did happened me few days earlier, but I wasn't sure what really happened.

I know that this is not reproducible, I do understand that there is not enough data to track a bug, but please help me set up debug for such behavior and I will provide as much info I can.

Can it be some kind of hardware issue ?

Can it be even possible this happened ? I always thought that scope position in Kstars is taken as feedback from mount physical encoders that just can't be wrong.

knro, 2022-03-22:
For this driver specifically, there are multiple translations steps since it goes via the mount model. Enable logging to see what actually happens at the lowest level, this is only way to possible track bugs of this kind. I couldn't use the in-built mount model for this, and had to develop the Nearest plugin which turns out to be more reliable in real-life scenarios.

Are you using this for Alt-Az mount? or Equatorial? I never tested on the latter and there are possibly quite a few bugs with equatorial handling.

knro, 2022-11-16:
Error reporting instructions here: https://indilib.org/individuals/logs-howto.html

A better method is to use Ekos Debugger

`sudo apt-get -y install ekosdebugger`

knro, 2022-11-20:
@luzik Any further updates on this? INDI v1.9.9 is due next week so we have time to fix if we could identify the root cause.

closed 2024-03-17 by knro