indilib/indi-3rdparty #557: indi-qhy fails to compile

RononDex, 2022-03-04:
**Describe the bug**
When trying to compile / build indi-qhy it fails with the following error:

```
-- Found CFITSIO: /usr/lib/libcfitsio.so
-- Found libqhyccd: /usr/lib/libqhyccd.so
-- Found libnova: /usr/lib/libnova.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cobra/packages/indi-3rdparty/build/indi-qhy
Consolidate compiler generated dependencies of target qhy_ccd_test
Consolidate compiler generated dependencies of target qhy_video_test
Consolidate compiler generated dependencies of target indi_qhy_ccd
[ 16%] Built target qhy_ccd_test
[ 16%] Built target qhy_video_test
[ 16%] Building CXX object CMakeFiles/indi_qhy_ccd.dir/qhy_ccd.cpp.o
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp: In member function ‘virtual bool QHYCCD::updateProperties()’:
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:410:9: error: ‘CaptureFormat’ was not declared in this scope
  410 |         CaptureFormat format;
      |         ^~~~~~~~~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:413:13: error: ‘format’ was not declared in this scope
  413 |             format = {"INDI_RAW", "RAW", 16, true};
      |             ^~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:417:13: error: ‘format’ was not declared in this scope
  417 |             format = {"INDI_MONO", "Mono", 16, true};
      |             ^~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:419:26: error: ‘format’ was not declared in this scope
  419 |         addCaptureFormat(format);
      |                          ^~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:419:9: error: ‘addCaptureFormat’ was not declared in this scope
  419 |         addCaptureFormat(format);
      |         ^~~~~~~~~~~~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp: In member function ‘virtual bool QHYCCD::StartExposure(float)’:
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:1310:37: warning: comparison of integer expressions of different signedness: ‘uint32_t’ {aka ‘unsigned int’} and ‘long int’ [-Wsign-compare]
 1310 |         if (m_LastExposureRequestuS != uSecs)
      |             ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
make[2]: *** [CMakeFiles/indi_qhy_ccd.dir/build.make:76: CMakeFiles/indi_qhy_ccd.dir/qhy_ccd.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/indi_qhy_ccd.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
Consolidate compiler generated dependencies of target indi_qhy_ccd
[ 16%] Building CXX object CMakeFiles/indi_qhy_ccd.dir/qhy_ccd.cpp.o
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp: In member function ‘virtual bool QHYCCD::updateProperties()’:
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:410:9: error: ‘CaptureFormat’ was not declared in this scope
  410 |         CaptureFormat format;
      |         ^~~~~~~~~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:413:13: error: ‘format’ was not declared in this scope
  413 |             format = {"INDI_RAW", "RAW", 16, true};
      |             ^~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:417:13: error: ‘format’ was not declared in this scope
  417 |             format = {"INDI_MONO", "Mono", 16, true};
      |             ^~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:419:26: error: ‘format’ was not declared in this scope
  419 |         addCaptureFormat(format);
      |                          ^~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:419:9: error: ‘addCaptureFormat’ was not declared in this scope
  419 |         addCaptureFormat(format);
      |         ^~~~~~~~~~~~~~~~
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp: In member function ‘virtual bool QHYCCD::StartExposure(float)’:
/home/cobra/packages/indi-3rdparty/indi-qhy/qhy_ccd.cpp:1310:37: warning: comparison of integer expressions of different signedness: ‘uint32_t’ {aka ‘unsigned int’} and ‘long int’ [-Wsign-compare]
 1310 |         if (m_LastExposureRequestuS != uSecs)
      |             ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
make[2]: *** [CMakeFiles/indi_qhy_ccd.dir/build.make:76: CMakeFiles/indi_qhy_ccd.dir/qhy_ccd.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/indi_qhy_ccd.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

```

**To Reproduce**
Exact steps to reproduce the behavior.
1. Compile Driver
2. Error

**Expected behavior**
No compilation errors

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Arch Linux ARM (armv7h)
 - Current master branch commit

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

RononDex, 2022-03-04:
Installed is `libqhyccd.so.22.3.3`, maybe the code does not compile with that version of the sdk (it is the one shipped with `libqhy`)

RononDex, 2022-03-04:
Seems to be related to https://github.com/indilib/indi-3rdparty/commit/3b62bcb3cae93062421d041fe38b8e4eb09595c2 @knro

knro, 2022-03-04:
You need to update libindi first.

RononDex, 2022-03-07:
compiling libindi from source and the compiling the driver worked thanks.

Seems the libindi package is out of date in the package manger :)

closed 2022-03-07 by RononDex