indilib/indi-3rdparty #558: INDIRTL-SDR driver does not work properly

T-STUDIO, 2022-03-06:
I tried all the functions of the INDI RTL-SDR driver on the driver because I wanted to observe the radio waves, but there were many places where it did not work.

environment
・ Raspberry-Pi4 4GB Buster (latest for both OS and INDI)
・ RTL-SDR.COM module V3 (latest)

Defective part
・Even if the import location of the FITS image is changed, it is not reflected.
・ FITS viewer does not open even if I get a FITS image
・ Streaming mode does not work
・INDI Web Manager does not show drivers
・Crash when activating an item on the Signal Processing tab.

The log at the time of the above operation check is also attached.

Please check at the time of maintenance.

The log when it is operated is attached here.

https://indilib.org/forum/wish-list/11399-indi-rtl-sdr-driver-maintenance.html#81113

knro, 2023-08-11:
@iliaplatone Any update on this issue? does it work from your side?

iliaplatone, 2023-08-11:
Yes it works, Just some issue on libsdp, already made a PR for that: the convolution didn't work becasue axes were swapped.

iliaplatone, 2023-08-11:
Also consider integrating OpenVLBI from the official repositiories which includes libopenvlbi for radio interferometers and libdsp itself so we don't have to integrate it into INDI